espressif/esp-matter #214: Can't get the light example commissioned with neither CHIP, Smartthings, or Google.

momo1, 2023-02-12:
Hi,

I flashed the `light` example following the instructions [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html). I even deleted all previous IDF and Matter repos and tools and started fresh. I tried commissioning using Smarththings, Google, and CHIP and they all fail. I tried each method over 10 times.

I'm using an ESP32 (WROOM-32) with IDF 4.4.3 and I `erase-flash` before every flash. I've attached the logs when trying CHIP (ESP and CHIP logs), ST, and Google.

For reference:
```
> matter config
VendorId:        65521 (0xFFF1)
ProductId:       32768 (0x8000)
HardwareVersion: 0 (0x0)
PinCode:         20202021
Discriminator:   f00
Done
```

What am I missing? 

Thanks

[chip.log](https://github.com/espressif/esp-matter/files/10717574/chip.log)
[espChip.log](https://github.com/espressif/esp-matter/files/10717575/espChip.log)
[espGoogle.log](https://github.com/espressif/esp-matter/files/10717576/espGoogle.log)
[espST.log](https://github.com/espressif/esp-matter/files/10717577/espST.log)

dhrishi, 2023-02-13:
@momo1 All the logs except `espGoogle.log` show that the device did not receive the CASE Sigma1 message on the IP network. The reason could be that the host-machine running the chip-tool or the commissioner phone app are using a different Wi-Fi network than the one configured on the ESP32. Can you please confirm that all the devices are in the same Wi-Fi network?

`espGoogle.log` log indicates that the project is not registered on the Google developer console. Please refer the table `Table: Vendor ID (VID) and device type pairing scenarios` [here](https://developers.home.google.com/matter/get-started) and add a project in your account. In this case too, please confirm that you use the same Wi-Fi network on both the sides.

momo1, 2023-02-14:
Thanks. I can confirm that they're all on the same network. There's only one test network. As for the Google bit thanks! I'll drop the Google test until I register the device.

Not sure if it's related at all but I also tried `matter wifi connect <ssid> <password>` per the instructions [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/faq.html) and that results in `Error: 45`

momo1, 2023-02-14:
Update: After trying CHIP about 20 more times and constant resets to ESP I got the device to commission a couple of time out of all the trials. It's extremely flaky and I can't figure out a pattern. Not sure how helpful that is but that's the latest.

dhrishi, 2023-02-14:
@momo1 I think it can be some problem related to mDNS with the AP. During the commissioning, when the ESP32 connects to the Wi-Fi network, can you please enter the below command on a host-machine connected to the same Wi-Fi:

`avahi-browse _matterc._udp -r` or
`dns-sd -B _matterc._udp` (on MacOS)

This will help us determine if other devices on the network are able to receive mDNS advertisements from the ESP32.

Also, is it possible to try out with a different AP to see if you encounter this issue?

momo1, 2023-02-14:
I seem to be getting mDNS advertisements from the ESP. The logs below are from two different physical machines connected to the same network. This is from when the commissioning actually failed. I doubt (but will not discount) that it would be mDNS because I do use mDNS for a bunch of other unrelated machines on the same AP and it works fine. I will also try another AP but that would take a bit to setup.

```
m@imac:~ $ dns-sd -B _matterc._udp
Browsing for _matterc._udp
DATE: ---Mon 13 Feb 2023---
21:29:07.472  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
21:29:07.473  Add        2   9 local.               _matterc._udp.       0A778D4CE314C2C4
21:29:28.474  Rmv        0   9 local.               _matterc._udp.       0A778D4CE314C2C4
21:29:33.521  Add        2   9 local.               _matterc._udp.       DF8F4BDF195C2B89
```

```
m@desktop:~ $ avahi-browse _matterc._udp -r
+ enp4s0 IPv4 0A778D4CE314C2C4                              _matterc._udp        local
= enp4s0 IPv4 0A778D4CE314C2C4                              _matterc._udp        local
   hostname = [AC67B23666A0.local]
   address = [192.168.1.52]
   port = [5540]
   txt = ["PI=" "PH=33" "CM=1" "D=3840" "T=1" "VP=65521+32768"]

- enp4s0 IPv4 0A778D4CE314C2C4                              _matterc._udp        local
+ enp4s0 IPv4 DF8F4BDF195C2B89                              _matterc._udp        local
= enp4s0 IPv4 DF8F4BDF195C2B89                              _matterc._udp        local
   hostname = [AC67B23666A0.local]
   address = [192.168.1.52]
   port = [5540]
   txt = ["PI=" "PH=36" "CM=1" "D=3840" "T=1" "VP=65521+32768"]
```

Also after failing I can keep restarting the ESP to get:

```
Browsing for _matterc._udp
DATE: ---Mon 13 Feb 2023---
21:40:34.050  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
21:40:34.051  Add        3   9 local.               _matterc._udp.       DF8F4BDF195C2B89
21:40:34.051  Add        2   9 local.               _matterc._udp.       6E91BC311AD91006
21:40:39.170  Rmv        0   9 local.               _matterc._udp.       DF8F4BDF195C2B89
21:40:41.906  Add        2   9 local.               _matterc._udp.       ACDD59FD64596A2B
21:40:56.131  Add        2   9 local.               _matterc._udp.       FEB2E8D5367AC0FD
21:41:22.454  Add        2   9 local.               _matterc._udp.       BFBEC6F94C854BC5
21:41:33.512  Add        2   9 local.               _matterc._udp.       0FC2A3A1AFF181B9
```

HTH

dhrishi, 2023-02-14:
Ok. Can you please provide the make and model of your AP?

>Not sure if it's related at all but I also tried matter wifi connect <ssid> <password> per the instructions [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/faq.html) and that results in Error: 45

Thanks for reporting this. The CLI is now updated to - `matter esp wifi connect ssid passphrase`. We will update the documentation.

momo1, 2023-02-18:
It seems you're right. I changed the AP and it seems to work reliably with the other AP (Linksys EA6350). The AP I'm having an issue with is an Orbi RBR750 but I can't find anything in the settings so far that could be interfering with mDNS.

dhrishi, 2023-02-18:
>I changed the AP and it seems to work reliably with the other AP (Linksys EA6350). 
 
@momo1 Good to know that it works fine

Can you check if IGMP proxy is disabled on the Orbi RBR750 AP? If yes, it may cause some issues for multicast traffic. You can find the option in the configuration settings of your AP. If you want, you can try to enable the option and see if it helps.

dhrishi, 2023-02-21:
Closing as this doesn't look like an issue on the device side. @momo1 Feel free to post an update

closed 2023-02-21 by dhrishi

MukeshBisht, 2023-03-06:
> Thanks. I can confirm that they're all on the same network. There's only one test network. As for the Google bit thanks! I'll drop the Google test until I register the device.
>
> Not sure if it's [...]

matter esp wifi connect <ssid> <password> worked for me