espressif/esp-matter #215: About the generation of the discriminator and pincode

shun-dgut, 2023-02-13:
1. Is the pincode generated by salt and verifier and iteration-count? Does this mean that my device only needs to store the pincode
2. Can the chiptool tool that generates the QR code payload run on windows

shubhamdp, 2023-02-13:
1. Yes, indeed, pincode is generated using salt, verifier, and iteration-count. Is the opposite, as passcode is distributed using qrcode or manual pairing code, device should store salt, verifier, and iteration-count.

2. As far as setting matter on windows, it can be setup using WSL, take a look here: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#windows-10. If pairing code generation is the only dependency then take a look at python based tool as well,  https://github.com/project-chip/connectedhomeip/tree/master/src/setup_payload/python.

shun-dgut, 2023-02-13:
> 1. Yes, indeed, pincode is generated using salt, verifier, and iteration-count. Is the opposite, as passcode is distributed using qrcode or manual pairing code, device should store salt, verifier, [...]

Thanks, but I wonder why I can't use the pincode directly on the device, but need to regenerate.

shubhamdp, 2023-02-13:
Please check section 5.1.7. Generation of the Passcode  in Matter Specification  v1.0. It has more insights related to this.
In general you can not place the verifier and passcode at the same place as per specification. There is one option though, with some clause about how to store both on device.

I am not well aware about the regeneration logic, so probably https://github.com/project-chip/connectedhomeip/blob/master/src/protocols/secure_channel/PASESession.cpp is a good place to look for.

shubhamdp, 2023-02-13:
@shun-dgut Please close the issue if your question is answered.

closed 2023-02-13 by shun-dgut