espressif/esp-matter #218: Commisioning issue with android CHIP-tool

MartinPArmstrong, 2023-02-14:
Hi. 

I am unable to commison the device using the Android CHIP-tool. I am however, able to commsion using google home.
Using a ESP32S3 with the ZAP-light example, the same issue remains on the other examples aswell.

```
I (24) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (25) boot: compile time 02:00:20
I (25) boot: chip revision: 0
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (32) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (65) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (72) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (80) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (87) boot:  4 ota_0            OTA app          00 10 00020000 001e0000
I (95) boot:  5 ota_1            OTA app          00 11 00200000 001e0000
I (102) boot:  6 fctry            WiFi data        01 02 003e0000 00006000
I (110) boot: End of partition table
I (114) esp_image: segment 0: paddr=00020020 vaddr=3c0e0020 size=305b8h (198072) map
I (158) esp_image: segment 1: paddr=000505e0 vaddr=3fc97ca0 size=04b24h ( 19236) load
I (162) esp_image: segment 2: paddr=0005510c vaddr=40374000 size=0af0ch ( 44812) load
I (174) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=dd260h (905824) map
I (336) esp_image: segment 4: paddr=0013d288 vaddr=4037ef0c size=08d90h ( 36240) load
I (345) esp_image: segment 5: paddr=00146020 vaddr=50000000 size=00010h (    16) load
I (353) boot: Loaded app from partition at offset 0x20000
I (353) boot: Disabling RNG early entropy source...
I (365) cpu_start: Pro cpu up.
I (365) cpu_start: Starting app cpu, entry point is 0x40375458
0x40375458: call_start_cpu1 at /home/martin/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (379) cpu_start: Pro cpu start user code
I (379) cpu_start: cpu freq: 160000000
I (379) cpu_start: Application information:
I (382) cpu_start: Project name:     zap_light
I (387) cpu_start: App version:      v1.0
I (392) cpu_start: Compile time:     Feb  3 2023 02:00:16
I (398) cpu_start: ELF file SHA256:  b3ab3a573693aeb8...
I (404) cpu_start: ESP-IDF:          v4.4.3-dirty
I (409) heap_init: Initializing. RAM available for dynamic allocation:
I (417) heap_init: At 3FCACAA8 len 0003CC68 (243 KiB): D/IRAM
I (423) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (430) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (436) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (443) spi_flash: detected chip: generic
I (447) spi_flash: flash io: dio
W (451) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (465) sleep: Configure to isolate all GPIO pins in sleep state
I (471) sleep: Enable automatic switching of GPIO sleep configuration
I (478) coexist: coexist rom version e7ae62f
I (483) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (546) led_driver_ws2812: Initializing light driver
I (546) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (546) pp: pp rom version: e7ae62f
I (556) net80211: net80211 rom version: e7ae62f
I (566) wifi:wifi driver task: 3fceda58, prio:23, stack:6656, core=0
I (566) system_api: Base MAC address is not set
I (566) system_api: read default base MAC address from EFUSE
I (576) wifi:wifi firmware version: 8cb87ff
I (576) wifi:wifi certification version: v7.0
I (586) wifi:config NVS flash: enabled
I (586) wifi:config nano formating: disabled
I (586) wifi:Init data frame dynamic rx buffer num: 32
I (596) wifi:Init management frame dynamic rx buffer num: 32
[... 271 lines cut ...]
I (172306) NimBLE: 

I (172306) chip[DL]: CHIPoBLE advertising started
I (172306) app_main: Commissioning window opened
E (172316) chip[BLE]: no endpoint for BLE sent data ack
E (172316) chip[BLE]: no endpoint for unsub recvd
I (173166) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
I (173166) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (173176) chip[DL]: Device already advertising, stop active advertisement and restart
I (173176) NimBLE: GAP procedure initiated: stop advertising.

I (173186) NimBLE: GAP procedure initiated: advertise; 
I (173196) NimBLE: disc_mode=2
I (173196) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (173206) NimBLE: 

I (202146) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
I (202146) chip[DL]: Device already advertising, stop active advertisement and restart
I (202156) NimBLE: GAP procedure initiated: stop advertising.

I (202156) NimBLE: GAP procedure initiated: advertise; 
I (202166) NimBLE: disc_mode=2
I (202166) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (202186) NimBLE:
```

jonsmirl, 2023-02-14:
Android CHIPTool is not maintained very well and it is broken much of the time. Try GSAFM instead... 
https://github.com/google-home/sample-app-for-matter-android

dhrishi, 2023-02-15:
@MartinPArmstrong As per the logs, the CSR (after the attestation request is successful) request is not received from the commissioner and hence the fail saif timer expires and the commissioning fails. Is it possible to check using other commissioners like host-based chip-tool, Google Home, Apple Home or the custom Android app as mentioned by @jonsmirl above?

closed 2023-02-15 by MartinPArmstrong