espressif/esp-matter #219: Communication between esp32 using Windows 11 as development host and ESP-IDF

SergiSoler8, 2023-02-15:
Hi,

I have three LOLIN D32 boards with esp32 chips and two temperature sensors DS18B20. My objective is to have two boards with a temperature sensor each one and the other board to act as a central node, which would receive the information from the two boards with temperature sensors. My question is to know if the board that acts as the central node can receive the information from the two boards with sensors through the MATTER protocol instead of using the ESP-NOW protocol, all this with Windows 11 using WSL and the ESP-IDF framework.

Regards, thank you.

dhrishi, 2023-02-16:
@SergiSoler8 What is the use/purpose of the central node? Does it send the temperature values to the cloud or something and the other nodes/boards do not connect to the internet? If you explain your usecase then I can provide an appropriate solution.

>all this with Windows 11 using WSL and the ESP-IDF framework.

We have tested Windows 10 with WSL and not Windows 11. Please see the documentation [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#windows-10)

SergiSoler8, 2023-02-16:
@dhrishi Hi,

The purpose of the central node is to send the temperature values to a web server. Only the board that acts as the central node would be connected to the internet to send the values to the web server, while the rest of the boards would not be connected to the internet, because they would send the temperature values to the central board using the Matter protocol.

Regards, thank you.

dhrishi, 2023-02-16:
@SergiSoler8 Ok. There are 2 options -

1. 
The 2 other boards having temperature sensors connected can run the Matter stack with a device type having the temperature measurement server cluster. The central node can run the Matter stack with a device type having the temperature measurement client cluster. This would not require using the ESP-Now protocol at all.

Now, a user can commission/setup all the three devices into the Matter network. Once done, a binding relationship can be established between the central node and the 2 boards, after which the central node can subscribe to the temperature attribute and get the updates over Matter when the temperature changes. It can then send it to the web server, in the same way that it is currently doing.

2.
Your central node runs Matter to ESP-Now bridge and bridges the non-Matter temperature sensor boards so that they are available over Matter. In this case, you won't have to change anything on the temperature sensor boards and they would continue using ESP-Now. But, you will have to additionally add Matter bridge functionality on the central node. Sending the temperature to the web server still stays the same.

dhrishi, 2023-02-17:
Closing as the question is answered

closed 2023-02-17 by dhrishi