espressif/esp-matter #220: Time synchronization

jonsmirl, 2023-02-16:
What is the state of time synchronization? Do Google/Apple/Amazon implement the Time Synchronization server? Can esp-matter sync to it as a client?  Or is the assumption that esp-matter nodes are going to use NTP?

emmby, 2023-02-16:
I'm able to create a TimeSynchronization cluster, but the TimeZone attribute doesn't seem to be available when I try to retrieve it:

```c
  endpoint_t *root_endpoint = endpoint::get(node,0);

  // NOTE: time_cluster is NULL when using get
  // esp_matter::cluster_t *time_cluster = esp_matter::cluster::get(root_endpoint, TimeSynchronization::Id);

  esp_matter::cluster::time_synchronization::config_t time_config;
  esp_matter::cluster_t *time_cluster = esp_matter::cluster::time_synchronization::create(root_endpoint, &time_config, CLUSTER_FLAG_CLIENT);  
  
  attribute_t *timezone_attribute = attribute::get(time_cluster,TimeSynchronization::Attributes::TimeZone::Id);
  // timezone_attribute is NULL
```

jonsmirl, 2023-02-16:
I am not clear totally how this is supposed to work. I'd guess that you create a TimeSynchronization client cluster and then bind it to the G/A/A controller.  If it's not bound it doesn't know who to ask, right? So did G/A/A implement TimeSynchronization server, and did they set up the ACLs so that you can bind to it?

Plan B, each device uses NTP. But then I need some way to tell the device what timezone it is in.  

I also checked my various home routers, none of them are advertising NTP service in their DHCP, so it will have to be pool.ntp.org.  The basic idea is that the G/A/A devices use pool.ntp.org and then the rest of the devices piggyback off from the G/A/A device. But have they implemented this?

jonsmirl, 2023-02-16:
My suspicion is that none of this works in CHIP 1.0 so you're on your own.  Luckily ESPIDF has NTP support, so the only real hurdle is having the user set the time zone in somehow.

shubhamdp, 2023-02-17:
@jonsmirl 
> What is the state of time synchronization?

As per Matter Specification, Time Synchronization feature is provisional so, not yet implemented.

> Do Google/Apple/Amazon implement the Time Synchronization server?

I have not checked with Google/Apple Home app.

> Can esp-matter sync to it as a client? Or is the assumption that esp-matter nodes are going to use NTP?

If implemented as per specification then, it mentions administrator can set the time/time-source on the device or, ESP32 devices with Internet connectivity can use SNTP to get the current time.

shubhamdp, 2023-02-17:
Spec mentions few scenarios/options regarding how this is supposed to work:
1. Administrator can configure time, time source and then device uses them to get time.
2. SNTP is one option
3. During commissioning, if any node is capable of fetching time then that can be configured as time server and administrator can configure it as the time source for other sleepy/lazy nodes.
4. There are few more scenarios 

G/A/A does not support bindings yet, no idea how that's gonna work.

jonsmirl, 2023-02-17:
I'll probably just use the IDF support for NTP on my device which needs time (only one device type needs it).

However, it doesn't look too hard to implement TimeSync server as an ESP demo since the IDF has NTP support. This doesn't appear to be production ready, so it'd just be a demo. The best solution here is for G/A/A to implement because they know the timezone and DST status.

emmby, 2023-02-28:
As of today, time synchronization and localization are listed as unsupported: https://developers.home.google.com/matter/get-started?authuser=0#unsupported_features

jonsmirl, 2023-03-24:
I'll leave this for next release. I want the time from Matter so that my devices will know which timezone and if DST is in effect. It is annoying to set that into each device manually.   The controllers know timezone/DST, they need to share that info with the devices.

closed 2023-03-24 by jonsmirl

abu-matterize, 2025-12-10:
what's it's status now? can a ESP32H2 (Thread) act as sntp client to sync time from ntp pool?