espressif/esp-matter #222: Read basic information

stefano055415, 2023-02-17:
I'm trying to read the basic information of the device.

with mfg_tool I created the partition with all the data (manufacturer, model, serial...)

if i try to read this data with chip-tool i read it correctly

./chip-tool any read-by-id 0x28 0x1 0x11 0

but if I try to read them from code they return empty

cluster_t *basic_information_cluster = cluster::get(endpoint, BasicInformation::Id);
attribute_t *basic_information_attribute = attribute::get(basic_information_cluster, BasicInformation::Attributes::VendorName::Id);
esp_matter_attr_val_t basic_information_val = esp_matter_invalid(NULL);
attribute::get_val(basic_information_attribute, &basic_information_val);
	
if (basic_information_val.type == ESP_MATTER_VAL_TYPE_CHAR_STRING){
		ESP_LOGI(TAG, "********** Endpoint 0x%04X's Cluster 0x%04X's Attribute 0x%04X is %.*s **********", 0,
						 BasicInformation::Id, BasicInformation::Attributes::VendorName::Id, basic_information_val.val.a.s, basic_information_val.val.a.b);
}

jonsmirl, 2023-02-17:
I added some Kconfig and put this in CMakelists.txt. I am not finding mfg_tool useful until production time. You probably did not copy the mfg_partition onto the device.
~~~
message(STATUS "Vendor ID: " ${CONFIG_DEVICE_VENDOR_ID})
message(STATUS "Vendor Name: " ${CONFIG_DEVICE_VENDOR_NAME})
message(STATUS "Product ID: " ${CONFIG_DEVICE_PRODUCT_ID})
message(STATUS "Product Name: " ${CONFIG_DEVICE_PRODUCT_NAME})
message(STATUS "SW Version String: " ${CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING})
idf_build_set_property(COMPILE_OPTIONS "-DCHIP_DEVICE_CONFIG_DEVICE_VENDOR_ID=${CONFIG_DEVICE_VENDOR_ID}" APPEND)
idf_build_set_property(COMPILE_OPTIONS -DCHIP_DEVICE_CONFIG_DEVICE_VENDOR_NAME="${CONFIG_DEVICE_VENDOR_NAME}" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-DCHIP_DEVICE_CONFIG_DEVICE_PRODUCT_ID=${CONFIG_DEVICE_PRODUCT_ID}" APPEND)
idf_build_set_property(COMPILE_OPTIONS -DCHIP_DEVICE_CONFIG_DEVICE_PRODUCT_NAME="${CONFIG_DEVICE_PRODUCT_NAME}" APPEND)
if(NOT ${CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING} STREQUAL "")
    idf_build_set_property(COMPILE_OPTIONS "-DCHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING=\"${CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING}\"" APPEND)
endif()
~~~

stefano055415, 2023-02-17:
Hi @jonsmirl , I created the partition and wrote it on the device and in fact with chip-tool I can correctly read the vendor name, product name,...

but i can't read them from the code

cluster_t *basic_information_cluster = cluster::get(endpoint, BasicInformation::Id);
attribute_t *basic_information_attribute = attribute::get(basic_information_cluster, BasicInformation::Attributes::VendorName::Id);
esp_matter_attr_val_t basic_information_val = esp_matter_invalid(NULL);
attribute::get_val(basic_information_attribute, &basic_information_val);

if (basic_information_val.type == ESP_MATTER_VAL_TYPE_CHAR_STRING){
ESP_LOGI(TAG, "********** Endpoint 0x%04X's Cluster 0x%04X's Attribute 0x%04X is %.*s **********", 0,
BasicInformation::Id, BasicInformation::Attributes::VendorName::Id, basic_information_val.val.a.s, basic_information_val.val.a.b);
}

its the same if i try to read with matter esp console

matter esp attribute get 0x0 0x28 0x1
I (432668) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0028's Attribute 0x0001 is **********
I (432670) esp_matter_attribute: ********** Endpoint 0x0000's Cluster 0x0028's Attribute 0x0001 is **********
Done

dhrishi, 2023-02-17:
@stefano055415 Some of the generic attributes are internally managed by the Matter stack and the callbacks for read/write are not given to the application. The basic information cluster attributes are managed that way.

Also, if you see, the esp-matter data model does not allocate memory for such attributes. See [this](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.cpp#L199-L210).

If you want to locally read such attributes, you can do so from the factory partition. Please see how it is read from the Matter stack [here](https://github.com/project-chip/connectedhomeip/blob/v1.0.0.2/src/platform/ESP32/ESP32FactoryDataProvider.cpp#L158)

stefano055415, 2023-02-20:
@dhrishi 
Ok perfect, thanks

closed 2023-02-20 by stefano055415

svenehlers, 2024-03-08:
Hi @jonsmirl , i tried your suggestion to add the lines to the CMakeLists.txt but i while building i get this error: "I have no idea what this is." After removing the lines from CMakeLists.txt it is working.

Added Lines:
`
message(STATUS "Vendor Name: " ${CONFIG_DEVICE_VENDOR_NAME})
message(STATUS "Product Name: " ${CONFIG_DEVICE_PRODUCT_NAME})
message(STATUS "HW Version String: " ${CONFIG_DEVICE_HARDWARE_VERSION_STRING})
idf_build_set_property(COMPILE_OPTIONS -DCHIP_DEVICE_CONFIG_DEVICE_VENDOR_NAME="${CONFIG_DEVICE_VENDOR_NAME}" APPEND)
idf_build_set_property(COMPILE_OPTIONS -DCHIP_DEVICE_CONFIG_DEVICE_PRODUCT_NAME="${CONFIG_DEVICE_PRODUCT_NAME}" APPEND)
if(NOT ${CHIP_DEVICE_CONFIG_DEVICE_HARDWARE_VERSION_STRING} STREQUAL "")
    idf_build_set_property(COMPILE_OPTIONS "-DCHIP_DEVICE_CONFIG_DEVICE_HARDWARE_VERSION_STRING=\"${CONFIG_DEVICE_HARDWARE_VERSION_STRING}\"" APPEND)
endif()
`

The same error occurs when removing the last 4 lines which i modified from your suggestion for adding the hardware version string
Do you have any idea what could cause this?
Where exactly do i have to place these lines?