espressif/esp-matter #223: ttyACM0 console on C3

jonsmirl, 2023-02-20:
I checked the light_switch app --- working.
I checked the blemesh_bridge --- not working.

set-target esp32c3
menuconfig 
jonsmirl@ares:~/aosp/esp-matter/examples/blemesh_bridge$ grep CONSOLE_USB sdkconfig
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y

build flash monitor
no console on blemesh

wqx6, 2023-02-20:
https://github.com/project-chip/connectedhomeip/blob/master/src/lib/shell/BUILD.gn The shell only supports chip_device_platform named esp32 but in the external platform the chip_device_platform has another name. Maybe you can edit the BUILD.gn and add  the external platform. Then you can use the shell.

jonsmirl, 2023-02-20:
This didn't fix it...
I changed the default case to the esp32 loop.
~~~
jonsmirl@ares:~/aosp/esp-matter/connectedhomeip/connectedhomeip$ git diff
diff --git a/src/lib/shell/BUILD.gn b/src/lib/shell/BUILD.gn
index d3fc4a4ee..64db1a4d7 100644
--- a/src/lib/shell/BUILD.gn
+++ b/src/lib/shell/BUILD.gn
@@ -110,7 +110,10 @@ static_library("shell") {
       "streamer_openiotsdk.cpp",
     ]
   } else {
-    sources += [ "MainLoopDefault.cpp" ]
+    sources += [
+      "MainLoopESP32.cpp",
+      "streamer_esp32.cpp",
+    ]
   }
 
   if (current_os == "mbed") {
~~~

jonsmirl, 2023-02-20:
That previous diff does help if you also turn on:
sdkconfig:CONFIG_ENABLE_CHIP_SHELL=y

then it is missing something to link to
~~~
/home/jonsmirl/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/chip/lib/libCHIP.a(shell_core.Engine.cpp.o): in function `chip::Shell::Engine::RegisterDefaultCommands()':
/home/jonsmirl/aosp/esp-matter/examples/blemesh_bridge/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/shell/Engine.cpp:109: undefined reference to `chip::Shell::RegisterWiFiCommands()'
~~~

jonsmirl, 2023-02-20:
I believe the platform name being used is 'external'

so this should have a case for external where it doesn't link anything
src/lib/shell/BUILD.gn b/src/lib/shell/BUILD.gn

And then the extern directory build.gn should add these to the build
      "MainLoopESP32.cpp",
      "streamer_esp32.cpp",

And then add in RegisterWifiCommands

jonsmirl, 2023-02-20:
I got this to function by hacking on the BUILD.gn file, but this is in no way a good fix. I just hacked on it until it worked for me. This just shows where the problems are.
~~~
jonsmirl@ares:~/aosp/esp-matter/connectedhomeip/connectedhomeip$ git diff
diff --git a/src/lib/shell/BUILD.gn b/src/lib/shell/BUILD.gn
index d3fc4a4ee..64db1a4d7 100644
--- a/src/lib/shell/BUILD.gn
+++ b/src/lib/shell/BUILD.gn
@@ -110,7 +110,10 @@ static_library("shell") {
       "streamer_openiotsdk.cpp",
     ]
   } else {
-    sources += [ "MainLoopDefault.cpp" ]
+    sources += [
+      "MainLoopESP32.cpp",
+      "streamer_esp32.cpp",
+    ]
   }
 
   if (current_os == "mbed") {
diff --git a/src/lib/shell/commands/BUILD.gn b/src/lib/shell/commands/BUILD.gn
index a635c3ac8..43e408775 100644
--- a/src/lib/shell/commands/BUILD.gn
+++ b/src/lib/shell/commands/BUILD.gn
@@ -35,6 +35,7 @@ source_set("commands") {
     ]
   }
 
+  sources += [ "WiFi.cpp" ]
   if (chip_enable_wifi) {
     sources += [ "WiFi.cpp" ]
   }
~~~

abdulrehmanee12, 2023-03-15:
PR [ESP32: Enable shell command for external platform feature #25648](https://github.com/project-chip/connectedhomeip/pull/25648) been raised in upstream repo to add shell command support for external platform feature

dhrishi, 2023-04-30:
Closing this. The above fix should be a part of connectedhomeip submodule that is now preset on the main branch

closed 2023-04-30 by dhrishi