espressif/esp-matter #225: Debugging in VS Code using WSL

stage-01, 2023-02-20:
I'm happy to report that I've successfully set up build and flash functionality through WSL on VS Code for my esp32c3 devkit, using the built-in USB to JTAG connection. However, I've hit a roadblock in getting debugging to work. After setting a breakpoint and pressing f5, VS Code prompts me to select a debugger, but when I choose ESP-IDF, nothing seems to happen.

stage-01, 2023-02-20:
I haven't got the build button working either. The only way I can currently build is via the terminal.

jonsmirl, 2023-02-20:
Try running openocd from the command line and make sure it works that way. You need to set the board and interface parameters when you run it.

brianignacio5, 2023-02-21:
Have you configured the ESP-IDF extension for Visual Studio Code to use WSL ? Please take a look at the [WSL tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/wsl.md)

Please the extension [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting)

stage-01, 2023-02-21:
Thanks for the quick responses. I've extensively followed the tutorial but even after doing everything the "right" way I still couldn't get openocd to run. It kept giving a permission error with the usb port. Because of the trouble programming in windows gave me I've decided to continue my development on Linux. So far everything in Linux seems to work. I've installed VS Code with esp-idf and esp-matter.

closed 2023-02-21 by stage-01