espressif/esp-matter #227: The coexistance of Matter WiFi Provisioning alongwith a custom one

HamzaHajeir, 2023-02-21:
Hi there,

I am in development phase of an IoT system and framework, that includes WiFi-provisioning over SoftAP, and WiFi management.

And we're considering to support matter in our under-development products.

However, I'm not sure whether I can rely on matter provisioning or not?
1. Can I send additional configuration/secret codes to the device at commisioning phase by using the matter provisioning?
2. If not (thus I should rely on my wifi provisioning), could that be a reason for failure of matter compliance test at future?

And regarding WiFi management:
1. Should I call matter APIs informing `WiFiConnected` and `WiFiDisconnected`? 
3. Would matter SDK call any of WiFi APIs internally?

Thanks in advance,
Hamza Hajeir

dhrishi, 2023-02-21:
@HamzaHajeir Matter setup, termed as commissioning, also involves connecting to the Wi-Fi network as one of the steps.

>Can I send additional configuration/secret codes to the device at commisioning phase by using the matter provisioning?

This assumes that the other end (like a phone app) commissioning the Matter device also belongs to you as both the sides need to understand your custom data. If the data is private like the secret code, you can use a custom cluster for your functionality in the Matter data model of the device. Once the commissioning is complete, you can re-use the existing secure session established by Matter to transfer your custom data.

>Would matter SDK call any of WiFi APIs internally?

Yes, as mentioned above, one part of Matter commissioning is getting the device on the IP network (Wi-Fi in this case) and is taken care of by the Matter SDK. Having said that, the SDK also provides an option to provide the Wi-Fi (IP) connectivity using your own mechanism and still commission the device on the IP network termed as on-network commissioning.

HamzaHajeir, 2023-02-21:
Thanks @dhrishi for your input.

I'd be reading more deeply in matter specs to understand more about the custom cluster (I'm not really sure what 'cluster' would indicate).

I think matter would normally manage the Wi-Fi events (Connect on Disconnect), is that correct? If so, is it possible to override this behaviour?

dhrishi, 2023-02-21:
>I'm not really sure what 'cluster' would indicate

Espressif has published a [blog series](https://blog.espressif.com/matter-38ccf1d60bcd) that provides details on different features of Matter for our customers and users. You can take a look. The no. 2 in the list can help with your question.

>I think matter would normally manage the Wi-Fi events (Connect on Disconnect), is that correct? 

That's right.

> If so, is it possible for us to take the actions of Wi-Fi and pass important events/information to matter?

Can you provide an example? I didn't quite understand this.

jonsmirl, 2023-02-21:
If your device is already established onto the wifi network before using Matter, then you can use Wifi to commission it instead of Bluetooth if your wish. One aspect of Matter Bluetooth commissioning is to also get the device onto the Wifi or Thread network. But if you are already on Wifi/Thread you don't need that part. This is how Ethernet devices work. They are automatically on the LAN because you plugged them in.

You are probably looking for these Matter events:
https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L32

HamzaHajeir, 2023-02-23:
Thanks for your input again @dhrishi.

> > If so, is it possible for us to take the actions of Wi-Fi and pass important events/information to matter?
> 
> Can you provide an example? I didn't quite understand this.

Here's an example of what I'm using now (similar concept), where my code listens for the events and manage the WiFi.
https://github.com/philbowles/h4plugins/blob/master/src/H4P_WiFi.cpp#L104

HamzaHajeir, 2023-02-23:
> If your device is already established onto the wifi network before using Matter, then you can use Wifi to commission it instead of Bluetooth if your wish. One aspect of Matter Bluetooth [...]

Thanks, I'll consider these events when integrating it to my code ofcourse, I'll read the blog (perhaps more docs) to have a better whole understanding of what will be going on.

dhrishi, 2023-02-24:
>where my code listens for the events and manage the WiFi.

You do not manage the Wi-Fi? But, based on the events given from the Wi-Fi, you take application specific actions in the project?

>So the actions would be handled by my code, and possibly inform esp-matter APIs of the occurance of such an event (as WiFi Disconnected).

Shouldn't it be the other way round? The esp-matter event handler will provide you with the Wi-Fi events and you take actions in your code?

Or, if you want better control on the Wi-Fi side and customize the implementation as per your need, then please refer [2.4.4 Advanced Setup](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#advanced-setup) section in the documentation.

HamzaHajeir, 2023-02-24:
> You do not manage the Wi-Fi? But, based on the events given from the Wi-Fi, you take application specific actions in the project?
>
>
> Shouldn't it be the other way round? The esp-matter event handler [...]

I don't really develop an application, but a framework of applications (has similarity in core with H4Plugins that I've mentioned a snippet of it).
So if it'd be our provisioning method, we'd for sure interact with the Wi-Fi APIs.

Also, I try to avoid `#if` `#ifdef` blocks to simplify management.
If it's mandatory then I'd be doing it, I'm now getting insights trying to fully understand, just before designing the integration solution.

> Or, if you want better control on the Wi-Fi side and customize the implementation as per your need, then please refer [2.4.4 Advanced Setup](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#advanced-setup) section in the documentation.
I've added it up to the reading list, thanks :).

closed 2023-02-25 by dhrishi

HamzaHajeir, 2023-02-26:
@dhrishi 

After some deep reading through the documentation and code, I think if my code would be managing WiFi, I'd just need calling `SetWiFiStationMode(kWiFiStationMode_ApplicationControlled)` and `SetWiFiAPMode(kWiFiAPMode_OnDemand)` * on my setup code, without the need to make any advance setup, correct?

HamzaHajeir, 2024-06-26:
Hi all @dhrishi @jonsmirl, searching for important issue regarding Matter and found this issue :).

To your knowledge, is having custom WiFi provisioning over WiFiAP (as described earlier) would result in the product not compliant to Matter specificiations? Where can I have a specific answer to this?

With thanks,

jonsmirl, 2024-06-26:
You want this section, 5.2.2.4. Commissioner Discovery, from an On-Network Device
From the Matter Core specification

HamzaHajeir, 2024-07-01:
> You want this section, 5.2.2.4. Commissioner Discovery, from an On-Network Device From the Matter Core specification

Thanks @jonsmirl, I've read that, but it's a different case than the intended, as it is meant for UI devices discovering commissioners by help of ther user. Our case is the inverse: The commissioner discovers On-Network WiFi devices to commision them.

There's a warning on top of the 5.2.1 module says "The list of user journeys provided here is not meant to be exhaustive; there
may be other journeys not listed here which can be realized using Matter.", and while my exact case was not mentioned in User Journeys 5.2.2.1 to 5.2.2.4, it's left open to an answer.

Is there a community that I can engage with?

jonsmirl, 2024-07-01:
5.2.2.3 is what Amazon uses in their FireTV devices. You can run the CHIP examples for tv-app and tv-casting-app to play with it.

jonsmirl, 2024-07-02:
I would try to stick with 5.7.1 or 5.7.2. When you venture off the common path you will run into a lot of poorly tested code.

HamzaHajeir, 2024-07-02:
> I would try to stick with 5.7.1 or 5.7.2. When you venture off the common path you will run into a lot of poorly tested code.

I'll review if we can rely on these basic ones, but required either our system not requiring custom data provisioning, or to consider the Matter commissioning process begins after our WiFi provisioning process.