espressif/esp-matter #228: commissioning window

stefano055415, 2023-02-21:
is it possible to open the commissioning window with a physical button?

15 minutes after starting up the device, the commissioning window closes and I have to restart the device to be able to reopen it.

could i use a physical button to restart it?

and is it possible not to open the commissioning window at boot?

dhrishi, 2023-02-21:
@stefano055415 

>is it possible to open the commissioning window with a physical button? could i use a physical button to restart it?

Yes, it is possible. We have an API to open the commissioning window. See [this](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L75) for reference. You can pass the timeout you require (min: 3 minutes, max: 15 minutes as per the spec) and the transport on which the commissioning is being advertised i.e. BLE or IP.

>and is it possible not to open the commissioning window at boot?

Yes, it is possible. Please check `5.7.2. User-Intent Commissioning Flow` in the specification for additional details.

stefano055415, 2023-02-22:
@dhrishi 

For the physical button that reactivates the commissioning window it's ok thanks

For the User-Intent Commissioning Flow I read the documentation and I understand how it works, but I don't understand how to activate it

I also found this [issue](https://github.com/project-chip/connectedhomeip/issues/25210) but I don't understand exactly how to create the qrcode and how to disable it on boot

dhrishi, 2023-02-22:
>I also found this https://github.com/project-chip/connectedhomeip/issues/25210 but I don't understand exactly how to create the qrcode

@stefano055415 You can use the payload option from the chip-tool to create the QR code. Below, I have used the default QR code and updated the commissioning flow to user-intent (value 1) instead of default standard (value 0)

```
./chip-tool payload generate-qrcode --existing-payload MT:Y.K9042C00KA0648G00 --commissioning-mode 1
[1677053365.166765][919220:919220] CHIP:TOO: QR Code: MT:Y.K900ID00KA0648G00
```
Generated QR code: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT:Y.K900ID00KA0648G00

> and how to disable it on boot

So, currently, the upstream SDK support for this is not entirely complete. The option right now is to disable the option CHIP_DEVICE_CONFIG_ENABLE_PAIRING_AUTOSTART [from the code](https://github.com/espressif/connectedhomeip/blob/v1.0.0.2/src/include/platform/CHIPDeviceConfig.h#L461) so that it does not open commissioning window on bootup. And then open the commissioning window (using the above API) on the desired action or event in your application. The way about how this is enabled can be printed on the device manual

The upstream issue talks about how the user-intent functionality can be exposed in the right way

stefano055415, 2023-02-22:
@dhrishi 
ok perfect thanks

closed 2023-02-22 by stefano055415