espressif/esp-matter #230: Running Matter along with our IOT Cloud solution

aggaddam, 2023-02-22:
Hi, 

We have been using ESP8266 and EP32 series chips in our products for our own IOT cloud solution. Our existing solution uses soft AP mode to bring/onboard/set up the devices into the Wi-Fi network. We are now upgrading our products to support matter with ESP32-C3 Hardware. 

We are looking for approaches to enable the matter in our products for local control/interoperability and at the same time we want to provide end users to control the device with our IOT cloud solutions/mobile app remotely over the internet.  I am still exploring the matter's protocol and specifications. Meanwhile, I want to check, Is there a way we can run both matter and also in parallel connected to our cloud? 

Regards,
Ashok

dhrishi, 2023-02-22:
@aggaddam 

>Is there a way we can run both matter and also in parallel connected to our cloud?

Yes, it is possible. Once your Matter phone app performs the setup (termed as commissioning) of the Matter device, it gets Wi-Fi (IP) connectivity too. The device can then connect to your cloud in the firmware. And your phone app can perform any exchange for the cloud communication with the device. You can also re-use the secure session established through Matter for this communication. With this option, you can totally give away your existing soft AP based solution to onboard devices to the Wi-Fi network.

Note that, Matter is a local network protocol and the spec does not talk anything related to remote control. It can be taken care of by the implementers, as per their requirements.

aggaddam, 2023-02-22:
@dhrishi Thanks for your fast response.

In order to connect to our cloud IOT solution, we need to pass information about cloud endpoint details and other key details. At which step of the matter commissioning process, this information needs to be passed ?

I also read from a few blogs that espressif already implemented a similar kind of implementation with both ESP-Matter SDK and ESP rainmaker agent (Cloud IOT solution) running together in a matter-certified product. How does this implementation work with matter? Does espressif add custom information during the matter commissioning process to connect to the rainmaker cloud?

dhrishi, 2023-02-22:
@aggaddam 

>At which step of the matter commissioning process, this information needs to be passed ?

As mentioned before, this is outside the scope of Matter. Ideally, it can be done once the commissioning is complete so that you have Wi-Fi connectivity and also can re-use the Matter secure session that is established.

>Does espressif add custom information during the matter commissioning process to connect to the rainmaker cloud

That's right. We use a custom cluster for RainMaker user node association. That is done post commissioning

aggaddam, 2023-02-23:
@dhrishi  Appreciated your response!  Could you please direct me to an example that implements ESP matter and rainmaker integration?

dhrishi, 2023-02-23:
@aggaddam The example isn't out yet. The component code having the RainMaker custom cluster is [here](https://github.com/espressif/esp-matter/blob/main/components/esp_matter_rainmaker/esp_matter_rainmaker.cpp#L216)

You can track issue #149 as it will be updated once the example is out.

I will close this issue as the original question is answered.

closed 2023-02-23 by dhrishi