espressif/esp-matter #231: Creating the bridge device failed

maenkai, 2023-02-23:
I updated the repository to bd9fc0d97459a9 and the following exception occurred when I used to create a bridge device

esp32:
W (00:00:16.421) esp_matter_core: Server Cluster 0x001d on endpoint 0x0008 already exists. Not creating again.
W (00:00:16.423) esp_matter_core: Attribute 0xfffd on cluster 0x001d already exists. Not creating again.
W (00:00:16.435) esp_matter_core: Attribute 0x0000 on cluster 0x001d already exists. Not creating again.
W (00:00:16.447) esp_matter_core: Attribute 0x0001 on cluster 0x001d already exists. Not creating again.
W (00:00:16.459) esp_matter_core: Attribute 0x0002 on cluster 0x001d already exists. Not creating again.
W (00:00:16.470) esp_matter_core: Attribute 0x0003 on cluster 0x001d already exists. Not creating again.
W (00:00:16.482) esp_matter_core: Attribute 0xfffc on cluster 0x001d already exists. Not creating again.
I (00:00:16.504) esp_matter_bridge: Cluster plugin init for the new added endpoint
.....plugin_init_callback_endpoint........... 168 
.....get_plugin_server_init_callback........... 1674 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400ec4d8  PS      : 0x00060d30  A0      : 0x801177b6  A1      : 0x3ffe6f20  
0x400ec4d8: esp_matter::cluster::get_plugin_server_init_callback(unsigned int*) at /home/mek/esp/esp-matter/components/esp_matter/esp_matter_core.cpp:1676

A2      : 0x00020008  A3      : 0x3ffe7310  A4      : 0x000000a8  A5      : 0x3ffc37a0  
A6      : 0x3f41e512  A7      : 0x00000001  A8      : 0x800ec4d8  A9      : 0x3ffe6ed0  
A10     : 0x00000037  A11     : 0x3ffe7310  A12     : 0x0000068a  A13     : 0x3ffe6df0  
A14     : 0x00000034  A15     : 0x3ffe6df0  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00020008  LBEG    : 0x40092259  LEND    : 0x40092269  LCOUNT  : 0xfffffff7

maenkai, 2023-02-23:
I solved this problem by adding scope to esp_matter_bridge.cpp

static esp_err_t plugin_init_callback_endpoint(endpoint_t *endpoint)
{
    if (!endpoint) {
        ESP_LOGE(TAG, "endpoint cannot be NULL");
        return ESP_ERR_INVALID_ARG;
    }
    ESP_LOGI(TAG, "Cluster plugin init for the new added endpoint");
    cluster_t *cluster = cluster::get_first(endpoint);
    while (cluster) {
        /* Plugin server init callback */
        cluster::plugin_server_init_callback_t plugin_server_init_callback =
            cluster::get_plugin_server_init_callback(cluster);

        if (plugin_server_init_callback) {
            plugin_server_init_callback();
        }
        cluster::plugin_client_init_callback_t plugin_client_init_callback =
            cluster::get_plugin_client_init_callback(cluster);
        if (plugin_client_init_callback) {
            plugin_client_init_callback();
        }
        cluster = cluster::get_next(cluster);
    }
    return ESP_OK;
}

wqx6, 2023-02-23:
Thanks for your report of this issue. We will fix it.

dhrishi, 2023-02-23:
Thanks @maenkai. This has been fixed with https://github.com/espressif/esp-matter/commit/38a893b1acc8d81ec8c5c18bc747a2de3072a781

closed 2023-02-23 by dhrishi