espressif/esp-matter #232: Details required on ESP-MATTER SDK/IDF versions for proof of concept and development

aggaddam, 2023-02-23:
I am looking for the latest ESP-MATTER SDK for my project which is under the proof of concept phase. Which version of ESP-IDF and ESP-MATTER SDK can I use it for my development purpose? When is the official release planned for ESP-IDF with the matter as an integrated component inside ESP-IDF? Also, is there any documentation available on how to use the API's of ESP-MATTER SDK?

dhrishi, 2023-02-23:
@aggaddam Please follow the documentation [link](https://github.com/espressif/esp-matter#documentation) present at the top-level README.md. This specifically should help you: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html

You can also quickly create a PoC using [pre-built images](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml) ESP Launchpad, again present in the top-level README.md

dhrishi, 2023-02-24:
Closing as the question is answered

closed 2023-02-24 by dhrishi

aggaddam, 2023-02-24:
Currently, there is very minimal documentation related to APIs of ESP-MATTER. When is the next official release planned for ESP-IDF with the matter?

dhrishi, 2023-02-24:
>Currently, there is very minimal documentation related to APIs of ESP-MATTER.

Currently, we have the [API reference documentation](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/api-reference/index.html). Which specific APIs are you looking into?

>When is the next official release planned for ESP-IDF with the matter?

ESP-IDF does not do official releases for Matter as it is the generic core SDK. Matter, sort of a specific project/implementation, sits on top of ESP-IDF and uses the IDF functionality as the base (like many other ESP projects do)