espressif/esp-matter #236: Attribute initialization in the ColorControl cluster.

jonsmirl, 2023-02-24:
I made a color control cluster like this:
~~~
    color_control::create(epLight, &ext_config.color_control, CLUSTER_FLAG_SERVER | CLUSTER_FLAG_CLIENT,
                        color_control::feature::color_temperature::get_id() | color_control::feature::xy::get_id());
~~~
It works and I can set the colorXY. The problem is that the CurrentX and CurrentY attributes are not getting initialized when the cluster is first created.
~~~
I (1312) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x4010 is 250 **********
I (1312) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x0007 is 250 **********
I (1312) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x400B is 0 **********
I (1312) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x400C is 65279 **********
I (1312) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x0007 is 250 **********
I (1312) ep: app_driver_attribute_update EP 1 CL 768 AT 7
I (1312) ep: Temperature1 255 255 255
I (1312) ep: Temperature2 255 0 0
I (1312) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x0008 is 2 **********
I (1322) ep: app_driver_attribute_update EP 1 CL 768 AT 8
I (1322) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x4001 is 2 **********
I (1322) ep: app_driver_attribute_update EP 1 CL 768 AT 16385
I (1322) esp_matter_core: Dynamic endpoint 1 added
~~~
I'm not getting the attribute init call for attributes 3 and 4 on the ColorControl cluster. 0xf and 0x400A, 0x400d also seem to be missing, but I don't use those. I'm on v1.0.0.2 CHIP.

shripad621git, 2023-02-28:
I have made the Color Control Cluster as specified above in esp_matter_endpoint.cpp in color_temperature_light.
Also, I have read the CurrentX and CurrentY attributes using chip-tool and I found their values consistent with the spec.

[image]

[image]

I performed a wildcard read on all the attributes on the Color Control Cluster using the command
``` chip-tool any read-all 0x0300,0x0300,0x0300 0xFFFFFFFF,0xFFFFFFFF 0 0x7283 1,1,1 ```
and I was able to retrieve the attributes 3 and 4 .
Also, the attributes 0x400a and 0x400d were retrieved .

[image]

Can you please give it one more try with the steps mentioned above?

jonsmirl, 2023-02-28:
There is no problem reading the clusters, I can also read them.  The issue is when the cluster is first created you get attribute update messages containing the default values for the attributes. I'm getting those initialization message for all of the attributes except ColorXY. Look earlier in your log back when the cluster was first created.

jonsmirl, 2023-02-28:
I don't understand the pattern, why do I get an initialization message for: 0x400c ColorTempPhysical­-MaxMireds default 0xfeff
I (1328) esp_matter_attribute: ********** Endpoint 0x0002's Cluster 0x0300's Attribute 0x400C is 65279 **********

But then I don't get one for  0x0003 CurrentX  default 0x616b?

It is not consistent, some attributes I get messages for and some I don't.

jonsmirl, 2023-02-28:
I'm starting to get a handle on this, these messages are confusing:
I (1312) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x4010 is 250 **********
I (1312) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x0007 is 250 **********
I (1312) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x400B is 0 **********
I (1312) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x400C is 65279 **********
I (1312) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0300's Attribute 0x0007 is 250 **********

Those messages don't indicate whether it is a read or write. I have been assuming you'd only print on writes, but that's not the case. Reads are getting printed too, but there is no way to tell from the message whether it is a read or a write. I can identify read/write now by setting a breakpoint and looking up the stack.

Now I am able to see that CHIP is not sending messages when the attributes are initially set. Those entries that LOOK like messages setting the attributes are in fact CHIP internally reading the default value from the attribute.

Fix on the Espressif side: add a R/W to those messages to indicate Read or Write.
Fix on my side: I need to add code like this
https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_driver.cpp#L100

shripad621git, 2023-03-28:
Will update soon with the fix for 'R 'or 'W'.

commit 14697712 (2023-04-06): Indicating Read or Write on Attribute

Fixes #236.
For better debuggig added 'R' or 'W' on the prints for attributes.