espressif/esp-matter #237: Binding logic

jonsmirl, 2023-02-25:
Currently the code does this
~~~
        client::command_handle_t cmd_handle;
        uint16_t local_endpoint_id = strtol((const char *)&argv[1][2], NULL, 16);
        cmd_handle.cluster_id = strtol((const char *)&argv[2][2], NULL, 16);
        cmd_handle.command_id = strtol((const char *)&argv[3][2], NULL, 16);
        cmd_handle.is_group = false;

        if (argc > 4) {
           console_buffer[0] = argc - 4;
           for (int i = 0; i < (argc - 4); i++) {
                if ((argv[4+i][0] != '0') || (argv[4+i][1] != 'x') || (strlen((const char*)&argv[4+i][2]) > 10)) {
                    ESP_LOGE(TAG, "Incorrect arguments. Check help for more details.");
                    return ESP_ERR_INVALID_ARG;
                }
                strcpy((console_buffer + 1 + 10*i), &argv[4+i][2]);
           }

           cmd_handle.command_data = console_buffer;
        }

        client::cluster_update(local_endpoint_id, &cmd_handle);
~~~
to this
~~~
    command_handle_t *context = chip::Platform::New<command_handle_t>(cmd_handle);
    if (!context) {
        ESP_LOGE(TAG, "failed to alloc memory for the command handle");
        return ESP_ERR_NO_MEM;
    }
    if (CHIP_NO_ERROR !=
        chip::BindingManager::GetInstance().NotifyBoundClusterChanged(local_endpoint_id, cmd_handle->cluster_id,
                                                                      static_cast<void *>(context))) {
        chip::Platform::Delete(context);
        ESP_LOGE(TAG, "failed to notify the bound cluster changed");
        return ESP_FAIL;
    }
~~~
to this
~~~
void app_driver_client_command_callback(client::peer_device_t *peer_device, client::command_handle_t *cmd_handle,
                                         void *priv_data)
                level_control::command::send_move_to_level(peer_device, cmd_handle->endpoint_id, strtol((const char *)(cmd_handle->command_data) + 1, NULL, 16),
                        strtol((const char *)(cmd_handle->command_data) + 11, NULL, 16), strtol((const char *)(cmd_handle->command_data) + 21, NULL, 16),
                        strtol((const char *)(cmd_handle->command_data) + 31, NULL, 16));
~~~
and then to this
~~~
    LevelControl::Commands::MoveToLevel::Type command_data;
    command_data.level = level;
    command_data.transitionTime.SetNonNull(transition_time);
    command_data.optionsMask.SetRaw(option_mask);
    command_data.optionsOverride.SetRaw(option_override);

    chip::Controller::LevelControlCluster cluster(*remote_device->GetExchangeManager(),
                                                  remote_device->GetSecureSession().Value(), remote_endpoint_id);
    cluster.InvokeCommand(command_data, NULL, send_command_success_callback, send_command_failure_callback);
    return ESP_OK;
~~~
Could this work instead? Don't pass strings around, instead make the command_data structure much earlier. The long code logic from app_driver_client_command_callback() moves over to app_driver_client_console_handler()
First step becomes:
~~~
    LevelControl::Commands::MoveToLevel::Type command_data;
    command_data.level = level;
    command_data.transitionTime.SetNonNull(transition_time);
    command_data.optionsMask.SetRaw(option_mask);
    command_data.optionsOverride.SetRaw(option_override);
 client::cluster_update(local_endpoint_id, &cmd_handle);
~~~
Now the last two steps disappear and just turn into:
~~~
   cluster.InvokeCommand(command_data, NULL, send_command_success_callback, send_command_failure_callback);
~~~
Why do I want this? Because I already have the command specific data structure. To use this API I have to turn it into strings, give to the API and then the API turns it back into the same structure.

jonsmirl, 2023-02-26:
Example of calling the binding API.  And how do I know how to set the is_group flag? 
~~~
void UI_Switch::long_pressed(int direction, int position) {
    ESP_LOGI(TAG, "Long button pressed %x", position);
    uint16_t epGenericID = endpoint::get_id(epGeneric);
    uint16_t epLightID = endpoint::get_id(epLight);
    long_press_active = true;
   
    LevelControl::Commands::MoveWithOnOff::DecodableType commandData;
    commandData.moveMode = (position == UP ? EMBER_ZCL_MOVE_MODE_UP : EMBER_ZCL_MOVE_MODE_DOWN);
    commandData.rate.SetNonNull(250);
    app::ConcreteCommandPath commandPath = {epLightID, LevelControl::Id, LevelControl::Commands::MoveWithOnOff::Id};
    app::CommandHandler commandObj(nullptr);

    // Press moves Position from 0 (idle) to 1 (press)
    lock::chip_stack_lock(portMAX_DELAY);
    emberAfLevelControlClusterMoveWithOnOffCallback(&commandObj, commandPath, commandData);
    chip::app::Clusters::Switch::Attributes::CurrentPosition::Set(epGenericID, position);
    // LongPress event takes newPosition as event data
    switch_cluster::event::send_long_press(epGenericID, position);
    lock::chip_stack_unlock();

    // notify bound devices
    client::command_handle_t cmd_handle;
    cmd_handle.endpoint_id = epLightID;
    cmd_handle.cluster_id = LevelControl::Id;
    cmd_handle.command_id = LevelControl::Commands::MoveWithOnOff::Id;
    cmd_handle.is_group = false;
    snprintf(console_buffer, sizeof(console_buffer), "%-10d%-10d", commandData.moveMode, commandData.rate.Value());
    cmd_handle.command_data = console_buffer;
    client::cluster_update(epLightID, &cmd_handle);
}
~~~

jonsmirl, 2023-02-26:
It should be possible to write this
~~~
void UI_Switch::long_pressed(int direction, int position) {
    uint16_t epGenericID = endpoint::get_id(epGeneric);
    uint16_t epLightID = endpoint::get_id(epLight);
   
    level_control::command::send_move_with_onoff(epLight, moveMode, rate);
    switch_cluster::event::send_long_press(epGenericID, position);
}
~~~

Send_move_with_onoff()  would build 'commandPath, commandData' then calls into a common routine which internally checks the EP to see if there is sever cluster, if so sends the command to it. Then if bound unicast and group and if they exist, sends the command to them too. Note that if you send 'commandPath, commandData'  into app_driver_client_command_callback() it completely disappears and turns into a simple invoke call. No need for that giant switch in user code to implementing it.

send_long_press() would both send the event and update the attribute.

jonsmirl, 2023-02-27:
Amazon uses a different approach and directly read the bindings table.
https://github.com/project-chip/connectedhomeip/blob/master/examples/tv-casting-app/tv-casting-common/src/CastingServer.cpp#L186

jonsmirl, 2023-02-28:
I noticed a problem using with using bridges and bindings. 

Previously I said this: "Send_move_with_onoff() would build 'commandPath, commandData' and check the EP to see if there is a sever cluster, if so send the command to it using emberAf...Callback()" Which calls this:
~~~
    // run the command locally
    LevelControl::Commands::MoveWithOnOff::DecodableType commandData;
    commandData.moveMode = (position == UP ? EMBER_ZCL_MOVE_MODE_UP : EMBER_ZCL_MOVE_MODE_DOWN);
    commandData.rate.SetNonNull(250);
    app::ConcreteCommandPath commandPath = {epLightID, LevelControl::Id, LevelControl::Commands::MoveWithOnOff::Id};
    app::CommandHandler commandObj(nullptr);

    lock::chip_stack_lock(portMAX_DELAY);
    emberAfLevelControlClusterMoveWithOnOffCallback(&commandObj, commandPath, commandData);
    lock::chip_stack_unlock();
~~~

But then I realized there are a lot more local cases than just a fixed server cluster. There are also all of the bridged devices.

Fortunately there is a very simple solution to this, implement loopback bindings instead of simply searching for server clusters.  For a loopback binding I'd add a binding like this:
~~~
 binding write binding '[{"node":0x7283, "endpoint":1, "cluster":6}]' 0x7283 0x1
~~~
and then when you are processing the bindings, check for source==dest node ID. Don't send those off on the network, instead call the emberAf...Callback() code. Now I can do this to bind to the BLE lights (38,39,40) in my bridge. 

Note that I have a light switch which contains a BLE bridge. The physical switch implements DimmerSwitch client. The BLE bridge is on the same device. So the DimmerSwitch client needs to bind to those bridged BLE bulbs and they are all on the same node.
~~~
 binding write binding '[{"node":0x7283, "endpoint":38, "cluster":6}]' 0x7283 0x1
 binding write binding '[{"node":0x7283, "endpoint":39, "cluster":6}]' 0x7283 0x1
 binding write binding '[{"node":0x7283, "endpoint":40, "cluster":6}]' 0x7283 0x1
~~~
And as the spec says, it you leave out cluster ID it applies to all client clusters.
~~~
 binding write binding '[{"node":0x7283, "endpoint":38}]' 0x7283 0x1
 binding write binding '[{"node":0x7283, "endpoint":39}]' 0x7283 0x1
 binding write binding '[{"node":0x7283, "endpoint":40}]' 0x7283 0x1
~~~
That would loop back OnOff, LevelControl and ColorControl in my case.

wqx6, 2023-03-01:
Thanks for your suggestion. In our logic, `client::cluster_update()` is only used to binding notify. We also wants to implement a clusters client which can invoke the command without a binding cluster. The logic `turn the command specific data structure into strings, and then turns it back into the same structure` should be changed. we will fixed it later.

wqx6, 2023-03-01:
> Do I need to call this twice? Once for is_group true and false? My device has no clue whether it has been bound unicast or group.

I think the button pressing should have a single function, in the example, it is toggling the unicast-bound light. If you want to notify the group-bound device to toggle, you should use the `bound` console command.

jonsmirl, 2023-03-01:
I don't believe you need to use the switch logic anywhere except in the command line implementation which gets compiled out during production.

The trick is always use CHIP command arg union instead of your own. LIke I did here
~~~
    // run the command locally
    LevelControl::Commands::MoveWithOnOff::DecodableType commandData;
    commandData.moveMode = (position == UP ? EMBER_ZCL_MOVE_MODE_UP : EMBER_ZCL_MOVE_MODE_DOWN);
    commandData.rate.SetNonNull(250);
    app::ConcreteCommandPath commandPath = {epLightID, LevelControl::Id, LevelControl::Commands::MoveWithOnOff::Id};
    app::CommandHandler commandObj(nullptr);
~~~

If you initially build the CHIP structure you can just pass it intact into invoke later in the process.

So this:
~~~
 level_control::command::send_move_with_onoff(epLight, moveMode, rate);
~~~

Is implemented like this:
~~~
 level_control::command::send_move_with_onoff(epLight, moveMode, rate) {

    LevelControl::Commands::MoveWithOnOff::DecodableType commandData;
    commandData.moveMode = (position == UP ? EMBER_ZCL_MOVE_MODE_UP : EMBER_ZCL_MOVE_MODE_DOWN);
    commandData.rate.SetNonNull(250);
    app::ConcreteCommandPath commandPath = {epLightID, LevelControl::Id, LevelControl::Commands::MoveWithOnOff::Id};
    app::CommandHandler commandObj(nullptr);

   make a struct...

  struct {
      commandData
      commandPath
      commandObect
      localCall 
  } myData;

  myData.localCall = &emberAfLevelControlClusterMoveWithOnOffCallback

   process_binding( myData);
}
~~~
and then process_binding looks like this. This is only one process_binding, it works for every command.
~~~
process_binding( myData)) {
    for (loop over the relevant bindings) {
          if (unicast)
               invoke(myData.commandData, myData.commandPath, myData.commandObect)
         if (group)
               invoke-group(myData.commandData, myData.commandPath, myData.commandObect)
         if (local / loopback)
               myData.localCall(myData.commandData, myData.commandPath, myData.commandObect)
   }
}
~~~

~~~

wqx6, 2023-03-02:
What type should the `commandData` should be? If it is a normal type such as `void *`, you need do the `switch(cluster_id){switch(command_id){}}` again in `invoke()` or `invoke-group()` since you have done this before you call the `level_control::command::send_move_with_onoff()`. Then why don't we just save the console inputs and do the `switch(cluster_id){switch(command_id){}}` after we establish the session and invoke the command?

BTW, the `LevelControl::Commands::MoveWithOnOff::DecodableType` is different from the `LevelControl::Commands::MoveWithOnOff::Type`. If you want to invoke the command, you should convert the `DecodableType` to `Type`. That will be a more complicated logic.

And in your process, when is the session establishment for the unicast command-sending? There are two conditions when we call the command-invoke functions like `level_control::command::send_move_with_onoff()`.  One is the device invokes the command as a simple client, and we will call the `client::connect()` and call the command-invoke functions after the session is established. Another is the device send the command as a binding device, we will use the `client::cluster_update()` to call the `NotifyBoundDevice`.  And the `loop over the relevant bindings` will be done in the `NotifyBoundDevice`.  The session will be also establish in `NotifyBoundDevice`, and then we use the registered `BoundDeviceChangedCallback` to do the command-invoke.

Also there is no local/loopback binding logic in the upstream binding codes. And I didn't see the description about the local/loopback binding in SPEC. The CSA does not define the logic for the self-binding, maybe you can raise an issue or a PR in the SPEC repo. It might help on your issue.

jonsmirl, 2023-03-02:
I'll reply more in the morning..  but I see that I missed the bit between Type and DecodableType. Just make both of them, don't try to convert.  I will talk to CHIP people about adjusting zap to make these use a shared structure.
~~~
 level_control::command::send_move_with_onoff(epLight, moveMode, rate) {

    LevelControl::Commands::MoveWithOnOff::DecodableType commandDataDT;
    commandDataDT.moveMode = (position == UP ? EMBER_ZCL_MOVE_MODE_UP : EMBER_ZCL_MOVE_MODE_DOWN);
    commandData.rateDT.SetNonNull(250);
    LevelControl::Commands::MoveWithOnOff::Type commandDataT;
    commandDataT.moveMode = (position == UP ? EMBER_ZCL_MOVE_MODE_UP : EMBER_ZCL_MOVE_MODE_DOWN);
    commandData.rateT.SetNonNull(250);
    app::ConcreteCommandPath commandPath = {epLightID, LevelControl::Id, LevelControl::Commands::MoveWithOnOff::Id};
    app::CommandHandler commandObj(nullptr);

   make a struct...

  struct {
      commandDataDT
      commandDataT
      commandPath
      commandObect
      localCall 
  } myData;

  myData.localCall = &emberAfLevelControlClusterMoveWithOnOffCallback

   process_binding( myData);
}
~~~
CHIP should have generated this as a shared structure and not made two copies. Then you could cast Type to DecodableType if you knew you weren't going use Decode().
~~~
namespace MoveWithOnOff {
enum class Fields
{
    kMoveMode        = 0,
    kRate            = 1,
    kOptionsMask     = 2,
    kOptionsOverride = 3,
};

struct Type
{
public:
    // Use GetCommandId instead of commandId directly to avoid naming conflict with CommandIdentification in ExecutionOfACommand
    static constexpr CommandId GetCommandId() { return Commands::MoveWithOnOff::Id; }
    static constexpr ClusterId GetClusterId() { return Clusters::LevelControl::Id; }

    MoveMode moveMode = static_cast<MoveMode>(0);
    DataModel::Nullable<uint8_t> rate;
    chip::BitMask<LevelControlOptions> optionsMask     = static_cast<chip::BitMask<LevelControlOptions>>(0);
    chip::BitMask<LevelControlOptions> optionsOverride = static_cast<chip::BitMask<LevelControlOptions>>(0);

    CHIP_ERROR Encode(TLV::TLVWriter & writer, TLV::Tag tag) const;

    using ResponseType = DataModel::NullObjectType;

    static constexpr bool MustUseTimedInvoke() { return false; }
};

struct DecodableType
{
public:
    static constexpr CommandId GetCommandId() { return Commands::MoveWithOnOff::Id; }
    static constexpr ClusterId GetClusterId() { return Clusters::LevelControl::Id; }

    MoveMode moveMode = static_cast<MoveMode>(0);
    DataModel::Nullable<uint8_t> rate;
    chip::BitMask<LevelControlOptions> optionsMask     = static_cast<chip::BitMask<LevelControlOptions>>(0);
    chip::BitMask<LevelControlOptions> optionsOverride = static_cast<chip::BitMask<LevelControlOptions>>(0);
    CHIP_ERROR Decode(TLV::TLVReader & reader);
};
}; // namespace MoveWithOnOff
~~~

jonsmirl, 2023-03-02:
Loopback is going to be officially spelled out in the next CHIP major release. However, it is not prevented today, the spec simply says nothing about it.  An example of why I have to have loopback. I'm implementing light switches which contain the BLE Mesh bridge. So my light switch has a Dimmer_Switch Endpoint, and a LIght Endpoint (the bridged BLE blub). Now I need to bind the Dimmer_Switch to the Light. But how? they are both on the same NodeID.

The needed syntax already works. 
~~~
binding write binding '["node":0x7283, "endpoint":28, "cluster":6}]' 0x7283 0x1
~~~
You can make that binding right and CHIP will accept it.  But if you try and use it you will get an MDNS failure because the CHIP library doesn't know how to resolve its own nodeID.  But that MDNS failure doesn't matter if the binding code never tries to send the command onto the network. That is what this code is doing, local means source == destination. In that case skip the network and directly call into the emberAf...Callback() function.
~~~
process_binding( myData)) {
    for (loop over the relevant bindings) {
          // binding contains remoteID, groupID, localEP
          if (unicast)
               invoke(remoteID, myData.commandData, myData.commandPath, myData.commandObect)
         if (group)
               invoke-group(groupID, myData.commandData, myData.commandPath, myData.commandObect)
         if (local / loopback) {
               copy command path, set in the local EP
               myData.localCall(localCluster, myData.commandData, commandPathCopy, myData.commandObect)
        }
   }
}
~~~

jonsmirl, 2023-03-03:
I've been implementing this locally and since Type and DecodableType are not the same, loopback bindings don't make sense any more. There is no point in doing this:
~~~
 myData.localCall = &emberAfLevelControlClusterMoveWithOnOffCallback
 process_binding( myData);
~~~
The local binding is pointless and I should just do what I was already doing:
~~~
    // run the command locally
    LevelControl::Commands::MoveWithOnOff::DecodableType commandData;
    commandData.moveMode = (position == UP ? EMBER_ZCL_MOVE_MODE_UP : EMBER_ZCL_MOVE_MODE_DOWN);
    commandData.rate.SetNonNull(250);
    app::ConcreteCommandPath commandPath = {epLightID, LevelControl::Id, LevelControl::Commands::MoveWithOnOff::Id};
    app::CommandHandler commandObj(nullptr);

    lock::chip_stack_lock(portMAX_DELAY);
    emberAfLevelControlClusterMoveWithOnOffCallback(&commandObj, commandPath, commandData);
    lock::chip_stack_unlock();
~~~

jonsmirl, 2023-03-03:
Can a structure like this work? this is general idea, some of the details missing. Then this sendBound() function would get added to esp-matter.
~~~
static sendBound(uint16_t endpoint, CLusterID clusterID, RequestObjectT & requestCommandData)
{
    for (const auto & binding : BindingTable::GetInstance())
    {
       if (binding.local == endpoint && (!bindin.clusterId.HasValue() || binding.clusterId.Value() == clusterID))
        {
            if (binding.type == EMBER_UNICAST_BINDING)
            {
                chip::Controller::LevelControlCluster cluster(*binding.remote_device->GetExchangeManager(),
                                                            binding.remote_device->GetSecureSession().Value(), binding.remote_endpoint_id);
                cluster.InvokeCommand(requestCommandData, NULL, send_command_success_callback, send_command_failure_callback);
            }
            else if (binding.type == EMBER_MULTICAST_BINDING)
            {
                chip::Messaging::ExchangeManager &exchange_mgr = chip::Server::GetInstance().GetExchangeManager();
                chip::Controller::InvokeGroupCommandRequest(&exchange_mgr, binding.fabric_index, binding.group_id, requestCommandData);
            }
        }

    }
}

void UI_Switch::long_pressed(int direction, int position)
{
    ESP_LOGI(TAG, "Long button pressed %x", position);
    uint16_t epGenericID = endpoint::get_id(epGeneric);
    uint16_t epLightID = endpoint::get_id(epLight);
    long_press_active = true;

    LevelControl::Commands::MoveWithOnOff::DecodableType commandDataDT;
    commandDataDT.moveMode = (position == UP ? EMBER_ZCL_MOVE_MODE_UP : EMBER_ZCL_MOVE_MODE_DOWN);
    commandDataDT.rate.SetNonNull(250);
    app::ConcreteCommandPath commandPath = {epLightID, LevelControl::Id, LevelControl::Commands::MoveWithOnOff::Id};
    app::CommandHandler commandObj(nullptr);

    // Press moves Position from 0 (idle) to 1 (press)
    lock::chip_stack_lock(portMAX_DELAY);
    emberAfLevelControlClusterMoveWithOnOffCallback(&commandObj, commandPath, commandDataDT);
    chip::app::Clusters::Switch::Attributes::CurrentPosition::Set(epGenericID, position);
    // LongPress event takes newPosition as event data
    switch_cluster::event::send_long_press(epGenericID, position);
    lock::chip_stack_unlock();

    LevelControl::Commands::MoveWithOnOff::Type commandDataT;
    commandDataT.moveMode = (position == UP ? EMBER_ZCL_MOVE_MODE_UP : EMBER_ZCL_MOVE_MODE_DOWN);
    commandDataT.rate.SetNonNull(250);

    send_bound(epLightID, LevelControl::Id, commandDataT);
}
~~~

jonsmirl, 2023-03-04:
I wrote the problem up as a stackoverflow question, maybe someone will tell me how to convince the C++ compiler to do what I want.   https://stackoverflow.com/questions/75637945/taking-the-address-of-template-function-where-template-uses-type-as-parameter-i

jonsmirl, 2023-03-04:
Meanwhile, I will just give on a common routine and make the giant switch case like you are doing.

jonsmirl, 2023-03-05:
So far I have reduced the callback down to this:
~~~
void Device::unicast_cb(client::peer_device_t *peer_device, client::command_handle_t *cmd_handle, void *priv_data)
{
    auto onSuccess = [](const ConcreteCommandPath & commandPath, const chip::app::StatusIB & status, const auto & dataResponse) {
        ChipLogProgress(NotSpecified, "OnOff command succeeds");
    };

    auto onFailure = [](CHIP_ERROR error) {
        ChipLogError(NotSpecified, "OnOff command failed: %" CHIP_ERROR_FORMAT, error.Format());
    };

    if (cmd_handle->cluster_id == OnOff::Id) {
    } else if (cmd_handle->cluster_id == LevelControl::Id) {
        switch(cmd_handle->command_id) {
            case LevelControl::Commands::MoveWithOnOff::Id:
            {
                Controller::InvokeCommandRequest(peer_device->GetExchangeManager(),
                    peer_device->GetSecureSession().Value(), cmd_handle->endpoint_id,
                    *((LevelControl::Commands::MoveToLevelWithOnOff::Type *)cmd_handle->command_data),
                    onSuccess, onFailure);
                break;
            };
        }
    }
}

void Device::group_cb(uint8_t fabric_index, client::command_handle_t *cmd_handle, void *priv_data)
{
    Messaging::ExchangeManager & exchangeMgr = Server::GetInstance().GetExchangeManager();

    if (cmd_handle->cluster_id == OnOff::Id) {
    } else if (cmd_handle->cluster_id == LevelControl::Id) {
        switch(cmd_handle->command_id) {
            case LevelControl::Commands::MoveWithOnOff::Id:
            {
                Controller::InvokeGroupCommandRequest(&exchangeMgr, fabric_index, cmd_handle->group_id,
                    *((LevelControl::Commands::MoveToLevelWithOnOff::Type *)cmd_handle->command_data));
                break;
            };
        }
    }
}
~~~
But if can figure out how to take the address of the functions it can be reduced down to this common routine which does not need to be customized. 
~~~
void Device::unicast_cb(client::peer_device_t *peer_device, client::command_handle_t *cmd_handle, void *priv_data)
{
    auto onSuccess = [](const ConcreteCommandPath & commandPath, const chip::app::StatusIB & status, const auto & dataResponse) {
        ChipLogProgress(NotSpecified, "OnOff command succeeds");
    };

    auto onFailure = [](CHIP_ERROR error) {
        ChipLogError(NotSpecified, "OnOff command failed: %" CHIP_ERROR_FORMAT, error.Format());
    };

    cmd_handle->unicast_invoke(peer_device->GetExchangeManager(),
        peer_device->GetSecureSession().Value(), cmd_handle->endpoint_id,
        cmd_handle->command_data, onSuccess, onFailure);
}

void Device::group_cb(uint8_t fabric_index, client::command_handle_t *cmd_handle, void *priv_data)
{
    Messaging::ExchangeManager & exchangeMgr = Server::GetInstance().GetExchangeManager();

    cmd_handle->group_invoke(&exchangeMgr, fabric_index, cmd_handle->group_id, cmd_handle->command_data));
}
~~~
you'd call it like this:
~~~
    ESP_LOGI(TAG, "Long button pressed %x", position);
    uint16_t epGenericID = endpoint::get_id(epGeneric);
    uint16_t epLightID = endpoint::get_id(epLight);
    long_press_active = true;

    LevelControl::Commands::MoveWithOnOff::DecodableType commandDataDT;
    commandDataDT.moveMode = (position == UP ? EMBER_ZCL_MOVE_MODE_UP : EMBER_ZCL_MOVE_MODE_DOWN);
    commandDataDT.rate.SetNonNull(250);
    app::ConcreteCommandPath commandPath = {epLightID, LevelControl::Id, LevelControl::Commands::MoveWithOnOff::Id};
    app::CommandHandler commandObj(nullptr);

    // Press moves Position from 0 (idle) to 1 (press)
    lock::chip_stack_lock(portMAX_DELAY);
    emberAfLevelControlClusterMoveWithOnOffCallback(&commandObj, commandPath, commandDataDT);
    chip::app::Clusters::Switch::Attributes::CurrentPosition::Set(epGenericID, position);
    // LongPress event takes newPosition as event data
    switch_cluster::event::send_long_press(epGenericID, position);
    lock::chip_stack_unlock();

    LevelControl::Commands::MoveWithOnOff::Type commandDataT;
    commandDataT.moveMode = (position == UP ? EMBER_ZCL_MOVE_MODE_UP : EMBER_ZCL_MOVE_MODE_DOWN);
    commandDataT.rate.SetNonNull(250);

    client::command_handle_t cmd_handle;
    uint16_t local_endpoint_id = epLightID;
    cmd_handle.cluster_id = LevelControl::Id;
    cmd_handle.command_id = LevelControl::Commands::MoveWithOnOff::Id;
    cmd_handle.command_data = &commandDataT;

    const decltype(&Controller::InvokeCommandRequest<chip::app::Clusters::LevelControl::Commands::MoveWithOnOff::Type>) invoke_address =
        static_cast<ChipError (*)(Messaging::ExchangeManager *, const SessionHandle &, chip::EndpointId,
                     const chip::app::Clusters::LevelControl::Commands::MoveWithOnOff::Type &,
                     chip::app::Clusters::LevelControl::Commands::MoveWithOnOff::Type::ResponseType::OnSuccessCallbackType,
                     chip::app::Clusters::LevelControl::Commands::MoveWithOnOff::Type::ResponseType::OnErrorCallbackType,
                     const Optional<uint16_t> &,
                     const Optional<System::Clock::Timeout> &)>
            (&Controller::InvokeCommandRequest<chip::app::Clusters::LevelControl::Commands::MoveWithOnOff::Type>);

   cmd_handle.unicast_invoke = invoke_address;

   // similar for group address.

    client::cluster_update(local_endpoint_id, &cmd_handle);
~~~

jonsmirl, 2024-05-28:
this has finally been fixed in CHIP. May 2024

closed 2024-05-28 by jonsmirl