espressif/esp-matter #238: Implementing blinking LED for Indentify

jonsmirl, 2023-02-26:
What is the appropriate way to implement timer effects for identify? How do I delay the 1s to 8s for the blinks without allowing recursion into the Matter stack disrupting it? I think I need to use EmberEventControl.

jonsmirl, 2023-02-26:
SiLabs is doing this, I'll give it a try
~~~
silabs/efr32/src/AppTask.cpp:    switch (sIdentifyEffect)
silabs/efr32/src/AppTask.cpp:    case EMBER_ZCL_IDENTIFY_EFFECT_IDENTIFIER_BLINK:
silabs/efr32/src/AppTask.cpp:    case EMBER_ZCL_IDENTIFY_EFFECT_IDENTIFIER_BREATHE:
silabs/efr32/src/AppTask.cpp:    case EMBER_ZCL_IDENTIFY_EFFECT_IDENTIFIER_OKAY:
silabs/efr32/src/AppTask.cpp:        (void) chip::DeviceLayer::SystemLayer().StartTimer(chip::System::Clock::Seconds16(5), OnTriggerIdentifyEffectCompleted,
silabs/efr32/src/AppTask.cpp:                                                           identify);
silabs/efr32/src/AppTask.cpp:    case EMBER_ZCL_IDENTIFY_EFFECT_IDENTIFIER_FINISH_EFFECT:
silabs/efr32/src/AppTask.cpp:        (void) chip::DeviceLayer::SystemLayer().CancelTimer(OnTriggerIdentifyEffectCompleted, identify);
silabs/efr32/src/AppTask.cpp:        (void) chip::DeviceLayer::SystemLayer().StartTimer(chip::System::Clock::Seconds16(1), OnTriggerIdentifyEffectCompleted,
silabs/efr32/src/AppTask.cpp:                                                           identify);
silabs/efr32/src/AppTask.cpp:    case EMBER_ZCL_IDENTIFY_EFFECT_IDENTIFIER_STOP_EFFECT:
silabs/efr32/src/AppTask.cpp:        (void) chip::DeviceLayer::SystemLayer().CancelTimer(OnTriggerIdentifyEffectCompleted, identify);
silabs/efr32/src/AppTask.cpp:        sIdentifyEffect = EMBER_ZCL_IDENTIFY_EFFECT_IDENTIFIER_STOP_EFFECT;
silabs/efr32/src/AppTask.cpp:        ChipLogProgress(Zcl, "No identifier effect");

~~~

jonsmirl, 2023-02-27:
Could you add mEffectVariant in addition to mCurrentEffectIdentifier on the callbacks?

https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_identify.cpp#L43

execute_callback(STOP, identify->mEndpoint, identify->mCurrentEffectIdentifier, identify->mEffectVariant);

shubhamdp, 2023-03-10:
`chip::DeviceLayer::SystemLayer().StartTimer()` adds the callback function to the chip stack queue and it will execute it in chip stack context.

If you want to reschedule the same callback from the callback that just works. `chip::DeviceLayer::SystemLayer().StartTimer()` can be called from the callback and it would post that to the chip stack and it would be recursion free.

Also, will release the fix for adding `mEffectVariant` to callback soon.

closed 2023-03-16 by espressif-bot