espressif/esp-matter #239: How do I know if I've already commissioned my device?

emmby, 2023-02-26:
I'd like to be able to distinguish between two states:

1. Brand new out of box experience, device needs to be commissioned onto the network, versus:
2. Device has already been commissioned and can make a network connection when it wants to.

I'd like to be able to display the commissioning QR code for state #1, but jump directly to making a network connection for state #2.

Basically i'm looking for some sort of "bool isCommissioned()" type method. Where is the right place to look?

emmby, 2023-02-26:
I found `chip::DeviceLayer::Internal::ESP32Utils::IsStationProvisioned()` in ESP32Utils.h, but it always returns false for me. Presumably because i'm calling it without calling `esp_matter::start` first? But I don't actually want to start matter (and incur the associated battery cost) just to see whether it's provisioned.

jonsmirl, 2023-02-27:
Did you notice these events in the samples?
https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L32
you could use them to save your own state.

emmby, 2023-02-27:
I’m trying to avoid spinning up anything battery intensive like the WiFi stack. It looks like those events require me to call matter start(). Is there a way to do that without actually starting up matter/turning on the radio?

jonsmirl, 2023-02-27:
Make an entry in flash commissioned = false. Now use those events to change it to commissioned = true when you are commissioned. When you want to know the state check the entry you made in flash to determine commissioned or not.

jonsmirl, 2023-02-27:
If you are extremely power sensitive consider looking at the BLE mesh bridge. Note the BLE mesh nodes can be either routing (mains power) or leaf (battery).

shubhamdp, 2023-02-27:
There is `FabricCount()` method which tells how many fabrics are present on the device, and that would be enough to know if device is commissioned or not.

https://github.com/project-chip/connectedhomeip/blob/3b80154ce7cde02bb19825da5d5b5026f90db9bd/src/credentials/FabricTable.h#L538-L541

But, for this we need chip server running, see that usage, below
https://github.com/project-chip/connectedhomeip/blob/3b80154ce7cde02bb19825da5d5b5026f90db9bd/src/app/server/Server.cpp#L262-L269

emmby, 2023-02-27:
I was able to rearrange my code to display the QR code later in the boot sequence, after i've initialized matter. It means a longer delay to seeing the QR, but it's not very noticeable. It would still be convenient to be able to determine provisioning status without starting up all the radios, but i'll close this for now.

closed 2023-02-27 by emmby