espressif/esp-matter #241: on_off_light endpoint is not working

SehribanTas, 2023-02-28:
I wanted to write the on_off_light endpoint example using the light examples, but the led is not blinking.
app_main.cpp ;
`extern "C" void app_main()
{
    esp_err_t err = ESP_OK;

    /* Initialize the ESP NVS layer */
    nvs_flash_init();

    /* Initialize driver */
    app_driver_handle_t light_handle = app_driver_light_init();
    app_driver_handle_t button_handle = app_driver_button_init();
    app_reset_button_register(button_handle);

    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

    on_off_light::config_t light_config;
    light_config.on_off.on_off = DEFAULT_POWER;
    light_config.on_off.lighting.start_up_on_off = nullptr;
  
    endpoint_t *endpoint = on_off_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);

    /* These node and endpoint handles can be used to create/add other endpoints and clusters. */
    if (!node || !endpoint) {
        ESP_LOGE(TAG, "Matter node creation failed");
    }

    light_endpoint_id = endpoint::get_id(endpoint);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id);

    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }

    /* Starting driver with default values */
    app_driver_light_set_defaults(light_endpoint_id);

#if CONFIG_ENABLE_CHIP_SHELL
    esp_matter::console::diagnostics_register_commands();
    esp_matter::console::wifi_register_commands();
    esp_matter::console::init();
#endif
}`

app_driver.cpp
`/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <esp_log.h>
#include <stdlib.h>
#include <string.h>

#include <device.h>
#include <esp_matter.h>
#include <led_driver.h>

#include <app_priv.h>

using namespace chip::app::Clusters;
using namespace esp_matter;

static const char *TAG = "app_driver";
extern uint16_t light_endpoint_id;

/* Do any conversions/remapping for the actual value here */
static esp_err_t app_driver_light_set_power(led_driver_handle_t handle, esp_matter_attr_val_t *val)
{
    return led_driver_set_power(handle, val->val.b);
}

static void app_driver_button_toggle_cb(void *arg, void *data)
{
    ESP_LOGI(TAG, "Toggle button pressed");
    uint16_t endpoint_id = light_endpoint_id;
    uint32_t cluster_id = OnOff::Id;
    uint32_t attribute_id = OnOff::Attributes::OnOff::Id;

    node_t *node = node::get();
    endpoint_t *endpoint = endpoint::get(node, endpoint_id);
    cluster_t *cluster = cluster::get(endpoint, cluster_id);
    attribute_t *attribute = attribute::get(cluster, attribute_id);

    esp_matter_attr_val_t val = esp_matter_invalid(NULL);
    attribute::get_val(attribute, &val);
    val.val.b = !val.val.b;
    attribute::update(endpoint_id, cluster_id, attribute_id, &val);
}

esp_err_t app_driver_attribute_update(app_driver_handle_t driver_handle, uint16_t endpoint_id, uint32_t cluster_id,
                                      uint32_t attribute_id, esp_matter_attr_val_t *val)
{
    esp_err_t err = ESP_OK;
    if (endpoint_id == light_endpoint_id) {
        led_driver_handle_t handle = (led_driver_handle_t)driver_handle;
        if (cluster_id == OnOff::Id) {
            if (attribute_id == OnOff::Attributes::OnOff::Id) {
                err = app_driver_light_set_power(handle, val);
            }
        }
    }
    return err;
}

esp_err_t app_driver_light_set_defaults(uint16_t endpoint_id)
{
    esp_err_t err = ESP_OK;
    void *priv_data = endpoint::get_priv_data(endpoint_id);
    led_driver_handle_t handle = (led_driver_handle_t)priv_data;
    node_t *node = node::get();
    endpoint_t *endpoint = endpoint::get(node, endpoint_id);
    cluster_t *cluster = NULL;
    attribute_t *attribute = NULL;
    esp_matter_attr_val_t val = esp_matter_invalid(NULL);

 

    /* Setting power */
    cluster = cluster::get(endpoint, OnOff::Id);
    attribute = attribute::get(cluster, OnOff::Attributes::OnOff::Id);
    attribute::get_val(attribute, &val);
    err |= app_driver_light_set_power(handle, &val);

    return err;
}

app_driver_handle_t app_driver_light_init()
{
    /* Initialize led */
    led_driver_config_t config = led_driver_get_config();
    led_driver_handle_t handle = led_driver_init(&config);
    return (app_driver_handle_t)handle;
}

app_driver_handle_t app_driver_button_init()
{
    /* Initialize button */
    button_config_t config = button_driver_get_config();
    button_handle_t handle = iot_button_create(&config);
    iot_button_register_cb(handle, BUTTON_PRESS_DOWN, app_driver_button_toggle_cb, NULL);
    return (app_driver_handle_t)handle;
}
`

dhrishi, 2023-02-28:
@SehribanTas What do you mean by led is not blinking? Which operation are you trying to perform where you expect blinking to happen? 

Are you using ESP32 or ESP32-C3? If you are using ESP32-C3, then it has an on-board led that is used to represent the on-off state.

SehribanTas, 2023-02-28:
I am using ESP32 , I turn the led on and off through the google home application, but the led does not switch to the on

dhrishi, 2023-02-28:
@SehribanTas On setting On/Off from the Google Home, if you see the device console prints like below, then it is working fine

```
I (307722) chip[EM]: >>> [E:49131r M:190523335] (S) Msg RX from 1:000000000001B669 [4552] --- Type 0001:08 (IM:InvokeCommandRequest)
I (307732) esp_matter_command: Received command 0x0001 for enpoint 0x0001's cluster 0x00000006
I (307742) chip[ZCL]: On/Off set value: 1 1
I (307742) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 0 **********
I (307752) chip[ZCL]: Toggle on/off from 0 to 1
```
Note: The above is for On command

There's an onboard LED on ESP32-C3 which reflects the state. So, if you use that, you can see the LED changing state

SehribanTas, 2023-03-01:
when ı flashing light example , ı can control on/off state with google home or matter console like below

[image]

But I dont want to level control , so ı change the endpoint as on_off_light (you can see the codes in the first question).  unfortunately  ı cant control on/off state with google home or matter console like below . 

[image]

in this situation device consoler prints like below , it seems no problem but leds on/off state is not changing

[image]
i couldn't solve the problem

jadhavrohit924, 2023-03-01:
@SehribanTas In the driver, we are setting RGB value based on HSL. OnOffLight don't have HSL so RGB values are always `0` and never change. That's why LED doesn’t blink.

We will fix it.

jadhavrohit924, 2023-03-09:
@SehribanTas Since we have decided not to update driver, you can use this driver patch for on/off light.

[on-off-driver.patch](https://github.com/espressif/esp-matter/files/10927564/on-off-driver.patch)

You just have to pass `true` if your device is on/off light  at `led_driver_set_power` in example/app_driver.cpp

dhrishi, 2023-03-10:
@SehribanTas Please close the issue if it works

dhrishi, 2023-03-14:
It should work with this patch. Please re-open in case of any problems

closed 2023-03-14 by dhrishi