espressif/esp-matter #242: BLE bridge not reliably provisioning.

jonsmirl, 2023-03-01:
I'm testing my BLE bridge and my test fixture has four BLE bulbs in it. So I put four unprovisioned bulbs in it and turned my bridge on.  The bridge is unable to get all four bulbs provisioned and it fails in multiple ways.

jonsmirl, 2023-03-01:
In this failure the wifi is provisioned by Matter, but some how the BLE system make wifi halt.
~~~
jonsmirl@ares:~/lowpan$ idf.py monitor
Executing action: monitor
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C3
Running idf_monitor in directory /home/jonsmirl/lowpan
Executing "/home/jonsmirl/.espressif/python_env/idf4.4_py3.10_env/bin/python /home/jonsmirl/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace /home/jonsmirl/lowpan/build/lowpan.elf -m '/home/jonsmirl/.espressif/python_env/idf4.4_py3.10_env/bin/python' '/home/jonsmirl/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c634
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x16bc
load:0x403cc710,len:0x91c
load:0x403ce710,len:0x2d00
entry 0x403cc710
I (24) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (24) boot: compile time 16:26:39
I (25) boot: chip revision: 3
I (26) boot.esp32c3: SPI Speed      : 80MHz
I (31) boot.esp32c3: SPI Mode       : DIO
I (36) boot.esp32c3: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (64) boot:  1 nvs              WiFi data        01 02 00010000 00008000
I (72) boot:  2 otadata          OTA data         01 00 00018000 00002000
I (79) boot:  3 phy_init         RF data          01 01 0001a000 00001000
I (87) boot:  4 ota_0            OTA app          00 10 00020000 001d0000
I (94) boot:  5 ota_1            OTA app          00 11 001f0000 001d0000
I (102) boot:  6 fctry            WiFi data        01 02 003c0000 00006000
I (109) boot: End of partition table
I (114) esp_image: segment 0: paddr=00020020 vaddr=3c140020 size=40678h (263800) map
I (163) esp_image: segment 1: paddr=000606a0 vaddr=3fc91c00 size=03fe0h ( 16352) load
I (167) esp_image: segment 2: paddr=00064688 vaddr=40380000 size=0b990h ( 47504) load
I (178) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=130418h (1246232) map
I (373) esp_image: segment 4: paddr=001a0440 vaddr=4038b990 size=06164h ( 24932) load
I (378) esp_image: segment 5: paddr=001a65ac vaddr=50000010 size=00010h (    16) load
I (385) boot: Loaded app from partition at offset 0x20000
I (385) boot: Disabling RNG early entropy source...
I (391) cpu_start: Pro cpu up.
I (406) cpu_start: Pro cpu start user code
I (406) cpu_start: cpu freq: 160000000
I (406) cpu_start: Application information:
I (408) cpu_start: Project name:     lowpan
I (413) cpu_start: App version:      v1.0
I (418) cpu_start: Compile time:     Feb 22 2023 16:26:32
I (424) cpu_start: ELF file SHA256:  c268663d7c7521a5...
I (430) cpu_start: ESP-IDF:          v4.4.3-dirty
I (435) heap_init: Initializing. RAM available for dynamic allocation:
I (443) heap_init: At 3FCA96B0 len 00033060 (204 KiB): DRAM
I (449) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (456) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (462) spi_flash: detected chip: generic
I (463) spi_flash: flash io: dio
I (464) sleep: Configure to isolate all GPIO pins in sleep state
I (469) sleep: Enable automatic switching of GPIO sleep configuration
I (476) coexist: coexist rom version 9387209
I (476) cpu_start: Starting scheduler.
I (520) gpio: GPIO[8]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (520) ep: Initializing LED strip
I (530) esp_matter_cluster: Creating default binding cluster
W (530) esp_matter_core: Command 0x0047 on cluster 0x0300 already exists. Not creating again.
W (540) esp_matter_core: Server Cluster 0x001e on endpoint 0x0002 already exists. Not creating again.
[... 804 lines cut ...]
I (10920) NimBLE: GATT procedure initiated: write no rsp; 
I (10920) NimBLE: att_handle=23 len=18

I (10950) NimBLE: GAP procedure initiated: 
I (10950) NimBLE: connection parameter update; conn_handle=1 itvl_min=16 itvl_max=32 latency=0 supervision_timeout=500 min_ce_len=0 max_ce_len=0
I (10950) NimBLE: 

I (11040) NimBLE: GATT procedure initiated: write no rsp; 
I (11040) NimBLE: att_handle=23 len=20

I (11040) NimBLE: GATT procedure initiated: write no rsp; 
I (11040) NimBLE: att_handle=23 len=16

I (13060) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

I (13100) chip[DL]: BLE GAP connection terminated (con 1 reason 0x216)
I (18500) wifi:bcn_timout,ap_probe_send_start
I (21000) wifi:ap_probe_send over, resett wifi status to disassoc
I (21010) wifi:state: run -> init (c800)
I (21010) wifi:pm stop, total sleep time: 11132133 us / 19585186 us

W (21010) wifi:<ba-del>idx
I (21010) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1

~~~

jonsmirl, 2023-03-01:
No free slots for segmented messages...
~~~
jonsmirl@ares:~/lowpan$ idf.py monitor
Executing action: monitor
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C3
Running idf_monitor in directory /home/jonsmirl/lowpan
Executing "/home/jonsmirl/.espressif/python_env/idf4.4_py3.10_env/bin/python /home/jonsmirl/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace /home/jonsmirl/lowpan/build/lowpan.elf -m '/home/jonsmirl/.espressif/python_env/idf4.4_py3.10_env/bin/python' '/home/jonsmirl/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c97a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x16bc
load:0x403cc710,len:0x91c
load:0x403ce710,len:0x2d00
entry 0x403cc710
I (24) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (24) boot: compile time 16:26:39
I (25) boot: chip revision: 3
I (26) boot.esp32c3: SPI Speed      : 80MHz
I (31) boot.esp32c3: SPI Mode       : DIO
I (36) boot.esp32c3: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (64) boot:  1 nvs              WiFi data        01 02 00010000 00008000
I (72) boot:  2 otadata          OTA data         01 00 00018000 00002000
I (79) boot:  3 phy_init         RF data          01 01 0001a000 00001000
I (87) boot:  4 ota_0            OTA app          00 10 00020000 001d0000
I (94) boot:  5 ota_1            OTA app          00 11 001f0000 001d0000
I (102) boot:  6 fctry            WiFi data        01 02 003c0000 00006000
I (109) boot: End of partition table
I (114) esp_image: segment 0: paddr=00020020 vaddr=3c140020 size=40678h (263800) map
I (163) esp_image: segment 1: paddr=000606a0 vaddr=3fc91c00 size=03fe0h ( 16352) load
I (167) esp_image: segment 2: paddr=00064688 vaddr=40380000 size=0b990h ( 47504) load
I (178) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=130418h (1246232) map
I (373) esp_image: segment 4: paddr=001a0440 vaddr=4038b990 size=06164h ( 24932) load
I (378) esp_image: segment 5: paddr=001a65ac vaddr=50000010 size=00010h (    16) load
I (385) boot: Loaded app from partition at offset 0x20000
I (385) boot: Disabling RNG early entropy source...
I (391) cpu_start: Pro cpu up.
I (406) cpu_start: Pro cpu start user code
I (406) cpu_start: cpu freq: 160000000
I (406) cpu_start: Application information:
I (408) cpu_start: Project name:     lowpan
I (413) cpu_start: App version:      v1.0
I (418) cpu_start: Compile time:     Feb 22 2023 16:26:32
I (424) cpu_start: ELF file SHA256:  c268663d7c7521a5...
I (430) cpu_start: ESP-IDF:          v4.4.3-dirty
I (435) heap_init: Initializing. RAM available for dynamic allocation:
I (443) heap_init: At 3FCA96B0 len 00033060 (204 KiB): DRAM
I (449) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (456) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (462) spi_flash: detected chip: generic
I (463) spi_flash: flash io: dio
I (464) sleep: Configure to isolate all GPIO pins in sleep state
I (469) sleep: Enable automatic switching of GPIO sleep configuration
I (476) coexist: coexist rom version 9387209
I (476) cpu_start: Starting scheduler.
I (520) gpio: GPIO[8]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (520) ep: Initializing LED strip
I (530) esp_matter_cluster: Creating default binding cluster
W (530) esp_matter_core: Command 0x0047 on cluster 0x0300 already exists. Not creating again.
W (540) esp_matter_core: Server Cluster 0x001e on endpoint 0x0002 already exists. Not creating again.
[... 834 lines cut ...]
I (15090) NimBLE: GAP procedure initiated: stop advertising.

I (15090) NimBLE: GAP procedure initiated: advertise; 
I (15090) NimBLE: disc_mode=0
I (15090) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (15090) NimBLE: 

I (15180) NimBLE: GAP procedure initiated: stop advertising.

I (15600) NimBLE: GAP procedure initiated: advertise; 
I (15600) NimBLE: disc_mode=0
I (15600) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (15600) NimBLE: 

I (15690) NimBLE: GAP procedure initiated: stop advertising.

I (22820) wifi:bcn_timout,ap_probe_send_start
I (25330) wifi:ap_probe_send over, resett wifi status to disassoc
I (25330) wifi:state: run -> init (c800)
I (25330) wifi:pm stop, total sleep time: 14108434 us / 23984823 us

W (25330) wifi:<ba-del>idx
I (25330) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1

~~~

jonsmirl, 2023-03-01:
error with composition message
~~~
jonsmirl@ares:~/lowpan$ idf.py monitor
Executing action: monitor
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C3
Running idf_monitor in directory /home/jonsmirl/lowpan
Executing "/home/jonsmirl/.espressif/python_env/idf4.4_py3.10_env/bin/python /home/jonsmirl/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace /home/jonsmirl/lowpan/build/lowpan.elf -m '/home/jonsmirl/.espressif/python_env/idf4.4_py3.10_env/bin/python' '/home/jonsmirl/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c97a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x16bc
load:0x403cc710,len:0x91c
load:0x403ce710,len:0x2d00
entry 0x403cc710
I (24) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (24) boot: compile time 16:26:39
I (25) boot: chip revision: 3
I (26) boot.esp32c3: SPI Speed      : 80MHz
I (31) boot.esp32c3: SPI Mode       : DIO
I (36) boot.esp32c3: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (64) boot:  1 nvs              WiFi data        01 02 00010000 00008000
I (72) boot:  2 otadata          OTA data         01 00 00018000 00002000
I (79) boot:  3 phy_init         RF data          01 01 0001a000 00001000
I (87) boot:  4 ota_0            OTA app          00 10 00020000 001d0000
I (94) boot:  5 ota_1            OTA app          00 11 001f0000 001d0000
I (102) boot:  6 fctry            WiFi data        01 02 003c0000 00006000
I (109) boot: End of partition table
I (114) esp_image: segment 0: paddr=00020020 vaddr=3c140020 size=40678h (263800) map
I (163) esp_image: segment 1: paddr=000606a0 vaddr=3fc91c00 size=03fe0h ( 16352) load
I (167) esp_image: segment 2: paddr=00064688 vaddr=40380000 size=0b990h ( 47504) load
I (178) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=130418h (1246232) map
I (373) esp_image: segment 4: paddr=001a0440 vaddr=4038b990 size=06164h ( 24932) load
I (378) esp_image: segment 5: paddr=001a65ac vaddr=50000010 size=00010h (    16) load
I (385) boot: Loaded app from partition at offset 0x20000
I (385) boot: Disabling RNG early entropy source...
I (391) cpu_start: Pro cpu up.
I (406) cpu_start: Pro cpu start user code
I (406) cpu_start: cpu freq: 160000000
I (406) cpu_start: Application information:
I (408) cpu_start: Project name:     lowpan
I (413) cpu_start: App version:      v1.0
I (418) cpu_start: Compile time:     Feb 22 2023 16:26:32
I (424) cpu_start: ELF file SHA256:  c268663d7c7521a5...
I (430) cpu_start: ESP-IDF:          v4.4.3-dirty
I (435) heap_init: Initializing. RAM available for dynamic allocation:
I (443) heap_init: At 3FCA96B0 len 00033060 (204 KiB): DRAM
I (449) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (456) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (462) spi_flash: detected chip: generic
I (463) spi_flash: flash io: dio
I (464) sleep: Configure to isolate all GPIO pins in sleep state
I (469) sleep: Enable automatic switching of GPIO sleep configuration
I (476) coexist: coexist rom version 9387209
I (476) cpu_start: Starting scheduler.
I (521) gpio: GPIO[8]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (531) ep: Initializing LED strip
I (531) esp_matter_cluster: Creating default binding cluster
W (541) esp_matter_core: Command 0x0047 on cluster 0x0300 already exists. Not creating again.
W (541) esp_matter_core: Server Cluster 0x001e on endpoint 0x0002 already exists. Not creating again.
[... 855 lines cut ...]
I (11471) NimBLE: GATT procedure initiated: write no rsp; 
I (11471) NimBLE: att_handle=23 len=18

I (11501) NimBLE: GAP procedure initiated: 
I (11501) NimBLE: connection parameter update; conn_handle=1 itvl_min=16 itvl_max=32 latency=0 supervision_timeout=500 min_ce_len=0 max_ce_len=0
I (11501) NimBLE: 

I (11591) NimBLE: GATT procedure initiated: write no rsp; 
I (11591) NimBLE: att_handle=23 len=20

I (11601) NimBLE: GATT procedure initiated: write no rsp; 
I (11601) NimBLE: att_handle=23 len=16

I (13611) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

I (13651) chip[DL]: BLE GAP connection terminated (con 1 reason 0x216)
I (16901) wifi:bcn_timout,ap_probe_send_start
I (19401) wifi:ap_probe_send over, resett wifi status to disassoc
I (19401) wifi:state: run -> init (c800)
I (19401) wifi:pm stop, total sleep time: 9579577 us / 18051858 us

W (19401) wifi:<ba-del>idx
I (19401) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1

~~~

wqx6, 2023-03-01:
Can you update the log with a file instead of the long comment?

closed 2023-03-01 by jonsmirl