espressif/esp-matter #245: qrcode device category

stefano055415, 2023-03-01:
How is it possible to generate a qrcode that also contains the information of the device category

With apple home, if you scan the qrcode generated, the app shows me a generic accessory.

[image]

I wish he would show me the correct category, in my case a light.

I've tried other devices matter that show it

[image]

but using mfg_tool I don't see a parameter to set this setting

dhrishi, 2023-03-01:
@stefano055415 

>How is it possible to generate a qrcode that also contains the information of the device category

QR code does not contain the device type/category

>I wish he would show me the correct category, in my case a light.

If the product model (including the vid, pid and other details) is uploaded in the DCL, then the correct category would be shown by the Apple Home. It reaches to the DCL to check if there's an entry for that device after scanning the QR code

> I've tried other devices matter that show it

Please check the vid of the device that is showing this and it should be present [here](https://webui.dcl.csa-iot.org/models)

stefano055415, 2023-03-02:
@dhrishi  ok i understand thanks

closed 2023-03-02 by stefano055415