espressif/esp-matter #246: Device name Apple home

stefano055415, 2023-03-01:
When I add an accessory to Apple home, the default name is always Matter Accessory.

[image]

I tried to use the fixed_label, common_name, product_name,... with mfg_tool, but the result doesn't change

Can someone help me?

Thank you

dhrishi, 2023-03-01:
@stefano055415 

>When I add an accessory to Apple home, the default name is always Matter Accessory.

As per our observations -- This name, similar to https://github.com/espressif/esp-matter/issues/245#issuecomment-1450368745, is pre-populated or set as default to the Product Name field in the DCL model, if the entry is present for that particular vid and pid. If not, it is set as Matter Accessory.

stefano055415, 2023-03-02:
@dhrishi  ok i understand thanks

closed 2023-03-02 by stefano055415