espressif/esp-matter #247: Expanding the bridges for more device types.

jonsmirl, 2023-03-01:
I've been looking at the BLE mesh bridge and what is there works, but what is there is only a small piece of a complete bridge. What are Espressif's plans for these bridges (both Zigbee and  BLE)?  To be more useful the device support needs to expand up through Extended Color bulb. There also needs to be some sort of keep alive scheme to keep the bridge and bulb state synchronized.  Also missing support for OOB keys.

Is the plan just do a bare minimum demo and then leave it up to the customer to complete the bridge? In that case the customers are going to be developing the same code over and over.  Implementing this code requires detailed knowledge of both the Matter and BLE/Zigbee device types because you have to build the mappings between the two types.

Or is the plan to provide an app_ble or app_zigbee which implements a turnkey complete bridge which can just be added as a component to your app?

dhrishi, 2023-03-01:
@jonsmirl 

> but what is there is only a small piece of a complete bridge.

Yes, I agree.

The plan is to add the common support that would be generically required under the bridge functionality for majority of the usecases. The applications can then bridge the application device types of their choice. We will review the suggestions proposed here and start with adding the relevant ones

jonsmirl, 2023-03-01:
The core of bridge is not a place for vendor value-add. It is best for consumers if all of the bridge cores are identical (ie the same code) then bridge behavior will be consistent.  Vendor value add can occur in the UI for controlling the bridge.

Think of wifi routers. They all run the identical Linux core bridging software yet we have thousands of models.

dhrishi, 2023-03-10:
https://github.com/espressif/esp-matter/issues/264#issuecomment-1458186367

jonsmirl, 2023-10-31:
A solution for allowing the user to expand the device types is in this issue
https://github.com/espressif/esp-matter/issues/647

closed 2023-10-31 by jonsmirl