espressif/esp-matter #248: RAM consumption

jonsmirl, 2023-03-01:
I keep running out of RAM and the reason is always the same. For example I implement a single ExtendedColor light and it fits into RAM, and then I create four to eight instances of the same code and I run out of RAM on the C3. Is it possible to make the esp-matter code more RAM efficient when allocating arrays of identical devices? I will also note that the majority of the RAM consumption is occurring in the esp-matter code, not the CHIP library (this is expected since esp-matter contains the buffers).

Can you use the knowledge that this is an array of eight OnOff clusters to reduce memory consumption over allocating eight standalone OnOff clusters?

For example, say a cluster has five string fields each allowable up to 1KB in length and you want eight instances of this cluster. With zap rules this will allocate 5 * 1KB * 8 = 40KB memory.  In normal use the strings are rarely 1KB and most are 30 bytes, but you don't know the string length until the command packet arrives.  Another approach would be to dynamically allocate a 1KB buffer, TLV decode into it, measure the string, allocate the exact amount of RAM needed, copy the string and free the 1KB temp buffer. Doing it that way uses less than 1KB instead of 40KB.  Of course this over commits the available RAM; just send out-of-memory errors when you run out.

Another issue is minimum heap allocations. The minimum chunks on ESP seem to be 8 or 16 bytes. So if you allocate a 16b word 8 times it consumes 8 * 8 = 64b memory. But it you allocate an array of 8 16b words the heap object would be 24b (2 * 8 + heap overhead). Even better if you can combine multiple attributes into a single array.  This heap granularity may be the main source of excess RAM consumption in the existing code.

I can tell you ahead of time how many instances to pre-allocate. For example I could say pre-allocate 8 copies of OnOff, LevelControl, Color Control, Descriptor, Groups, Scenes, etc.  Then use the existing API to tie those pre-allocated clusters into end points.  This would let you combine hundreds of tiny allocations for all of the fixed length fields into a single large one.

jadhavrohit924, 2023-03-01:
@jonsmirl Thanks for the suggestions!! We will look into it.

jonsmirl, 2023-03-24:
I am in much better shape now with esp-matter in PSRAM support. I will open new bug if I run out of memory again.

closed 2023-03-24 by jonsmirl