espressif/esp-matter #249: Unable to commission the device with on off device type

aggaddam, 2023-03-02:
Hi, 

I started exploring matter with example code for light provided in ESP-Matter SDK. I was able to commission the device the light with CHIP tool and android developer matter app.

Later, I modified the light example to use as on_off_light instead of color_temperature_light. With on off light, I am unable to commission the device. App shows Not a Matter - certified device.
I didn't modify PID, or VID from the menu config.

dhrishi, 2023-03-02:
@aggaddam This is very strange. Ideally, it should not be anything on the device side. Please try with chip-tool too and then upload the logs on the device side as well as chip-tool side.

aggaddam, 2023-03-03:
@dhrishi 
With the chip tool, it's working fine. 
With the Google Home Sample App for Matter (GHSAFM), it was working till last week. But from this week onwards it is getting failed with an Error message - "Not a Matter-certified  device"

Here is the log file 
`
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16bc
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
I (30) boot: ESP-IDF cf637563-dirty 2nd stage bootloader
I (30) boot: compile time 15:00:28
I (30) boot: chip revision: 3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (70) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (85) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (93) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (100) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (108) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (115) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (123) boot: End of partition table
I (127) boot: No factory image, trying OTA 0
I (132) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=30330h (197424) map
I (172) esp_image: segment 1: paddr=00050358 vaddr=3fc90400 size=03720h ( 14112) load
I (174) esp_image: segment 2: paddr=00053a80 vaddr=40380000 size=0c598h ( 50584) load
I (187) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=f183ch (989244) map
I (341) esp_image: segment 4: paddr=00151864 vaddr=4038c598 size=03c9ch ( 15516) load
I (344) esp_image: segment 5: paddr=00155508 vaddr=50000010 size=00010h (    16) load
I (352) boot: Loaded app from partition at offset 0x20000
I (395) boot: Set actual ota_seq=1 in otadata[0]
I (395) boot: Disabling RNG early entropy source...
I (406) cpu_start: Pro cpu up.
D (406) efuse: In EFUSE_BLK2__DATA4_REG is used 3 bits starting with 0 bit
D (406) efuse: In EFUSE_BLK2__DATA4_REG is used 8 bits starting with 12 bit
D (413) efuse: In EFUSE_BLK1__DATA3_REG is used 3 bits starting with 18 bit
D (420) efuse: In EFUSE_BLK1__DATA5_REG is used 5 bits starting with 5 bit
D (426) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 7 bit
D (433) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 14 bit
D (440) efuse: In EFUSE_BLK1__DATA4_REG is used 8 bits starting with 21 bit
D (447) efuse: In EFUSE_BLK1__DATA4_REG is used 3 bits starting with 29 bit
D (454) efuse: In EFUSE_BLK1__DATA5_REG is used 5 bits starting with 0 bit
D (469) clk: RTC_SLOW_CLK calibration value: 3786598
I (478) cpu_start: Pro cpu start user code
I (478) cpu_start: cpu freq: 160000000
I (478) cpu_start: Application information:
I (481) cpu_start: Project name:     light
I (485) cpu_start: App version:      v1.0
I (490) cpu_start: Compile time:     Mar  2 2023 15:00:24
I (496) cpu_start: ELF file SHA256:  130c93c0f41db7f9...
I (502) cpu_start: ESP-IDF:          cf637563-dirty
D (508) memory_layout: Checking 4 reserved memory ranges:
D (513) memory_layout: Reserved memory range 0x3fc80000 - 0x3fc90400
D (520) memory_layout: Reserved memory range 0x3fc90400 - 0x3fca3e60
D (526) memory_layout: Reserved memory range 0x3fcdf060 - 0x3fce0000
D (533) memory_layout: Reserved memory range 0x50000000 - 0x50000020
D (539) memory_layout: Building list of available memory regions:
D (545) memory_layout: Available memory region 0x3fca3e60 - 0x3fcc0000
D (552) memory_layout: Available memory region 0x3fcc0000 - 0x3fcdf060
[... 1107 lines cut ...]
I (108967) chip[IN]: (S) Sending msg 35018075 on secure session with LSID: 40349
D (108975) CHIP[DL]: Sending indication for CHIPoBLE TX characteristic (con 1, len 128)
I (108983) NimBLE: GATT procedure initiated: indicate; 
I (108989) NimBLE: att_handle=14

D (108994) chip[DMG]: Command handler moving to [CommandSen]
D (108999) chip[DMG]: Command handler moving to [AwaitingDe]
E (109005) chip[DL]: Long dispatch time: 619 ms, for event type 7
I (109052) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
D (109053) CHIP[DL]: Sending indication for CHIPoBLE TX characteristic (con 1, len 128)
I (109059) NimBLE: GATT procedure initiated: indicate; 
I (109065) NimBLE: att_handle=14

I (109142) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
D (109143) CHIP[DL]: Sending indication for CHIPoBLE TX characteristic (con 1, len 128)
I (109149) NimBLE: GATT procedure initiated: indicate; 
I (109155) NimBLE: att_handle=14

I (109232) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (110852) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 1 ) indicate = 0
I (110853) chip[DL]: CHIPoBLE unsubscribe received
I (110858) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
D (110865) chip[IN]: Clearing BLE pending packets.
I (110870) chip[BLE]: Releasing end point's BLE connection back to application.
ESP-ROM:ESP-`

dhrishi, 2023-03-03:
>With the chip tool, it's working fine.

@aggaddam This indicates that it should not be an issue on the ESP32 side. The logs above show that the CSRRequest is received by the ESP32 and then it sends the response. After that, the phone app should install the RootCA and NOC on the ESP32 but that doesn't happen. 

Would you mind raising an issue in the Google Matter sample app repository instead? I see a similar issue reported there: https://github.com/google-home/sample-app-for-matter-android/issues/93

dhrishi, 2023-03-04:
@aggaddam Can you please close the issue here and track the above?

closed 2023-03-06 by aggaddam