espressif/esp-matter #250: psram for matter source code

maenkai, 2023-03-02:
After testing, it takes 4kb to create a bridge device, and my chip supports psram, I want the Matter source code to only use psram, how should I achieve this

shubhamdp, 2023-03-03:
@maenkai Can you please take a look at the IDF's Support for [External RAM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html).
You can enable few config options to start using PSRAM.

jonsmirl, 2023-03-03:
We have had a lot of trouble with this. The problem is that CHIP uses a lot of BSS and DATA. The option in the IDF to move BSS/DATA into PSRAM on the S3 is broken with a note that it will be fixed in IDF5.  So you can turn on PSRAM support (we have it on and have moved everything except CHIP to it) and CHIP will continue to use a large amount of SRAM.

In our use case we would like to push ALL of CHIP into PSRAM so that we can have the LCD buffer in SRAM. Since we can't do that we were forced into making the LCD work out of PSRAM which is doable but it is tricky to make work.

maenkai, 2023-03-07:
> @maenkai能否请您看一下 IDF[对外部 RAM 的支持](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html)。 您可以启用几个配置选项以开始使用 PSRAM。

I want esp-matter's dynamic memory to use psram.
Where should I modify it.

dhrishi, 2023-03-07:
@maenkai We are working on providing an option to allocate the data model from PSRAM if available and chosen by the user. Please allow us a few days before we get this out.

jonsmirl, 2023-03-11:
I am stuck now. Now I have to go search these libraries and start hand moving buffers one by one to PSRAM. 6MB of empty PSRAM sitting there unused.
~~~
Total sizes:
Used static IRAM:   86162 bytes ( 276078 remain, 23.8% used)
      .text size:   85135 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:  174237 bytes (  -1309 remain, 100.8% used) Overflow detected! You can run idf.py size-files for more information.
      .data size:   27053 bytes
      .bss  size:  147184 bytes
~~~

jonsmirl, 2023-03-13:
Hmm... who are the top consumers????  What is that friend.c? I tried turing on Bluetooth in PSRAM but it fails. BLE Mesh issue maybe?
~~~
        Object File DRAM .data .rtc.data DRAM .bss IRAM0 .text & 0.vectors ram_st_total Flash .text & .rodata & .rodata_noload & .appdesc flash_total
friend.c.obj			24	0	62055	0	0	62079	6830	2678	0	0	9532
Server.cpp.obj			4	0	17264	0	0	17268	6042	1211	0	0	7257
InteractionModelEngine.c	0	0	9816	0	0	9816	7003	1237	0	0	8240
net.c.obj			5084	0	3762	0	0	8846	4988	1466	0	0	11538
tasks.c.obj			8	0	704	7856	0	8568	0	1542	0	0	9406
phy_chip_v7_newrom.o		203	0	0	6644	0	6847	0	0	0	0	6847
adv.c.obj			56	0	4580	0	0	4636	2553	1458	0	0	4067
heap_tlsf.c.obj			1589	0	0	2637	0	4226	0	0	0	0	4226
ringbuf.c.obj			0	0	0	3761	0	3761	0	2317	0	0	6078
portasm.S.obj			3084	0	0	424	0	3508	0	0	0	0	3508
queue.c.obj			0	0	0	3363	0	3363	0	1593	0	0	4956
wl_cnx.o			1	0	3183	58	0	3242	10219	1068	2287	0	13633
phy_chip_v7.o			2129	0	271	600	0	3000	7525	0	0	0	10254
pp.o				1153	0	16	1814	0	2983	4722	497	0	0	8186
spi_flash_chip_generic.c	559	0	0	2310	0	2869	0	0	0	0	2869
lld_con.o			0	0	84	2652	0	2736	249	10	0	0	2911
ESP32_custom.Connectivit	0	0	2644	0	0	2644	576	191	0	0	767
spi_flash_hal_iram.c.obj	0	0	0	2297	0	2297	0	0	0	0	2297
esp_flash_api.c.obj		20	0	0	2275	0	2295	80	947	0	0	3322
level-control.cpp.obj		0	0	2240	0	0	2240	3398	1148	0	0	4546
rtc_clk.c.obj			171	0	8	2053	0	2232	0	0	0	0	2224
opiram_psram.c.obj		640	0	10	1514	0	2164	0	0	0	0	2154
libTransportLayer.Sessio	0	0	2096	0	0	2096	5843	1676	0	0	7519
xtensa_vectors.S.obj		32	0	0	1504	424	1960	0	48	0	0	2008
rmt.c.obj			40	0	36	1778	0	1854	4550	1592	0	0	7960
spi_flash_chip_mxic_opi.	401	0	0	1440	0	1841	0	0	0	0	1841
pm.o				48	0	396	1307	0	1751	5472	228	0	0	7055
flash_mmap.c.obj		0	0	520	1186	0	1706	153	236	0	0	1575
spi_flash_hal_gpspi.c.ob	0	0	0	1699	0	1699	0	0	0	0	1699
color-control-server.cpp	0	0	1536	0	0	1536	6479	85	0	0	6564
dnssd.Resolver_ImplMinim	0	0	1448	0	0	1448	4447	1091	0	0	5538
heap_caps.c.obj			4	0	4	1381	0	1389	356	512	0	0	2253
ieee80211_sta.o			0	0	37	1350	0	1387	9936	985	1633	0	13904
wdt_hal_iram.c.obj		0	0	0	1357	0	1357	0	0	0	0	1357
dnssd.Advertiser_ImplMin	0	0	1320	0	0	1320	7498	1928	0	0	9426
dns.c.obj			0	0	1292	0	0	1292	1449	218	0	0	1667
esp_matter_core.cpp.obj		92	0	1193	0	0	1285	9069	3781	0	0	12942
coexist_scheme.o		1199	0	32	0	0	1231	1940	0	0	0	3139
ieee80211_output.o		2	0	12	1203	0	1217	6975	424	0	0	8604
spi_timing_config.c.obj		102	0	0	1112	0	1214	0	0	0	0	1214
alarm.c.obj			0	0	1208	0	0	1208	1324	560	0	0	1884
attribute-storage.cpp.ob	1120	0	6	0	0	1126	3414	266	0	0	4800
cpu_start.c.obj			1	0	5	1100	0	1106	57	437	0	0	1595
lld_adv.o			0	0	16	1063	0	1079	129	10	0	0	1202
xtensa_intr_asm.S.obj		1024	0	0	51	0	1075	0	0	0	0	1075
spi_flash_timing_tuning.	240	0	3	832	0	1075	0	0	0	0	1072
rwble.o				128	0	92	850	0	1070	676	100	0	0	1754
libaccess.ExampleAccessC	16	0	1044	0	0	1060	2255	270	0	0	2541
port.c.obj			0	0	24	1035	0	1059	157	601	0	0	1793
ieee80211_nvs.o			4	0	1048	0	0	1052	3938	928	0	0	4870
esp_adapter.c.obj		516	0	5	519	0	1040	1335	0	0	0	2370
provisioner_prov.c.obj		278	0	750	0	0	1028	14299	6113	0	0	20690
message.cpp.obj			0	0	1026	0	0	1026	210	23	0	0	233
memspi_host_driver.c.obj	360	0	0	658	0	1018	0	0	0	0	1018
~~~

jonsmirl, 2023-03-13:
I see now there are two different BLE in PSRAM options, one for BLE and one for BLE Mesh. I had been turning on BLE in PSRAM.  I just turned on the BLE Mesh in PSRAM option. No real difference, friend.c still 62KB.
~~~
   .vectors size:    1027 bytes
Used stat D/IRAM:  171845 bytes (   1083 remain, 99.4% used)
      .data size:   27053 bytes
      .bss  size:  144792 bytes
Used Flash size : 1397175 bytes
      .text     : 1121879 bytes
      .rodata   :  275040 bytes
Total image size: 1509430 bytes (.bin may be padded larger)
~~~

jonsmirl, 2023-03-13:
Look at this, I went in and set the LPN BLE Mesh defaults a lot lower.  BLE Mesh should be allocating those buffers from PSRAM and it isn't.
~~~
Total sizes:
Used static IRAM:   85202 bytes ( 277038 remain, 23.5% used)
      .text size:   84175 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:  108885 bytes (  64043 remain, 63.0% used)
      .data size:   23269 bytes
      .bss  size:   85616 bytes
Used Flash size : 1397099 bytes
      .text     : 1121803 bytes
      .rodata   :  275040 bytes
Total image size: 1505570 bytes (.bin may be padded larger)
~~~

dhrishi, 2023-03-16:
@maenkai Fixed with c8babee85ac3fc2c0df1eeac91a7abc7b932ab06. You can now allocate the esp-matter data model from PSRAM.

dhrishi, 2023-03-20:
I will close this as now the esp-matter data model now provides an option to use PSRAM.

closed 2023-03-20 by dhrishi