espressif/esp-matter #251: Matter Test-Harness

stefano055415, 2023-03-02:
I have a problem with TH version 2.6

If I load the PICS files the correct tests are not selected

Steps to reproduce it
Modify the project by uploading the Pics file:

[image]

I press on Run Test:

[image]

I create a new test:

[image]

But the selected tests do not match those of the pics file

what am I doing wrong thanks

this is the pics file

[On-Off Cluster Test Plan.xml.zip](https://github.com/espressif/esp-matter/files/10872949/On-Off.Cluster.Test.Plan.xml.zip)

PSONALl, 2023-03-03:
@stefano055415 
Can you reboot your test harness and then try again? follow these steps:
1. ./chip-certification-tool/scripts/stop.sh
2. ./chip-certification-tool/scripts/start.sh

stefano055415, 2023-03-03:
@PSONALl 

I installed test-harness on a raspberry pi4 and tried to reinstall it several times but with the same result I also tried to use version 2.5 but I have the same problem

stefano055415, 2023-03-03:
From project-chip repository, they suggested me to write on this repository 

[https://github.com/CHIP-Specifications/chip-certification-tool ](https://github.com/CHIP-Specifications/chip-certification-tool )

but being private I can't access it, does anyone know how I can request access?

dhrishi, 2023-03-04:
@stefano055415 chip-certification-tool is under CHIP-Specifications which is not open. If you/your company are a member of CSA then you can request them for access

closed 2023-03-06 by stefano055415