espressif/esp-matter #252: Updating temperature_measurement attribute value

MartinPArmstrong, 2023-03-02:
I've been able create a temperature_sesnor endpoint and set initial values like this:

```
temperature_sensor::config_t temp_sensor_config;
temp_sensor_config.temperature_measurement.measured_value = 27;
endpoint_t *endpoint1 = temperature_sensor::create(node, &temp_sensor_config, ENDPOINT_FLAG_NONE, temp_handle);
```
My question is wether there is a simple way to update the measured_value attribute from in other functions elswehere.

dhrishi, 2023-03-03:
@MartinPArmstrong Will this reference help: https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_driver.cpp#L58-L70 ? 

Here, we toggle the OnOff attribute of OnOff server cluster on a button press. Basically, `attribute::update` is the API.

MartinPArmstrong, 2023-03-03:
Thanks, just what i needed. I will now close the issue,

closed 2023-03-03 by MartinPArmstrong