espressif/esp-matter #254: ERROR when build example zigbee_birdge

githubworksonline, 2023-03-03:
esp-idf: main branch
esp-matter: main branch

after build esp-idf example esp_zigbee_rcp,  with same env, i build esp-matter example zigbee_birdge, however ,some ERROR occurred：

idf.py build

*********

[19/661] Building C object esp-idf/wpa_supplicant/CMakeFiles/__idf_wpa_supplicant.dir/src/ap/wpa_auth.c.obj**FAILED**: esp-idf/wpa_supplicant/CMakeFiles/__idf_wpa_supplicant.dir/src/ap/wpa_auth.c.obj 

**********
home/aaa/Documents/esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c: In function 'wpa_ap_join':

/home/aaa/Documents/esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:2547:17: error: implicit declaration of function 'esp_send_assoc_resp'; did you mean 'owe_process_assoc_resp'? [-Werror=implicit-function-declaration]

 2547 |             if (esp_send_assoc_resp(hapd, sta, bssid, resp, omit_rsnxe, subtype) != WLAN_STATUS_SUCCESS) {

      |                 ^~~~~~~~~~~~~~~~~~~

      |                 owe_process_assoc_resp

cc1: all warnings being treated as errors

[24/661] Building C object esp-idf/wpa_supplicant/CMakeFiles/__idf_wpa_supplicant.dir/src/eap_peer/eap_peap.c.objninja: build stopped: subcommand failed.

ninja failed with exit code 1, output of the command is in the /home/aaa/Documents/esp-matter/examples/zigbee_bridge/build/log/idf_py_stderr_output_12227 and /home/aaa/Documents/esp-matter/examples/zigbee_bridge/build/log/idf_py_stdout_output_12227

shubhamdp, 2023-03-04:
@githubworksonline zigbee_bridge example is tested with specific IDF commit 20949d444f as mentioned in the docs [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32h2/developing.html#getting-the-repositories). Can you try on that.

Also, there is no `main` branch for esp-idf, did you mean to say `master`?

githubworksonline, 2023-03-06:
> @githubworksonline zigbee_bridge example is tested with specific IDF commit 20949d444f as mentioned in the docs [...]

yeah, it's master branch.

thank u.

closed 2023-03-06 by githubworksonline