espressif/esp-matter #256: Unable to modify zap_light using zap-tool

markusp00, 2023-03-03:
I am unable to modify the zap_light example using the zap-tool. I suspect it is caused by a wrong version of some software.

This is my current setup:
- ESP32-S3-DevKitC-1
- ESP-IDF v4.4.3
- npm v9.6.0
- node v16.13.0
- electron v18.0.0
- zap v2022.12.18

Steps to replicate:
- Copy and paste the zap_light example and rename it zap_temp
- Place the zap-directory in esp-matter/connectedhomeip/connectedhomeip/third_party
- Run `./scripts/tools/zap/run_zaptool.sh`
- Open light.zap
- Add a Matter endpoint
- Generate files in zap-generated folder
- Build the example using idf.py build

Results:
- Notification in zap-tool

> 

160 | UPGRADE WARNING | Resource "${key}" not found among the context resources. Check your template.json file. You may need a Matter SDK upgrade. | 1
-- | -- | -- | --

- Console log:

> [1222/1326] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/src/app/EventManagement.cpp.obj
/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/src/app/EventManagement.cpp: In member function 'CHIP_ERROR chip::app::EventManagement::LogEventPrivate(chip::app::EventLoggingDelegate*, const chip::app::EventOptions&, chip::EventNumber&)':
/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/src/app/EventManagement.cpp:410:23: warning: 'writer' is used uninitialized in this function [-Wuninitialized]
     CircularTLVWriter checkpoint = writer;
                       ^~~~~~~~~~
[1263/1326] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/src/app/util/attribute-storage.cpp.obj
FAILED: esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/src/app/util/attribute-storage.cpp.obj 
ccache /home/vboxuser/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -I/home/vboxuser/esp-matter/examples/zap_temp/build/config -I/home/vboxuser/esp-matter/components/esp_matter -I/home/vboxuser/esp-matter/components/esp_matter/zap_common -I/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/zzz_generated/app-common -I/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/src -I/home/vboxuser/esp-matter/examples/zap_temp/main -I/home/vboxuser/esp-idf/components/newlib/platform_include -I/home/vboxuser/esp-idf/components/freertos/include -I/home/vboxuser/esp-idf/components/freertos/include/esp_additions/freertos -I/home/vboxuser/esp-idf/components/freertos/port/xtensa/include -I/home/vboxuser/esp-idf/components/freertos/include/esp_additions -I/home/vboxuser/esp-idf/components/esp_hw_support/include -I/home/vboxuser/esp-idf/components/esp_hw_support/include/soc -I/home/vboxuser/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/home/vboxuser/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/vboxuser/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/home/vboxuser/esp-idf/components/heap/include -I/home/vboxuser/esp-idf/components/log/include -I/home/vboxuser/esp-idf/components/lwip/include/apps -I/home/vboxuser/esp-idf/components/lwip/include/apps/sntp -I/home/vboxuser/esp-idf/components/lwip/lwip/src/include -I/home/vboxuser/esp-idf/components/lwip/port/esp32/include -I/home/vboxuser/esp-idf/components/lwip/port/esp32/include/arch -I/home/vboxuser/esp-idf/components/soc/include -I/home/vboxuser/esp-idf/components/soc/esp32s3/. -I/home/vboxuser/esp-idf/components/soc/esp32s3/include -I/home/vboxuser/esp-idf/components/hal/esp32s3/include -I/home/vboxuser/esp-idf/components/hal/include -I/home/vboxuser/esp-idf/components/hal/platform_port/include -I/home/vboxuser/esp-idf/components/esp_rom/include -I/home/vboxuser/esp-idf/components/esp_rom/include/esp32s3 -I/home/vboxuser/esp-idf/components/esp_rom/esp32s3 -I/home/vboxuser/esp-idf/components/esp_common/include -I/home/vboxuser/esp-idf/components/esp_system/include -I/home/vboxuser/esp-idf/components/esp_system/port/soc -I/home/vboxuser/esp-idf/components/esp_system/port/public_compat -I/home/vboxuser/esp-idf/components/xtensa/include -I/home/vboxuser/esp-idf/components/xtensa/esp32s3/include -I/home/vboxuser/esp-idf/components/driver/include -I/home/vboxuser/esp-idf/components/driver/esp32s3/include -I/home/vboxuser/esp-idf/components/esp_pm/include -I/home/vboxuser/esp-idf/components/esp_ringbuf/include -I/home/vboxuser/esp-idf/components/efuse/include -I/home/vboxuser/esp-idf/components/efuse/esp32s3/include -I/home/vboxuser/esp-idf/components/vfs/include -I/home/vboxuser/esp-idf/components/esp_wifi/include -I/home/vboxuser/esp-idf/components/esp_event/include -I/home/vboxuser/esp-idf/components/esp_netif/include -I/home/vboxuser/esp-idf/components/esp_eth/include -I/home/vboxuser/esp-idf/components/tcpip_adapter/include -I/home/vboxuser/esp-idf/components/esp_phy/include -I/home/vboxuser/esp-idf/components/esp_phy/esp32s3/include -I/home/vboxuser/esp-idf/components/esp_ipc/include -I/home/vboxuser/esp-idf/components/app_trace/include -I/home/vboxuser/esp-idf/components/esp_timer/include -I/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/src/platform/ESP32 -I/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/src/platform/OpenThread -I/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/src/include -I/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/src/lib -I/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/examples/platform/esp32 -I/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/third_party/nlassert/repo/include -I/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/third_party/nlio/repo/include -I/home/vboxuser/esp-matter/examples/zap_temp/build/esp-idf/chip/src/include -I/home/vboxuser/esp-matter/examples/zap_temp/build/esp-idf/chip/include -I/home/vboxuser/esp-matter/examples/zap_temp/build/esp-idf/chip/gen/include -I/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/config/esp32 -I/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/config/esp32/../.. -I/home/vboxuser/esp-idf/components/bt/common/osi/include -I/home/vboxuser/esp-idf/components/bt/include/esp32s3/include -I/home/vboxuser/esp-idf/components/bt/common/api/include/api -I/home/vboxuser/esp-idf/components/bt/common/btc/profile/esp/blufi/include -I/home/vboxuser/esp-idf/components/bt/common/btc/profile/esp/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/port/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ias/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/lls/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/tps/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/util/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include -I/home/vboxuser/esp-idf/components/bt/host/nimble/esp-hci/include -I/home/vboxuser/esp-idf/components/nvs_flash/include -I/home/vboxuser/esp-idf/components/spi_flash/include -I/home/vboxuser/esp-idf/components/mbedtls/port/include -I/home/vboxuser/esp-idf/components/mbedtls/mbedtls/include -I/home/vboxuser/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/vboxuser/esp-matter/components/esp_matter_console -I/home/vboxuser/esp-idf/components/app_update/include -I/home/vboxuser/esp-idf/components/bootloader_support/include -I/home/vboxuser/esp-matter/examples/zap_temp/managed_compon [... cut ...] ype, RemoteLockOpHandler, const chip::Optional<chip::Span<const unsigned char> >&)':
/home/vboxuser/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/door-lock-server/door-lock-server.cpp:3348:38: warning: '*((void*)& pinCredIdx +2)' may be used uninitialized in this function [-Wmaybe-uninitialized]
         foundCred[0].credentialIndex = pinCredIdx.Value();
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

jonsmirl, 2023-03-05:
Zap has to be matched exactly to the CHIP release. CHIP 1.0.0.2 works with this specfic zap-cli (other releases won't work) https://github.com/project-chip/zap/releases/tag/v2022.12.20-nightly

dhrishi, 2023-03-06:
@markusp00 As mentioned by @jonsmirl the ZAP should match the v1.0.0.2 tag of upstream connectedhomeip. Please give it a try and update in case of any issues

markusp00, 2023-03-07:
I have now tried multiple releases of ZAP without any luck. I encounter the same problems as the other members of my project group. We have decided to abandon ZAP and instead use the non-zap esp-matter platform.

closed 2023-03-07 by markusp00

Jayesh19101999, 2023-04-05:
Even i am facing the same error @dhrishi while building the lighning app when i enter the command idf.py build i am not able to build . I am getting these logs .

einfochips@PUNECPU0517:~/Desktop/matter/connectedhomeip/examples/lighting-app/esp32$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/einfochips/Desktop/matter/connectedhomeip/examples/lighting-app/esp32/build
Executing "ninja all"...
[1/41] Performing build step for 'chip_gn'
ninja: no work to do.
[2/39] Performing build step for 'bootloader'
[1/1] cd /home/einfochips/Desktop/matter/connectedhomeip/examples/lighting-app/esp32/build/bootloader/esp-idf/esptool_py && /home/einfochips/Desktop/matter/connectedhomeip/.environment/pigweed-venv/bin/python /home/einfochips/Desktop/ESP32/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/einfochips/Desktop/matter/connectedhomeip/examples/lighting-app/esp32/build/bootloader/bootloader.bin
Bootloader binary size 0x6330 bytes. 0xcd0 bytes (11%) free.
[4/37] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/__/third_party/connectedhomeip/src/app/util/attribute-storage.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/__/third_party/connectedhomeip/src/app/util/attribute-storage.cpp.obj
/home/einfochips/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -I/home/einfochips/Desktop/matter/connectedhomeip/examples/lighting-app/esp32/build/config -I/home/einfochips/Desktop/matter/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/zzz_generated/lighting-app -I/home/einfochips/Desktop/matter/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/examples/providers -I/home/einfochips/Desktop/matter/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/examples/lighting-app/lighting-common/include -I/home/einfochips/Desktop/matter/connectedhomeip/examples/lighting-app/esp32/main/include -I/home/einfochips/Desktop/matter/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/examples/platform/esp32 -I/home/einfochips/Desktop/matter/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app -I/home/einfochips/Desktop/matter/connectedhomeip/examples/lighting-app/esp32/build/gen/app-zapgen/zapgen/app-templates -I/home/einfochips/Desktop/ESP32/esp-idf/components/newlib/platform_include -I/home/einfochips/Desktop/ESP32/esp-idf/components/freertos/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/freertos/include/esp_additions/freertos -I/home/einfochips/Desktop/ESP32/esp-idf/components/freertos/port/xtensa/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/freertos/include/esp_additions -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_hw_support/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_hw_support/include/soc -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_hw_support/port/esp32/. -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_hw_support/port/esp32/private_include -I/home/einfochips/Desktop/ESP32/esp-idf/components/heap/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/log/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/lwip/include/apps -I/home/einfochips/Desktop/ESP32/esp-idf/components/lwip/include/apps/sntp -I/home/einfochips/Desktop/ESP32/esp-idf/components/lwip/lwip/src/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/lwip/port/esp32/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/lwip/port/esp32/include/arch -I/home/einfochips/Desktop/ESP32/esp-idf/components/soc/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/soc/esp32/. -I/home/einfochips/Desktop/ESP32/esp-idf/components/soc/esp32/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/hal/esp32/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/hal/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/hal/platform_port/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_rom/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_rom/include/esp32 -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_rom/esp32 -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_common/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_system/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_system/port/soc -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_system/port/public_compat -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp32/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/xtensa/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/xtensa/esp32/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/driver/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/driver/esp32/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_pm/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_ringbuf/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/efuse/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/efuse/esp32/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/vfs/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/esp_wifi/include -I/home/einfochips/Desktop/ESP32/esp-idf/components/ [... cut ...]
/home/einfochips/Desktop/matter/connectedhomeip/examples/lighting-app/esp32/third_party/connectedhomeip/src/app/util/attribute-storage.cpp:85:71: note: in expansion of macro 'GENERATED_ATTRIBUTES'
 constexpr const EmberAfAttributeMetadata generatedAttributes[]      = GENERATED_ATTRIBUTES;
                                                                       ^~~~~~~~~~~~~~~~~~~~
[9/37] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/__/third_party/connectedhomeip/src/app/clusters/general-diagnostics-server/general-diagnostics-server.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1