espressif/esp-matter #258: Build failed due to small App partition

edWin-m, 2023-03-04:
@dhrishi  My source failed to complete the `ninja` build step after I updated the esp-matter version to commit [b8f0e22](https://github.com/espressif/esp-matter/commit/b8f0e2237094dd99e78811c6429fcbbda289d1c7). Below is a snippet of the build log showing the failure.

```
FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size /Users/**/build/esp-idf/esptool_py/CMakeFiles/app_check_size 
cd /Users/**/build/esp-idf/esptool_py && /opt/homebrew/opt/python@3.10/bin/python3.10 /Users/edwin/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /Users/**/build/partition_table/partition-table.bin /Users/**/build/lightstrip.bin
Error: app partition is too small for binary lightstrip.bin size 0x10b380:
  - Part 'factory' 0/0 @ 0x10000 size 0x100000 (overflow 0xb380)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

After updating the esp-matter repo, I modified the `partitions.csv` to match the one used in the light example. *I have also tried the example and it succeeded.*

[image]

It looks like this can be fixed by updating the offsets in the `partitions.csv` If that's the case please advise on which sector I should add the `0xb380` offset to.

shubhamdp, 2023-03-04:
@edWin-m error says that app partition has overflown, error also specifies the partition label, `factory` at `0x10000`. As far I as remember that is the very basic partition table used in idf examples. If you are using that then you will have to increase the size of partition with label `factory`.

If you wish to use the custom partition table as per esp-matter examples then you may be missing some configuration options as specified in `sdkconfig.defaults`. https://github.com/espressif/esp-matter/blob/b8f0e2237094dd99e78811c6429fcbbda289d1c7/examples/light/sdkconfig.defaults#L19-L22

For more details you can refer idf docs which talks about partition table [subtypes](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#subtype).

edWin-m, 2023-03-06:
@shubhamdp Thanks.  The `CONFIG_PARTITION_TABLE_CUSTOM` option in my `sdkconfig` was not set.

closed 2023-03-06 by edWin-m