espressif/esp-matter #259: Color control cluster, command 0x47

jonsmirl, 2023-03-05:
This makes my color control cluster:
~~~
    color_control::create(epLight, &ext_config.color_control, CLUSTER_FLAG_SERVER | CLUSTER_FLAG_CLIENT,
                          color_control::feature::color_temperature::get_id() | color_control::feature::xy::get_id() | color_control::feature::color_loop::get_id());
~~~
turning on the two features seems to be missing a check for duplicating command 0x0047
~~~
W (1113) esp_matter_core: Command 0x0047 on cluster 0x0300 already exists. Not creating again.
~~~

PSONALl, 2023-03-06:
@jonsmirl This is not a missing check, during creating a command, it is getting checked at a lower level if the command already exists, and if it is existing already then it won't be created

jonsmirl, 2023-03-06:
This doesn't hurt anything, but in other places in the code you suppress this error message when it is internally triggered.

closed 2023-03-06 by jonsmirl