espressif/esp-matter #260: ModeSelect cluster support missing

jonsmirl, 2023-03-05:
esp-matter support for the ModeSelect cluster support is missing.

MartinPArmstrong, 2023-03-06:
I've noticed this myself, since we need this cluster for a project. If support isn't added we'll probably have to add it ourselves.

dhrishi, 2023-03-07:
@jonsmirl @MartinPArmstrong We will add the support of ModeSelect cluster soon.

jonsmirl, 2023-03-07:
Dhrishi will know better, but I suspect it will be done within week.

dhrishi, 2023-03-08:
@MartinPArmstrong It should be done within this week

jadhavrohit924, 2023-03-09:
@jonsmirl @MartinPArmstrong We are working on it. It will take a bit longer to add Mode Support in component.

MartinPArmstrong, 2023-03-09:
@jadhavrohit924 Can we perhaps expect it next week instead then?

jadhavrohit924, 2023-03-09:
@MartinPArmstrong It will be done ASAP!!

markusp00, 2023-03-21:
@jadhavrohit924 Any updates on the issue?

jonsmirl, 2023-03-21:
Did CHIP 1.0 ship without ANY vendor implementing ModeSelect?

jonsmirl, 2023-05-30:
What is update on this? I don't see mode_select in esp_matter yet.

jadhavrohit924, 2023-05-31:
@jonsmirl It will be added soon.

itavero, 2023-06-06:
I was trying to add it manually, but got stuck when trying to define the Supported Modes attribute (not sure how to create the list of ModeOptionStruct instances).

jonsmirl, 2023-06-19:
Any progress on this?

jadhavrohit924, 2023-07-03:
Hi @jonsmirl @itavero @markusp00 @MartinPArmstrong, sorry for the delay. It was blocked due to some dependency. We are unblocked now. Support of mode select cluster is in internal review, it will be merged soon :)

jadhavrohit924, 2023-07-07:
Its merged in commit [43fa905](https://github.com/espressif/esp-matter/commit/43fa90571765e6caca3f5fdd655ed54a585d80cf) please pull the latest code. Here is the documentation for link https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#mode-select Please close the issue.

closed 2023-07-07 by jonsmirl