espressif/esp-matter #262: Application stuck during commissioning

Clcudio10, 2023-03-06:
Since yesterday the example "light" (but also the others) stuck during the commission with Alexa. I leave the log below. After that expires the fail safe timer and restart to advertise.
The fact is for a month i've never got a single time this problem instead now i can't commission a new device and also Alexa see the device that i have already add before yesterday as "device doesn't respond". What could be the problem?

Log:

`I (30) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (30) boot: compile time 15:02:59
I (30) boot: chip revision: 3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (70) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (85) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (93) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (100) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (108) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (115) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (123) boot: End of partition table
I (127) boot: No factory image, trying OTA 0
I (132) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=2f0c8h (192712) map
I (171) esp_image: segment 1: paddr=0004f0f0 vaddr=3fc91c00 size=00f28h (  3880) load
I (172) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=f6658h (1009240) map
I (334) esp_image: segment 3: paddr=00146680 vaddr=3fc92b28 size=027e8h ( 10216) load
I (336) esp_image: segment 4: paddr=00148e70 vaddr=40380000 size=11a9ch ( 72348) load
I (353) esp_image: segment 5: paddr=0015a914 vaddr=50000010 size=00010h (    16) load
I (359) boot: Loaded app from partition at offset 0x20000
I (409) boot: Set actual ota_seq=1 in otadata[0]
I (409) boot: Disabling RNG early entropy source...
I (420) cpu_start: Pro cpu up.
I (429) cpu_start: Pro cpu start user code
I (429) cpu_start: cpu freq: 160000000
I (429) cpu_start: Application information:
I (432) cpu_start: Project name:     light
I (436) cpu_start: App version:      v1.0
I (441) cpu_start: Compile time:     Mar  6 2023 15:02:54
I (447) cpu_start: ELF file SHA256:  8eacba2061768fed...
I (453) cpu_start: ESP-IDF:          v4.4.3-dirty
I (459) heap_init: Initializing. RAM available for dynamic allocation:
I (466) heap_init: At 3FCA6070 len 000366A0 (217 KiB): DRAM
I (472) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (479) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (486) spi_flash: detected chip: generic
I (490) spi_flash: flash io: dio
I (495) sleep: Configure to isolate all GPIO pins in sleep state
I (500) sleep: Enable automatic switching of GPIO sleep configuration
I (508) coexist: coexist rom version 9387209
I (513) cpu_start: Starting scheduler.
I (521) led_driver_ws2812: Initializing light driver
I (521) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (541) app_main: Light created with endpoint_id 1
W (551) esp_matter_core: Command 0x0047 on cluster 0x0300 already exists. Not creating again.
I (551) pp: pp rom version: 9387209
I (551) net80211: net80211 rom version: 9387209
I (571) wifi:wifi driver task: 3fcb1f08, prio:23, stack:6656, core=0
I (571) system_api: Base MAC address is not set
I (571) system_api: read default base MAC address from EFUSE
I (581) wifi:wifi firmware version: 8cb87ff
I (581) wifi:wifi certification version: v7.0
I (581) wifi:config NVS flash: enabled
I (591) wifi:config nano formating: disabled
I (591) wifi:Init data frame dynamic rx buffer num: 32
I (601) wifi:Init management frame dynamic rx buffer num: 32
I (601) wifi:Init management short buffer num: 32
I (611) wifi:Init dynamic tx buffer num: 32
I (611) wifi:Init static tx FG buffer num: 2
[... 464 lines cut ...]
I (59521) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (59521) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 1 ) indicate = 0
I (59531) chip[DL]: CHIPoBLE unsubscribe received
I (59541) chip[BLE]: Releasing end point's BLE connection back to application.
I (60511) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
I (60551) chip[DL]: IP_EVENT_GOT_IP6
I (60551) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fe80:0000:0000:0000:86f7:03ff:fe12:c760
I (60561) app_main: Interface IP Address changed
I (60561) chip[DIS]: Updating services using commissioning mode 0
I (60571) chip[DIS]: CHIP minimal mDNS started advertising.
I (60581) chip[DIS]: Advertise operational node 9F75C201E1057317-00F03DC4F270B430
I (60581) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 9F75C201E1057317-00F03DC4F270B430.
I (60591) chip[DIS]: mDNS service published: _matter._tcp
W (62181) wifi:<ba-add>idx:0 (ifx:0, f0:9f:c2:7d:a8:79), tid:0, ssn:2, winSize:64
I (62551) esp_netif_handlers: sta ip: 192.168.1.39, mask: 255.255.255.0, gw: 192.168.1.1
I (62551) chip[DL]: IP_EVENT_STA_GOT_IP
I (62551) chip[DL]: IPv4 address changed on WiFi station interface: 192.168.1.39/255.255.255.0 gateway 192.168.1.1
I (62561) chip[DL]: IPv4 Internet connectivity ESTABLISHED
I (62571) app_main: Interface IP Address changed
I (62571) chip[DIS]: Updating services using commissioning mode 0
I (62581) chip[DIS]: CHIP minimal mDNS started advertising.
I (62591) chip[DIS]: Advertise operational node 9F75C201E1057317-00F03DC4F270B430
I (62601) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 9F75C201E1057317-00F03DC4F270B430.
I (62611) chip[DIS]: mDNS service published: _matter._tcp
I (62611) ROUTE_HOOK: Hook already installed on netif, skip...`

dhrishi, 2023-03-07:
@Clcudio10 The log above shows that the device is connected to the IP network and is now waiting for CASE Sigma1 message from the commissioner (Alexa). If you have used commissioning using Alexa with the same AP before, then this should be a transient failure.

Clcudio10, 2023-03-07:
Thank you, so what i miss in the procedure? If i test the same ESP device every time with different example type (light, light switch,...), but also with the same type, i only need to erase the device FLASH and remove that device from Alexa app side, or there are other things that i forget to do?

dhrishi, 2023-03-08:
@Clcudio10 There's nothing that you have missed. I wanted to say that it is some problem on the commissioner side. Are you able to commission your device with the host-based chip-tool or any other ecosystem app?

Clcudio10, 2023-03-08:
The fact is that:
- Before February in my tests i was cabable of commission and use the devices only with the CHIP-tool;
- Then, from one day to another, the devices started to run also with Alexa but never with the other ecosystems; consider i only have a Echo Dot and a Nest Mini to try, but Google Home said that "the device is not a certificated Matter device" (a different kind of problem, but OK...), so Alexa at this moment is my only option. In any case, for a month i've commissioned and used always the same two ESP32C3-DevkitC-02 with different example firmwares (light that is OK, light_switch and generic_switch that are not compatible device with Alexa yet) and never had i problem until this monday when i can't complete the commission procedure with the light example like i shown.
-  Since yesterday morning the commission seems to be OK again.

So i start to think that maybe there are sort of mantainance in Alexa Matter side that i cant't see or in any case something that doesn't depend on my work.

dhrishi, 2023-03-08:
>Since yesterday morning the commission seems to be OK again.

Good to know.

Can you close this issue if it works now?

Clcudio10, 2023-03-08:
Yes, thank you.

closed 2023-03-08 by Clcudio10