espressif/esp-matter #263: Endpoint names in apps like Google Home

MartinPArmstrong, 2023-03-06:
When commissioning my application with multiple endpoints into Google Home all the endpoints get the name TEST_PRODUCT as shown in the picture below. Is there a way to change these names in the code? Or is this a Google Home related issue.

[image]

MartinPArmstrong, 2023-03-06:
And If this is simply tied to manufacturer information, is there a way to have different name for the endpoints?

jonsmirl, 2023-03-07:
For development I am setting the device characteristics like this:
~~~
message(STATUS "Vendor ID: " ${CONFIG_DEVICE_VENDOR_ID})
message(STATUS "Vendor Name: " ${CONFIG_DEVICE_VENDOR_NAME})
message(STATUS "Product ID: " ${CONFIG_DEVICE_PRODUCT_ID})
message(STATUS "Product Name: " ${CONFIG_DEVICE_PRODUCT_NAME})
message(STATUS "SW Version String: " ${CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING})
idf_build_set_property(COMPILE_OPTIONS "-DCHIP_DEVICE_CONFIG_DEVICE_VENDOR_ID=${CONFIG_DEVICE_VENDOR_ID}" APPEND)
idf_build_set_property(COMPILE_OPTIONS -DCHIP_DEVICE_CONFIG_DEVICE_VENDOR_NAME="${CONFIG_DEVICE_VENDOR_NAME}" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-DCHIP_DEVICE_CONFIG_DEVICE_PRODUCT_ID=${CONFIG_DEVICE_PRODUCT_ID}" APPEND)
idf_build_set_property(COMPILE_OPTIONS -DCHIP_DEVICE_CONFIG_DEVICE_PRODUCT_NAME="${CONFIG_DEVICE_PRODUCT_NAME}" APPEND)
~~~

There are the Fixed Label and User Label clusters for setting names. But... I don't know what Google will do with them on random endpoints.

dhrishi, 2023-03-07:
@MartinPArmstrong 

>When commissioning my application with multiple endpoints into Google Home all the endpoints get the name TEST_PRODUCT as shown in the picture below.

This is the product name that is set on the device.

> Is there a way to change these names in the code?

Yes. You can use the [mfg_tool](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool#manufacturing-partition-generator-utility) to configure the product name of your choice. Once you setup the tool, please use `./tools/mfg_tool/mfg_tool.py --help` to see the available options.

closed 2023-03-15 by MartinPArmstrong