espressif/esp-matter #264: Example for reading attributes

jonsmirl, 2023-03-07:
I figured this out already, but none of the examples illustrate ATTRIBUTE_FLAG_OVERRIDE and how to implement READ for an attribute. For example a temp sensor and MeasuredValue.

jonsmirl, 2023-03-07:
This will set the override flag. Now how to do it after a resume reconstructs the device?
~~~
        if (device_type == ESP_MATTER_TEMPERATURE_SENSOR_DEVICE_TYPE_ID) {
            cluster_t *cluster = cluster::get(bridged_device->dev->endpoint, TemperatureMeasurement::Id);
            attribute_t *attrib = esp_matter::attribute::get(cluster, TemperatureMeasurement::Attributes::MeasuredValue::Id);
            esp_matter::attribute::set_override_callback(attrib, NULL);
        }
~~~

jonsmirl, 2023-03-07:
I'm not finding any obvious callbacks letting me modify a resumed device.

dhrishi, 2023-03-07:
@jonsmirl Currently, the read for attributes that are managed internally are handled by the Matter stack. For the attributes in the esp-matter data model, the current value is given by the esp-matter.

For your scenario, the BLE Mesh publish-subscribe should be used, Whenever there is a change in the value of the BLE mesh sensor, the bridge is notified and it should then update its attribute value. We are working on providing a sample implementation for publish-subscribe for the BLE Mesh bridge.

jonsmirl, 2023-03-07:
So when a sensor is resumed, how do I get my subscribe hooked back up to the attribute?

jonsmirl, 2023-03-07:
In the example, please turn on Friend support in the BLE bridge and LPN on the mesh sensor.  These sensors are going to be battery power and have LPN turned on.  Set the polling interval to something like two minutes so that you can easily see how it impacts things.

dhrishi, 2023-03-10:
As the question about reading the attribute is handled, I will close this. And tag tha above comment in the open issue for bridge

closed 2023-03-10 by dhrishi