espressif/esp-matter #265: Unable to setup temperature_measurement cluster and view temperature values on AppleHome

SimonMSN, 2023-03-07:
We are working on a multi endpoint device with two lights and one button. 
For testing purposes, we wanted to add the temperature_measurement cluster to the switch endpoint to test the ESP32's temperature sensor.
As stated in the docs (https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#defining-your-own-data-model), a cluster can be added to an existing endpoint by passing the endpoint (sw_endpoint) to the cluster's `create()` function:

```
// create a switch endpoint
generic_switch::config_t sw_config;
endpoint_t *sw_endpoint = generic_switch::create(node, &sw_config, ENDPOINT_FLAG_NONE, button_handle);

// add temperature measurement cluster to switch endpoint
cluster::temperature_measurement::config_t temperature_measurement_config;
temperature_measurement_config.measured_value = 22;
cluster_t *cluster = cluster::temperature_measurement::create(sw_endpoint, &temperature_measurement_config, CLUSTER_FLAG_SERVER);
```

We then were able to succesfully commission the device using an iPhone 14 and a Homepod. Both lights (1x dimmable light and dimmable color light) can be controlled without any issues. The switch endpoint is also working. 

But on the Home App we can't see any field, indicating the measured temperature of 22 °C. Our understanding was, that a temperature_measurement cluster can be added to any endpoint. Although it may not make sense to add temperature measurement functionality to a switch, but that's another topic.

We also tested a dedicated endpoint, using a TemperatureSensor endpoint. With the `chip tool`, we checked that both mandatory clusters `Identify` and `temperature-measurement` were present on the TemperatureSensor endpoint. But we had no success to display the measured temperature values on Apple Home.

Did we setup the endpoint and cluster correctly, or are there currently some bugs on AppleHome or temperature_measurement cluster?

jonsmirl, 2023-03-07:
The Matter spec does indeed allow any cluster on any endpoint. However, AppleHome does not support that. Neither do Google or Amazon.

What might work is to make a dedicated Temperature Sensor endpoint with the Temperature Sensor device_type.  Apple claims to support that. https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md

Check using one of the examples in CHIP, Apple tests with those.
https://github.com/project-chip/connectedhomeip/tree/master/examples/temperature-measurement-app/esp32

Let me know if works, I have not checked IOS lately.

SimonMSN, 2023-03-07:
Thanks for your answer jon

Do you know this from experience, or did you found any docs about what cluster on which endpoint is supported? Would be interesting to see what can be mixed.

I'll check your links and see if i can manage to properly setup the TemperatureSensor device. We currently use the esp-matter examples, since the examples from project-chip repo seem to be more complex to comprehend.

jonsmirl, 2023-03-07:
Apple/Google/Amazon are only doing to support the device types in the Device Library specification.

However, that in no way restricts what you do in your own apps.

stage-01, 2023-03-08:
Also if you want a temperature of 22 degrees celsius the value you put in has to be 100 times higher. So for your example you'd have to put in a value of 2200 instead of 22.

SimonMSN, 2023-03-09:
Thank you stage-01 and PSONALI for your inputs. 
It's good to read confirmation about this issue. 

Would be great to hear a feedback on this @PSONALl, in case you get a response from Apple ;)

SimonMSN, 2023-03-09:
> The Matter spec does indeed allow any cluster on any endpoint. However, AppleHome does not support that. Neither do Google or Amazon.
>
> What might work is to make a dedicated Temperature Sensor [...]

Unfortunately we're testing with the esp-matter repository instead of the project-chip repository. I can not confirm that I have time to check out the example from the chip repository. But I will give feedback if I am able to do so.

jonsmirl, 2023-03-13:
There is a copy of the CHIP tree inside esp-matter. checkout esp-matter/connecthomeip/connectedhomeip/.... It is a fully functioning copy.

SimonMSN, 2023-03-13:
I checked the temperature-app example from project-chip. This is working, although the reading shows always 0.0°C. But that's not a problem. Once I understand the program flow, this can be changed/fixed.

I went through main.cpp in the temperature-app example from project-chip, but I don't understand at which lines the node with its endpoint(s) gets configured, i.e., making it a temperature sensor node. I suspect the ZAP tool is used to generate the .matter and .zap file and then, these two files are used in the code to build the node. But I haven't figured out yet where.
Contrary, the examples from the esp-matter repo are very clear on how to setup a device with endpoints as well as accessing and/or modifying it's attributes. These examples can be extended easily.

Thank you for the help so far.

jonsmirl, 2023-03-13:
Something like this:
~~
scripts/tools/zap/run_zaptool.sh examples/temperature-measurement-app/esp32/main/temperature-measurement.zap
~~~

PSONALl, 2023-03-14:
@SimonMSN In project-chip endpoint configurations after running zap-tool are saved at `zzz_generated/temperature-measurement-app/zap-generated/endpoint_config.h`
The zap-tool can be run using the command:
```
scripts/tools/zap/run_zaptool.sh examples/temperature-measurement-app/esp32/main/temperature-measurement.zap
```

It will generate zap configurations and save them to the zap-generated folder at the path where temperature-measurement.zap is present instead of saving it directly to `zzz_generated/temperature-measurement-app/zap-generated`
I am trying out temperature measuremet app from project-chip on my end, I will check and let you know about the fix in the code

PSONALl, 2023-03-15:
@SimonMSN We have identified the fix for this issue, It will be maid available soon!

dhrishi, 2023-03-16:
Fixed with bcafaaab67a41246c077937abb2129d2715902d4. Please give it a try

closed 2023-03-16 by dhrishi