espressif/esp-matter #269: Added successfully but this function is not called back.

shun-dgut, 2023-03-10:
Added successfully but this function is not called back.

[image]

shun-dgut, 2023-03-10:
commit id ： f065091da

shun-dgut, 2023-03-10:
> @shun-dgut Can you please provide details? What does the below mean:
> 
> > Added successfully but this function is not called back.
> 
> Did you successfully commission the device and still not seeing kCommissioningComplete event in the app?

can be seen，But can't see the print in main.  esp_matter::start(app_event_cb);
[image]

shubhamdp, 2023-03-10:
@shun-dgut can you please share the complete logs. Also, can you please check if the event callback is being passed in to `esp_matter::start();` API.

shun-dgut, 2023-03-10:
> @shun-dgut can you please share the complete logs. Also, can you please check if the event callback is being passed in to `esp_matter::start();` API.

Make sure it is called. Is it the code in the picture below not PostEvent

[image]

shubhamdp, 2023-03-10:
@shun-dgut Complete logs would help.

PlatformManager calls the [PostEvent](https://github.com/project-chip/connectedhomeip/blob/v1.0.0.2/src/include/platform/PlatformManager.h#L423)

Above code in commissioning window manager is similar to what you have in app_main, CommissioningWindowManager also registers an event handler with platform manager.

So, I suspect it failed to register the event handler or post event may have failed.

shubhamdp, 2023-03-10:
@shun-dgut Please check the below, I tried `examples/generic_switch` on the SHA you mentioned and I can see the print in callback just fine.

[image]

shun-dgut, 2023-03-10:
Sorry，It `s my mistake