espressif/esp-matter #270: Devices not adding to Google Home

SYESHASWINI, 2023-03-10:
I'm trying to add devices to google home, but it is getting failed.

Board: esp32c3
Google Home version: 2.65.1.6
Nest Hub version(GUIK2 2nd gen): 1.56.324896

dhrishi, 2023-03-10:
@SYESHASWINI I would request you to provide sufficient information with the bug report. Can you please upload (not paste) the device side logs?

Also, can you confirm, if it worked anytime before? Have you registered the project in the Google developer console as mentioned in the table `Table: Vendor ID (VID) and device type pairing scenarios` [here](https://developers.home.google.com/matter/get-started)

SYESHASWINI, 2023-03-10:
No, I didn't registered the project with google developer console. It is required to test the example projects also.
Attaching the error log.
[GH.txt](https://github.com/espressif/esp-matter/files/10942475/GH.txt)

dhrishi, 2023-03-10:
@SYESHASWINI As per the link above, you are required to register the project for test devices. Please give it a try and it should work

closed 2023-03-10 by dhrishi