espressif/esp-matter #271: Ethernet and Wi-Fi support?

JonathanWitthoeft, 2023-03-10:
Does Matter support both Ethernet and WiFi concurrently? I see that if Ethernet Telemetry is enabled, Wi-Fi is not supported:

```
#if CONFIG_ENABLE_ETHERNET_TELEMETRY
#define CHIP_DEVICE_CONFIG_ENABLE_WIFI 0
#define CHIP_DEVICE_CONFIG_ENABLE_ETHERNET 1
```

dhrishi, 2023-03-11:
@JonathanWitthoeft As per the Matter specification only one of Matter over Wi-Fi, Thread or Ethernet can be supported. Please check `Features` of `Network Commissioning Cluster`, section `11.8.5` of v1.0 Matter core specification.

jonsmirl, 2023-03-13:
Once Matter is up and running it does not care about Wi-Fi, Thread or Ethernet since that is hidden behind IP networking.  My desktops functions as a router onto all three and Matter works fine on it.  For commissioning you have to pick one, but after you are commissioned you can use all three. To get this telemetry info you'd have to open the correct interface (not covered by the spec). Probably varies with Wifi STA or AP mode too.

Or is this a case of a device with both Wifi and Ethernet support where you have to chose one? I haven't consider how that case would work. Probably need some code of your own to detect that Ethernet was plugged in and to shutdown BLE/Wifi.

JonathanWitthoeft, 2023-03-13:
Yes @jonsmirl I am developing an update for a product to enable Matter support and network provisioning and that product has both Ethernet and Wi-Fi network interfaces.

You say Wi-Fi or Ethernet doesn't matter because it is hidden behind IP networking, but don't those interfaces have separate MAC addresses and therefore separate IPv6 addresses? 

If I were to disconnect the Wi-Fi radio of an already paired device and route all traffic through the Ethernet port, would that even work? I am thinking since IPv6 addresses are derived from the MAC addresses, the Matter accessory would be unreachable.

jonsmirl, 2023-03-13:
Yes, Matter would still work if you disconnect the wifi radio and use Ethernet, Matter is using IP networking and it does not rely on the MAC addresses. Matter nodes are located using mDNS on top of IP. This works even if the IP and MAC address change. Apple uses crazy rotating MAC addresses which change every few minutes.

Does your device function as a router? Is it routing IP traffic between Ethernet and Wifi independent of Matter? I suspect it isn't.  In that case I would suggest booting up in BLE/Wifi commissioning mode. Then detect if the Ethernet cable is plugged in. If Ethernet plugged in, switch from BLE/Wifi commissioning mode to Ethernet mode. Note that the only real difference between BLE/Wifi commissioning mode and Ethernet mode is that BLE/Wifi uses BLE to pass in your wifi SSID/PW.

JonathanWitthoeft, 2023-03-13:
Is this as simple as disabling Wi-Fi and BLE radios, or is there a clean API in this Matter SDK for disabling Wi-Fi/BLE commissioning and reflecting it is disabled? Similar Question for Ethernet, I do not see a single example of enabling Ethernet mode.

I see Wi-Fi commissioning is initialized here, but I don't know if there is a clean way to de-init it:
https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.cpp#L825-L827
```
static chip::app::Clusters::NetworkCommissioning::Instance sWiFiNetworkCommissioningInstance(0,
                                    &(chip::DeviceLayer::NetworkCommissioning::ESPWiFiDriver::GetInstance()));
sWiFiNetworkCommissioningInstance.Init();
```

jonsmirl, 2023-03-13:
@dhrishi  that's a question for the Espressif people, I have not tried doing that on the ESP platform.

jonsmirl, 2023-03-13:
Short term for testing, commission your device using BLE/Wifi and a working AP. Then reboot it onto Ethernet, that should work.

PSONALl, 2023-03-14:
```
static chip::app::Clusters::NetworkCommissioning::Instance sWiFiNetworkCommissioningInstance(0,
                                    &(chip::DeviceLayer::NetworkCommissioning::ESPWiFiDriver::GetInstance()));
sWiFiNetworkCommissioningInstance.Init();
```

@JonathanWitthoeft This happens when Wifi is enabled i.e, the CHIP_DEVICE_CONFIG_ENABLE_WIFI option is enabled
In the case of Ethernet since this config is disabled so it won't happen but 
I have tried creating an ESPEthernetDriver Instance for NetworkCommissioning in project-chip but still, the ethernet commissioning was not working with Apple Home and Google Home

JonathanWitthoeft, 2023-03-14:
@PSONALl Does this mean Ethernet Commissioning is unsupported in this SDK? 
Is there still an active effort to implement this, or is it low priority?

PSONALl, 2023-03-14:
@JonathanWitthoeft for ethernet commissioning we are facing issues with Apple Home and Google Home, We are checking them currently, But the ethernet commissioning works well with chip-tool, You can try it out using following steps:
1. While building an application, you need to enable `ENABLE_ETHERNET_TELEMETRY` config option, You can find it in menuconfig under: `Component config → CHIP Device Layer → Network Telemetry Options -> Enable Ethernet Telemetry`
2. And for commissioning use chip-tool and do on-network commissioning using this command:
    ```
    ./chip-tool pairing onnetwork <node-id> 20202021
    ```

jzwyssig, 2023-03-15:
@PSONALl I am also trying to commission via Ethernet. I use W5500 chipset and enabled the `ENABLE_ETHERNET_TELEMETRY` config option in menuconfig. When I try to commission with chip-tool it says `CHIP:DIS: Browsing for: _matterc._udp`. 

I get an IP address after the commission window is opened. Is that causing problem?  I also notice that the `OnEthernetPlatformEvent` is **not** executed when the `IP_EVENT_ETH_GOT_IP` occurs.

PSONALl, 2023-03-15:
@jzwyssig currently we support IP101 chipset for ethernet commissioning

h2zero, 2023-03-15:
I am having the same issue, I have modified the `ConnectivityManagerImpl::InitEthernet()` to support the rtl8201 which is working fine and getting an IP but commissioning fails. I have tested the modifications with the all-clusters-app example and commissioning works with chip-tool but does not with my esp-matter project.

dhrishi, 2023-03-15:
@h2zero Can you please upload the ESP side console logs as well as the chip-tool logs so that we can take a look?

h2zero, 2023-03-15:
@dhrishi Here you go, thanks for looking into it :smiley: 
[chip-tool-log.txt](https://github.com/espressif/esp-matter/files/10982872/chip-tool-log.txt)
[esp-log.txt](https://github.com/espressif/esp-matter/files/10982873/esp-log.txt)

dhrishi, 2023-03-15:
@h2zero 

The chip-tool-log.txt says:

>Failed in verifying 'Attestation Information' command received from the device: err 604. Look at AttestationVerificationResult enum to understand the errors

604 is `kCertificationDeclarationInvalidVendorId`

Can you confirm if the Vendor ID in the Certification Declaration matches that in the Basic Information cluster? Have you done any changes in the default app? Or changed the attestation certs (DAC, PAI) or the Certification Declaration (CD)? If yes, please provide the details

dhrishi, 2023-03-16:
@jzwyssig Can you please upload the ESP side console logs as well as the chip-tool logs so that we can take a look?

Sorry, I missed your previous comment :-)

dhrishi, 2023-03-29:
>After that I start matter. In order to make this work I had to comment out [esp_event_loop_create_default();](https://github.com/espressif/esp-matter/blob/af62fb0b254304e6114b91608957a421dd1ac58e/components/esp_matter/esp_matter_core.cpp#L956) and disable the initialization of the Ethernet driver in ConnectivityManagerImpl_Ethernet.cpp.

This is a bug. @PSONALl Please check

PSONALl, 2023-04-05:
> In order to make this work I had to comment out [...]

@jzwyssig there was an issue in ethernet commissioning on connectedhomeip due to the changes in Wifi Connectivity Manager but it has been fixed now and made be available soon

jzwyssig, 2023-04-07:
Thank you @PSONALl. Can I just pull the latest changes in esp-matter or do I also need to upgrade connectedhomeip?

PSONALl, 2023-04-12:
@jzwyssig you would need to upgrade the connectedhomeip because the fix will be available there

PSONALl, 2023-04-12:
@jzwyssig You also need to patch ConnectivityManagerImpl_Ethernet.cpp as it initialises network interface for ethernet, 
Which example are you trying out ? Are you commissioning using a Apple Home or chip-tool?

PSONALl, 2023-04-13:
yes, you need to add this patch: [ethernet-fix.patch](https://github.com/espressif/esp-matter/files/11219093/ethernet-fix.patch) on esp-matter side to get it working, Once the upstream PR gets merged, we will add this in esp-matter too

jzwyssig, 2023-04-19:
I already applied the patch as it is very similar to the code I posted [here](https://github.com/espressif/esp-matter/issues/271#issuecomment-1505001948) and I manually replaced the files of connectedhomeip-1.0.0.2 with your changes of the [upstream PR](https://github.com/project-chip/connectedhomeip/pull/25876/files). 
As I already mentioned, I use W5500 as my lan driver which is working perfectly fine. But after I commission with google home the bridged nodes are offline and with Apple HomeKit it is not working at all. I guess something is still not functioning correctly. I also upgraded to ESP-IDF-5.0.1, but the issue persists. 

What did you do to make it work?

riwalker, 2023-04-19:
@jzwyssig would you mind sharing your driver for the W5500?  I am using the W6100 and would also like to connect over ethernet.
also, either you or @PSONALl , would you be interested in some paid porting?  please email mrriwalker gmail
appreciated.

PSONALl, 2023-05-02:
> What did you do to make it work?

@jzwyssig the above two patches worked for me, I am not facing issues after commissioning the device and I can also control it from Apple Home, What issue are you facing? Can you share your device side logs here?

PSONALl, 2023-05-04:
@jzwyssig This looks like an issue with Google home, I will report it to Google issue tracker, but I am able to commission and control a bridged onoff light from Apple Home, Are you able to control your device using Apple Home now?

jzwyssig, 2023-05-04:
@PSONALI You are right! It works pefectly fine with the Apple Home app. Thank you for reporting to Google issue tracker.

jonsmirl, 2023-05-05:
I will confirm Google Home has these issues. I am hitting a similar issue which you will likely encounter too.

I'm using my own app to commission a BLE/wifi device from Android. Google does not want to share the wifi credentials with me, so when I ask Android to commission the device first Google does a hidden commissioning session to the device using its own fabric where sets in the Wifi credentials. That succeeds, it then opens a commissioning window and my commissioning proceeds over wifi (not BLE).  If this all works, everything is good.

But, let's say my app's commissioning fails for some reason. The device won't be commissioned in my app. So the user hits the commissioning button again. That calls the Google library which rescans the QRCode. Then the Google library is going to set there three minutes and time out.

Why does it do that? Because the hidden commissioning session has given the device wifi credentials so now it is going to use wifi for commissioning instead of BLE. When you retry commissioning, the retry sits there looking for the device on BLE since that is where the QRCode says it is. But Google has silently moved it onto Wifi. 
 
So now Google is stuck. They have to remember this secret commissioning session and then use it to OpenCommissioningWindow on the device using Wifi. That's the only way to get back into contact with the device. If they just sit there looking for it on BLE it is never going to be there unless you factory reset the device.  And that's how I am working around this, if my commissioning code gets an error, I have to factory reset the device.

JonathanWitthoeft, 2023-05-05:
@PSONALl Would you mind sharing your Ethernet Driver/Network Manager patches that work with the IP101? I am using a LAN8720 but that would give me a good starting point.

PSONALl, 2023-05-08:
> @PSONALl Would you mind sharing your Ethernet Driver/Network Manager patches that work with the IP101? I am using a LAN8720 but that would give me a good starting point.

@JonathanWitthoeft You can refer the basic ethernet example from esp-idf [here](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/basic)

PSONALl, 2023-05-17:
@jzwyssig It is raised on a Internal Issue Tracker, you wont have access for the link, There is no latest update on the issue

jzwyssig, 2023-05-17:
@PSONALl what brings you to the conclusion that this is a bug on Google side? Could you share some insights?

PSONALl, 2023-05-22:
> @PSONALl what brings you to the conclusion that this is a bug on Google's side? Could you share some insights?

@jzwyssig Ethernet commissioning works well with chip-tool so I think this is an issue on Google's side

jzwyssig, 2023-05-30:
@PSONALI Are there any news or is there any way to accelerate this issue? Today I updated to the newest esp-matter and connectedhomeip branch, but the problem persists. Commissioning is working, bridge is there (online), but bridged endpoints remain offline. Although it is possible to retrieve device informations when tapping on the device's properties. With Wi-Fi everything is working as expected. Thanks!

JonathanWitthoeft, 2023-06-14:
@PSONALl @mgoeller @jzwyssig I am now experiencing offline bridged endpoints with google using standard Wi-Fi commissioning. It was working at some point but lost functionality. I see the issue with this environment:
esp-idf 5.0.1:
https://github.com/espressif/esp-idf/tree/a4afa44435ef4488d018399e1de50ad2ee964be8

esp-matter:
https://github.com/espressif/esp-matter/tree/273efe2b7c4b8ae4eec257b01c11349a68429705

JonathanWitthoeft, 2023-06-14:
I loaded an old firmware binary that I knew worked with google home and I am experiencing the same issue. BLE commissioning to a Wi-Fi network and bridged endpoints are offline. So it seems to indeed be something google changed.

mgoeller, 2023-06-22:
For me the WiFi endpoints are still working if I compile using ESP-IDF 5.1 and the newest esp-matter, but still no progress on the wired side. This is very tiring! I wish Espressif would have more of a stand-point here towards Google, so many devices are based on an ESP...

And if its an error on Googles side, how do Philips Hue wired Ethernet Bridges work?

JonathanWitthoeft, 2023-06-26:
> @PSONALl @mgoeller @jzwyssig I am now experiencing offline bridged endpoints with google using standard Wi-Fi commissioning. It was working at some point but lost functionality. I see the issue with this environment:

It turns out my Google Home Mini was on the default "My Home" and I was trying to add the matter device to a separate Google Home name. This caused the accessories to pair, but show as offline. Once I had them both set up using the same home, everything started working again....my bad

mgoeller, 2023-06-27:
Ok, but the problem with Ethernet in Google Home still persists...

PSONALl, 2023-06-27:
@mgoeller I think this is some issue with Google Home App Commissioning, I have reported this problem to Google Issue tracker with all the details, I will let you know once there is an update

mgoeller, 2023-06-27:
@PSONALl I know, but this problem is persisting since 3 months now, and we urgently need a fix. If other platforms (ARM, Silabs etc.) manage to be able to matter pair and communicate via Ethernet (see examples on the market such as Philips Hue Bridge) it can not only be Googles fault...

PSONALl, 2023-06-27:
@mgoeller 

> If other platforms (ARM, Silabs etc.) manage to be able to matter pair and communicate via Ethernet

Oh, Ok I will look into this issue and will provide a fix soon

riwalker, 2023-06-27:
> Oh, Ok I will look into this issue and will provide a fix soon

FYI there are others of us also waiting for Ethernet to fully work.  Please fix the bug.

dhrishi, 2023-07-08:
@mgoeller @riwalker This should ideally fix the issue https://github.com/project-chip/connectedhomeip/pull/27808/files

We will soon update the connectedhomeip submodule in esp-matter with this. If you want, you can verify this fix by adding the changes locally

AtiosAG, 2023-07-10:
@dhrishi @PSONALl I can confirm Ethernet is working now with google home. As I use the W5500 chipset I already provided the necessary callbacks. Therefore, I cannot verify all of your changes. But the magic line that makes everything work is `ESP_ERROR_CHECK(esp_netif_create_ip6_linklocal(eth_netif_spi));` in the `ETHERNET_EVENT_CONNECTED` event.
Thanks!

jzwyssig, 2023-07-10:
@dhrishi @PSONALl I can confirm Ethernet is working now with google home. As I use the W5500 chipset I already provided the necessary callbacks. Therefore, I cannot verify all of your changes. But the magic line that makes everything work is `ESP_ERROR_CHECK(esp_netif_create_ip6_linklocal(eth_netif));` in the `ETHERNET_EVENT_CONNECTED` event.
Thanks!

jzwyssig, 2023-08-02:
@dhrishi  @PSONALl There is still a problem with ethernet: After successful pairing I add bridged endpoints. They work sometimes and go suddenly offline. I see the following debug output (imported are the last 4 lines):
```
I (340055) chip[DMG]: Registered a ReadHandler that will schedule a report between system Timestamp: 340057 and system Timestamp 340057.
I (340065) chip[IM]: Resuming subscriptionId 2607589860
I (340265) chip[DIS]: Checking node lookup status after 200 ms
I (340575) chip[DIS]: UDP:192.168.188.156%en1:5540: new best score: 2
I (340575) chip[DIS]: UDP:[2001:1708:2580:DD01:104E:1640:A26B:D121%en1]:5540: new best score: 4
I (340585) chip[DIS]: UDP:[2001:1708:2580:DD01:215F:5313:96E5:B383%en1]:5540: score has not improved: 4
I (340595) chip[DIS]: UDP:[2001:1708:2580:DD01:313E:CF5E:3D31:90D0%en1]:5540: score has not improved: 4
I (340605) chip[DIS]: UDP:[2001:1708:2580:DD01:7DA1:14F9:A889:64AE%en1]:5540: score has not improved: 4
I (340615) chip[DIS]: Checking node lookup status after 545 ms
I (340615) chip[SC]: Initiating session on local FabricIndex 1 from 0x000000001D16B0FB -> 0x00000000E34EC3E0
I (340685) chip[EM]: <<< [E:52478i S:0 M:73105355] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (340685) chip[IN]: (U) Sending msg 73105355 to IP address 'UDP:[2001:1708:2580:DD01:104E:1640:A26B:D121]:5540'
I (340695) chip[SC]: Sent Sigma1 msg
I (341045) chip[IN]: (U) Sending msg 73105355 to IP address 'UDP:[2001:1708:2580:DD01:104E:1640:A26B:D121]:5540'
I (341405) chip[IN]: (U) Sending msg 73105355 to IP address 'UDP:[2001:1708:2580:DD01:104E:1640:A26B:D121]:5540'
I (342005) chip[IN]: (U) Sending msg 73105355 to IP address 'UDP:[2001:1708:2580:DD01:104E:1640:A26B:D121]:5540'
E (342015) chip[EM]: Ignoring transient send error: 3000001 on exchange 52478i
I (342905) chip[IN]: (U) Sending msg 73105355 to IP address 'UDP:[2001:1708:2580:DD01:104E:1640:A26B:D121]:5540'
E (342905) chip[EM]: Ignoring transient send error: 3000001 on exchange 52478i
E (344525) chip[EM]: Failed to Send CHIP MessageCounter:73105355 on exchange 52478i sendCount: 4 max retries: 4
E (346905) chip[SC]: CASESession timed out while waiting for a response from the peer. Current state was 4
E (346905) chip[DMG]: Failed to establish CASE for subscription-resumption with error '32'
```

What could this be related to?

mgoeller, 2023-08-02:
@dhrishi I experience the same errors as @jzwyssig

PSONALl, 2023-08-04:
@jzwyssig @mgoeller Thank you for notifying of this issue. I have raised a PR to connectedhomeip here: https://github.com/project-chip/connectedhomeip/pull/28520 which solves this issue, We will merge it soon to the esp-matter submodule and I will notify you once it gets merged

dhrishi, 2023-09-14:
Fixed with 5d94b80517e7b1940667b43d7ebe4ef8c33a66d1

closed 2023-09-14 by dhrishi