espressif/esp-matter #272: apple home remove device

shun-dgut, 2023-03-11:
when apple home remove device，device will call  esp_matter_command_callback_remove_fabric ，if i want application know this event，what should I do.
[image]

shubhamdp, 2023-03-12:
esp-matter exposes that as an event `kFabricRemoved` https://github.com/espressif/esp-matter/blob/d886f9c76513417abab80ccbb25d3a730a2bbbd9/components/esp_matter/esp_matter.h#L128-L129

Find the usage in light example, [here](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L63). (NOTE: code inside the case starts the commissionable advertising on dns-sd on removal of last fabric).

If you are curious about the implementation, please find below more details.

You have to implement the `chip::FabricTable::Delegate` class and register it with fabrics table.
https://github.com/espressif/esp-matter/blob/d886f9c76513417abab80ccbb25d3a730a2bbbd9/components/esp_matter/esp_matter_core.cpp#L121-L147

https://github.com/espressif/esp-matter/blob/d886f9c76513417abab80ccbb25d3a730a2bbbd9/components/esp_matter/esp_matter_core.cpp#L797

shubhamdp, 2023-03-14:
@shun-dgut any update on this?

dhrishi, 2023-03-15:
I will close this as the fabric removed event is already present and received. Please re-open if you still see an issue.

closed 2023-03-15 by dhrishi