espressif/esp-matter #273: This accessory is not currently supported by the Home app. in the Home app.

vaemc, 2023-03-14:
Hello:

Run [generic_switch](https://github.com/espressif/esp-matter/tree/matter_1_1/te1/examples/generic_switch) example on the esp32c2 

The home app shows the Add this matter device, but says "Not Supported".

My apple ios version is 16.3.1

May I ask what is the reason

[image]

shubhamdp, 2023-03-14:
Hi @vaemc, by default this example implements [latching switch](https://github.com/espressif/esp-matter/blob/matter_1_1/te1/examples/generic_switch/main/Kconfig.projbuild#L5) and that is shows as unsupported matter accessory in Apple Home.

Momentary switch are supported in Apple Home. You will have to enable `CONFIG_ GENERIC_SWITCH_TYPE_MOMENTARY ` using menuconfig, it can be found in menu path: `(Top) -> Demo -> Generic Switch Type`

vaemc, 2023-03-14:
@shubhamdp Thanks for the quick response！ 
I try to change it to `CONFIG_GENERIC_SWITCH_TYPE_MOMENTARY`, it still doesn't support

shubhamdp, 2023-03-14:
Please check the attached screenshot if it's any help. My Hompod is on Version 16.3.2

[image]

dhrishi, 2023-03-20:
@vaemc Please close the issue if you are able to see the momentary switch in the Home app.

vaemc, 2023-03-20:
Thank you very much for your enthusiastic help, it should be my problem, I will try again

closed 2023-03-20 by vaemc