espressif/esp-matter #274: Why can't I get USB jtag serial IO working?

jonsmirl, 2023-03-14:
why does this simple program not work? I am using S3.
~~~
extern "C" void app_main(void)
{
    nvs_flash_init();

    /* Disable buffering on stdin */
    setvbuf(stdin, NULL, _IONBF, 0);

    /* Minicom, screen, idf_monitor send CR when ENTER key is pressed */
    esp_vfs_dev_usb_serial_jtag_set_rx_line_endings(ESP_LINE_ENDINGS_CR);
    /* Move the caret to the beginning of the next line on '\n' */
    esp_vfs_dev_usb_serial_jtag_set_tx_line_endings(ESP_LINE_ENDINGS_CRLF);

    /* Enable non-blocking mode on stdin and stdout */
    fcntl(fileno(stdout), F_SETFL, O_NONBLOCK);
    fcntl(fileno(stdin), F_SETFL, O_NONBLOCK);

    usb_serial_jtag_driver_config_t usb_serial_jtag_config;
    usb_serial_jtag_config.rx_buffer_size = 256;
    usb_serial_jtag_config.tx_buffer_size = 256;

    esp_err_t ret = ESP_OK;
    /* Install USB-SERIAL-JTAG driver for interrupt-driven reads and writes */
    ret = usb_serial_jtag_driver_install(&usb_serial_jtag_config);
    if (ret != ESP_OK) {
        printf("usb_serial_jtag_driver_install fail\n");
        return;
    }

    /* Tell vfs to use usb-serial-jtag driver */
    esp_vfs_usb_serial_jtag_use_driver();
    esp_vfs_dev_uart_register();

    while (true) {
        unsigned char c;
        usb_serial_jtag_read_bytes(&c, 1, 0);
        usb_serial_jtag_write_bytes(&c, 1, 0);
    }
}
~~~

PSONALl, 2023-03-16:
This code reads a single byte from the USB-to-serial adapter using the `usb_serial_jtag_read_bytes` function, and then immediately writes that byte back to the adapter using the `usb_serial_jtag_write_bytes` function.

One potential issue with this code can be : 
The `usb_serial_jtag_read_bytes` and `usb_serial_jtag_write_bytes` functions are blocking functions, which means that they will wait until data is available to be read or written before returning. In this code, if there is no data available to be read from the adapter, the `usb_serial_jtag_read_bytes` function will block the loop and prevent any further execution of the code.
The code can be modified as follows:
 ```
esp_err_t ret = usb_serial_jtag_read_bytes(&c, 1, 1000); // timeout after 1 second
    if (ret == ESP_OK) {
        usb_serial_jtag_write_bytes(&c, 1, 0);
    }
```

In above code `usb_serial_jtag_read_bytes` function will wait for up to 1 second for data to be available and will return an error code if no data is available within that time. If data is available, the byte is read and immediately written back to the adapter using the `usb_serial_jtag_write_bytes` function

PSONALl, 2023-03-16:
@jonsmirl can you also provide additional information like, what issue you are facing exactly and what you are trying to achieve?
because there are some possible reasons why the code may not work like: missing header files or hardware issues or incorrect configurations of the USB-to-serial adapter

jonsmirl, 2023-03-16:
I want to turn off the esp-matter REPL console and replace it with my own I/O.  I can do this without issue on a UART but  I can't get the USB Serial com port to work.

Here is another symptom:
 printf("X");
 usb_serial_jtag_write_bytes(&"Y", 1, 0);

You will see the X and not the Y.  So I am able to printf over USB serial but the read/write byte functions don't work. If I use the UART I see both X and Y;

 printf("X");
 uart_write_bytes(&"Y", 1, 0);

That loop should have had an if in it. 
~~~
    while (true) {
        unsigned char c;
        if (usb_serial_jtag_read_bytes(&c, 1, 0))
             usb_serial_jtag_write_bytes(&c, 1, 0);
    }
~~~
Can you provide me with the simplest possible program which will just echo back what you type using /dev/ttyACM0? The USB serial code is not broken, the REPL console obviously uses it.  So how do I get it into a state where I can use it? This might be some issue with the VFS having control of it and I don't know how to turn it off.

jonsmirl, 2023-03-24:
I am attempting to avoid addressing this issue by using the code in the esp-console component instead. I don't have it completely working yet but it seems to be a possible solution.

closed 2023-03-24 by jonsmirl