espressif/esp-matter #275: Getting stuck on ble comissioning

ivishaltejwani, 2023-03-16:
I (32018) NimBLE: GAP procedure initiated: advertise;
I (32028) NimBLE: disc_mode=2
I (32038) NimBLE: adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (32048) NimBLE:

this is where it gets stuck , I can see in apple home as matter accessory but cannot add it
same with google home it sometimes appear as matter device found but never gets added to the app

no changes in the example code

I have not used homepod or google nest mini to add , are they necessary to add in apple home?

I have tried idf 5.0.1 and idf 4.4.4 same issue with both , esp matter example as well as connectedhomeip example
ESP32-D0WD (revision 1)
ESP32-D0WD (revision 3)

both of the chips are tested

os- macos

Platform
esp32

Platform Version(s)
ESPIDF 5.0.1
ESPIDF 4.4.4

dhrishi, 2023-03-16:
@ivishaltejwani The log shows that the commissioning does not even start. Can you try with host based chip-tool once?

Yes. Google and Apple both require a hub (like a Nest Hub or HomePod) to be present in the same network for successful commissioning and control.

Also, note that, for ESP32 we support IDF v4.4.3 right now. See [this](https://github.com/espressif/esp-matter#supported-esp-idf-and-connectedhomeip-versions)

dhrishi, 2023-03-17:
Closing this as this is being discussed in https://github.com/project-chip/connectedhomeip/issues/25707

closed 2023-03-17 by dhrishi