espressif/esp-matter #276: Initialize the mesh node causing the program to crash

maenkai, 2023-03-16:
I use the routine example/blemesh_bridge.
Use esp_ble_mesh_node_prov_enable to become a mesh node.
I used my mobile phone for mesh allocation, and the following error occurred. How should I solve it

```c
assert failed: bt_mesh_gatts_service_stop mesh_bearer_adapt.c:1283 (rc == 0)
Backtrace:0x40081cce:0x3fff74300x40090ab1:0x3fff7450 0x400971a5:0x3fff7470 0x4013cb56:0x3fff7590 0x401222ff:0x3fff75c0 0x4011ec0a:0x3fff75e0 0x40120f97:0x3fff7610 0x401214f8:0x3fff76b0 0x40121dba:0x3fff76e0 0x40121f63:0x3fff7760 0x4013c445:0x3fff7780 0x40138caf:0x3fff77a0 0x40138ce5:0x3fff77d0 0x401425c5:0x3fff7810 0x401427b3:0x3fff7840 0x4014333b:0x3fff7860 0x401342f9:0x3fff7890 0x4013419e:0x3fff78c0 0x40133a15:0x3fff78f0 0x40133a27:0x3fff7910 0x4013b256:0x3fff7930 0x4016e90f:0x3fff7950 0x40093a96:0x3fff7970 
0x40081cce: panic_abort at /home/mek/esp/esp-idf/components/esp_system/panic.c:402

0x40090ab1: esp_system_abort at /home/mek/esp/esp-idf/components/esp_system/esp_system.c:128

0x400971a5: __assert_func at /home/mek/esp/esp-idf/components/newlib/assert.c:85

0x4013cb56: bt_mesh_gatts_service_stop at /home/mek/esp/esp-idf/components/bt/esp_ble_mesh/mesh_core/nimble_host/mesh_bearer_adapt.c:1283 (discriminator 1)

0x401222ff: bt_mesh_proxy_server_prov_disable at /home/mek/esp/esp-idf/components/bt/esp_ble_mesh/mesh_core/proxy_server.c:781

0x4011ec0a: bt_mesh_provision at /home/mek/esp/esp-idf/components/bt/esp_ble_mesh/mesh_core/main.c:55

0x40120f97: prov_data at /home/mek/esp/esp-idf/components/bt/esp_ble_mesh/mesh_core/prov.c:1256

0x401214f8: bt_mesh_pb_gatt_recv at /home/mek/esp/esp-idf/components/bt/esp_ble_mesh/mesh_core/prov.c:1691

0x40121dba: proxy_complete_pdu at /home/mek/esp/esp-idf/components/bt/esp_ble_mesh/mesh_core/proxy_server.c:487 (discriminator 3)

0x40121f63: proxy_recv at /home/mek/esp/esp-idf/components/bt/esp_ble_mesh/mesh_core/proxy_server.c:577

0x4013c445: proxy_char_access_cb at /home/mek/esp/esp-idf/components/bt/esp_ble_mesh/mesh_core/nimble_host/mesh_bearer_adapt.c:1594

0x40138caf: ble_gatts_val_access at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gatts.c:395

0x40138ce5: ble_gatts_chr_val_access at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gatts.c:421

0x401425c5: ble_att_svr_write at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:524

0x401427b3: ble_att_svr_write_handle at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:555

0x4014333b: ble_att_svr_rx_write_no_rsp at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:2038

0x401342f9: ble_att_rx at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:503

0x4013419e: ble_hs_hci_evt_acl_process at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:927

0x40133a15: ble_hs_process_rx_data_queue at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:238

0x40133a27: ble_hs_event_rx_data at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:532

0x4013b256: ble_npl_event_run at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:125
 (inlined by) nimble_port_run at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:78

0x4016e90f: chip::DeviceLayer::Internal::BLEManagerImpl::bleprph_host_task(void*) at /home/mek/code/Lipro/project/meizu_light/build/esp-idf/chip/../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:712

0x40093a96: vPortTaskWrapper at /home/mek/esp/esp-idf/components/freertos/port/xtensa/port.c:131

maenkai, 2023-03-20:
Now check this area.
```c
I (00:00:14.826) chip[DL]: BLE GAP connection established (con 0)
I (00:00:14.826) NimBLE: connection established; status=0 
E (00:00:14.826) chip[DL]: adv stop
I (00:00:14.828) NimBLE: 

I (00:00:14.832) chip[DL]: CHIPoBLE advertising stopped
I (00:00:14.841) NimBLE: GAP procedure initiated: advertise; 
I (00:00:14.853) NimBLE: disc_mode=2
I (00:00:14.857) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (00:00:14.868) NimBLE: 

I (00:00:14.868) CHIP[DL]: ESP_BLE_MESH_STOP_BLE_ADVERTISING_COMP_EVT, err_code 0
E (21555) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (21555) task_wdt:  - IDLE (CPU 0)
E (21555) task_wdt: Tasks currently running:
E (21555) task_wdt: CPU 0: ble
E (21555) task_wdt: CPU 1: IDLE
E (21555) task_wdt: Aborting.

abort() was called at PC 0x4019c8fb on core 0
0x4019c8fb: task_wdt_isr at /home/mek/esp/esp-idf/components/esp_system/task_wdt.c:176 (discriminator 3)

Backtrace:0x40081cce:0x3ffbe8700x40090ab1:0x3ffbe890 0x400970ae:0x3ffbe8b0 0x4019c8fb:0x3ffbe920 0x400831a5:0x3ffbe940 0x4013fc4e:0x3fff32f0 0x40131389:0x3fff3370 0x4013122e:0x3fff33a0 0x40130aa5:0x3fff33d0 0x40130ab7:0x3fff33f0 0x40137d52:0x3fff3410 0x4016b3f3:0x3fff3430 0x40093a96:0x3fff3450 
0x40081cce: panic_abort at /home/mek/esp/esp-idf/components/esp_system/panic.c:402

0x40090ab1: esp_system_abort at /home/mek/esp/esp-idf/components/esp_system/esp_system.c:128

0x400970ae: abort at /home/mek/esp/esp-idf/components/newlib/abort.c:46

0x4019c8fb: task_wdt_isr at /home/mek/esp/esp-idf/components/esp_system/task_wdt.c:176 (discriminator 3)

0x400831a5: _xt_lowint1 at /home/mek/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x4013fc4e: ble_att_svr_build_read_group_type_rsp at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:1731
 (inlined by) ble_att_svr_rx_read_group_type at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:1918

0x40131389: ble_att_rx at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:503

0x4013122e: ble_hs_hci_evt_acl_process at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:927

0x40130aa5: ble_hs_process_rx_data_queue at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:238

0x40130ab7: ble_hs_event_rx_data at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:532

0x40137d52: ble_npl_event_run at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:125
 (inlined by) nimble_port_run at /home/mek/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:78

0x4016b3f3: chip::DeviceLayer::Internal::BLEManagerImpl::bleprph_host_task(void*) at /home/mek/code/Lipro/project/meizu_light/build/esp-idf/chip/../../../../../../../esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:670

0x40093a96: vPortTaskWrapper at /home/mek/esp/esp-idf/components/freertos/port/xtensa/port.c:131

InfiniteYuan, 2023-03-21:
Could you please provide the reproducible exmpale？

closed 2023-04-03 by maenkai