espressif/esp-matter #277: OTA esp Matter:Received QueryImage failure

HKSIWAR, 2023-03-16:
Hello,
I have two esp32, the first one is provider and flashed by the ota-provider example, the second one is requestor and flashed by the ota-requestor example.
ota-requestor. then, i put both devices in the same network. finally, i run the AnnounceOTAProvider command:
out/chip-tool otasoftwareupdaterequestor announce-otaprovider 0xDEADBEEF 0 0 0x1234567890 0.
The following figure shows the error message:
[image]

dhrishi, 2023-03-16:
@HKSIWAR Are you directly using the upstream connectedhomeip `ota-requestor-app` and `ota-provider-app` and not the esp-matter examples? Please provide details about the examples used and branch/tag/commit you are using.

If you are using v1.0.0.2 connectedhomeip tag, have you followed the instructions here: https://github.com/project-chip/connectedhomeip/tree/v1.0.0.2/examples/ota-requestor-app/esp32#prerequisite ? If you face the issue even after following the documentation, then please upload (and not paste) the entire logs on OTA provider, requestor and the chip-tool

HkiriSiwar, 2023-03-16:
the following paths that i used are :

-  https://github.com/espressif/connectedhomeip/tree/v1.0.0.2/examples/ota-provider-app/esp32
- https://github.com/espressif/connectedhomeip/tree/v1.0.0.2/examples/ota-requestor-app/esp32

the steps are :
-  comissioning provider : I use the following command "sudo out/debug/chip-tool pairing ble-wifi 12345 Orange-710D qtFAgq2N2h8 20202021 3840"
- comissioning requestor : I use the following command "sudo out/debug/chip-tool pairing ble-wifi 1234 Orange-710D qtFAgq2N2h8 20202021 3840"
- update image: I use the following command "sudo ./out/debug/chip-tool otasoftwareupdaterequestor announce-ota-provider 12345 0 0 0 1234 0"
I use the version 1.0.0.2 and the logs output are  :
 **- Log output ota-requestor :**
[log_requestor.log](https://github.com/espressif/esp-matter/files/10995109/log_requestor.log)
 **- Log output ota-provider :**
[log_provider.log](https://github.com/espressif/esp-matter/files/10995117/log_provider.log)
 **- Log chip-tools :**
[log_chip_tool.log](https://github.com/espressif/esp-matter/files/10995121/log_chip_tool.log)

jadhavrohit924, 2023-03-17:
Hi @HkiriSiwar,
From the logs, looks like you have missed one step of `Writing into Access Control List`.
Please check the steps [here](https://github.com/project-chip/connectedhomeip/tree/v1.0.0.2/examples/ota-provider-app/esp32#access-control-list-requirements)

dhrishi, 2023-04-06:
@HKSIWAR This issue has been fixed in the upstream codebase https://github.com/project-chip/connectedhomeip/pull/25859

This will be addressed in the esp-matter repository once the connectedhomeip submodule is updated, having this commit. For now, you can try this on your side by applying the fix in `esp-matter/connectedhomeip/connectedhomeip`

jadhavrohit924, 2023-04-07:
@HkiriSiwar I could able to perform OTA on lateste esp matter. 
From your logs, looks like the problem is in your OTA image. Can you verify you have OTA image with updated software version ( i.e OTA image software version > current image software version )?

dhrishi, 2023-04-12:
@HkiriSiwar Is your issue resolved? Please close this if it was.

dhrishi, 2023-04-27:
I will close this now. Please reopen in case of anything

closed 2023-04-27 by dhrishi