espressif/esp-matter #279: git reports filename too long

ChrisEAlfred, 2023-03-17:
On Windows:

git.exe clone --progress --recursive -v "https://github.com/espressif/esp-matter.git" "C:\MYDEV\esp-matter"

reports:

fatal: cannot write keep file 'C:/MYDEV/esp-matter/.git/modules/connectedhomeip/connectedhomeip/modules/third_party/mbed-mcu-boot/repo/modules/boot/espressif/hal/esp-idf/modules/components/bootloader/subproject/components/micro-ecc/micro-ecc/objects/pack/pack-565943cb8d3498c31f42c650fde09ed4d3676a1b.keep': Filename too long

This is the first of many filenames too long.

dhrishi, 2023-03-17:
@ChrisEAlfred Are you directly using Windows as the development environment? Upstream connectedhomeip [officially supports](https://github.com/espressif/connectedhomeip/blob/v1.0.0.2/docs/guides/BUILDING.md#building-matter) Ubuntu 20.04 or 22.04 LTS, Debian 11 (64 bit) and MacOS 10.15 or later.

In this repository, we have additionally supported Windows 10 using WSL. Please see [this](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#windows-10) and check if it helps.

dhrishi, 2023-04-04:
I see that this is being tracked through https://github.com/project-chip/connectedhomeip/issues/25726 . I will close this issue now.

closed 2023-04-04 by dhrishi