espressif/esp-matter #281: heatin/cooling unit device type

stage-01, 2023-03-21:
I'm working on developing a smart heating unit for in-home use to which I'd like to add matter functionality. When looking in the matter device library specifications the category that suits this best would be a heating/cooling unit. However I can't find this device type in the esp-matter endpoints. I've noticed google doesn't support this device type either. 

Is there any way I can implement this device type in my heater? Or is there any way for me to just give it basic on/off functionality?

dhrishi, 2023-03-21:
@stage-01 As per the Matter v1.0 specification, the support for Heating/Cooling Unit is provisional. This means that it is a part of the specification, but the implementation is not certifiable in the v1.0 release. This can be the reason that the ecosystems don't support it yet.

>Or is there any way for me to just give it basic on/off functionality?

Basic on/off functionality can be achieved through `On/Off Plug-in Unit`

stage-01, 2023-03-22:
Thanks for de confirmation. I will close this issue because my question has been answered. 
It's too bad that there is not a way to just have a standard "matter device" that can have the on/off functionality seeing as it would be a bit weird to buy a heater and have to set it up in google home as a smart plug. Guess matter v1.0 and the implementation in google home is still lacking a bit too much features for it to be useable for most home appliances.

closed 2023-03-22 by stage-01

dhrishi, 2023-03-22:
@stage-01 Even though it is not directly supported, Google Home (as well as Apple Home) gives an option to represent the outlet as per the connected accessory type. After commissioning the outlet, click on it and then choose settings and then the device type. List has Light, AC, Fan, Dehumidifer, Heater, Kettle, TV, Switch etc. You can choose as per your accessory. Then it changes the icon for the accessory accordingly and you can control it using the voice commands too. Hope this helps.

stage-01, 2023-03-22:
Yes I've found that feature and that is what I will be using for test purposes. That sadly wont really work for an end product :( Thank you very much for your help!

germouse12, 2023-03-22:
Would the "Thermostat" device type meet your needs? Your application would need to operate autonomously based on the set temperature, but you could control that set temperature through Google Home and the like.

stage-01, 2023-03-23:
I might be able to get that to work. It wont be as good of a fit as the heating/cooling unit device type but it's better than nothing. Thanks for the suggestion