espressif/esp-matter #282: Some dependencies missing for WSL

MartyMcFlyInTheSky, 2023-03-22:
I tried to set up matter in WSL 2.0 running ubuntu 22.. Eventually  I got it to work but there's some information missing in the [official doc](https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#development-setup). Namely `pkg-config` and glib-2.0 are missing in a default installation of WSL ubuntu and should be installed first. Before I got this error:
Error:

```
  Setting up CIPD package manager...done (3.9s)
  Setting up Python environment.....[/]
ERROR at //build/config/linux/pkg_config.gni:116:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out/
Command: python3 /home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/build/config/linux/pkg-config.py openssl
Returned 1.
stderr:

Could not run pkg-config.

See //src/crypto/BUILD.gn:84:3: whence it was called.
  pkg_config("openssl_config") {
  ^-----------------------------
See //BUILD.gn:152:7: which caused the file to be included.
      "${chip_root}/src/crypto",
      ^------------------------
['gn', 'gen', '/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']

Traceback (most recent call last):
  File "/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 335, in install_packages
    subprocess.check_call(
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gn', 'gen', '/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 857, in <module>
    sys.exit(main())
  File "/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 849, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 501, in setup
    result = step(spin)
  File "/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 661, in virtualenv
    if not virtualenv_setup.install(
  File "/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 370, in install
    install_packages(gn_target)
  File "/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 343, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['gn', 'gen', '/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/home/glades/dev/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit s
tatus 1.```

Installing the missing packages gets rid of the problem:

`apt-get install glib-2.0 pkg-config`

dhrishi, 2023-03-23:
@MartyMcFlyInTheSky Thank you for reporting the issue. Can you please check if this is covered through the [connectedhomeip pre-requisites](https://github.com/espressif/connectedhomeip/blob/v1.0.0.2/docs/guides/BUILDING.md#installing-prerequisites-on-linux) mentioned in the esp-matter documentation [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#host-setup)

shubhamdp, 2023-03-29:
@MartyMcFlyInTheSky Do you have any update on this?

Last step in setting up WSL docs mentions that
> Here onwards process for setting esp-matter and building examples is same as other hosts.

I suppose after following those steps and installing prerequisites, this error would go away.

closed 2023-03-29 by dhrishi

kunjumonkk, 2023-08-06:
It got fixed for me after installing following pkg and able to install matter successfully:

sudo apt-get install git gcc g++ pkg-config libssl-dev libdbus-1-dev
libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev
python3-pip unzip libgirepository1.0-dev libcairo2-dev libreadline-dev