espressif/esp-matter #283: ESP-IDF v5.xx for matter enabled product development

aggaddam, 2023-03-23:
Hi, 

My company is planning to start development for matter-enabled IOT products. We want to start off our development with IDF v5.xx instead of v4.xx and our development will get completed in the next year 2023. Considering long-term support till 2025 by espressif for IDF v5.xx, we would like to start matter development with v5.xx instead of v4.xx  

I would like to hear some suggestions from this team regarding IDF versions

[image]

shubhamdp, 2023-03-29:
@aggaddam We have recently released the support of esp-matter branch on IDF v5.xx. Latest `main` branch of esp-matter supports IDF commit [bb9200ace](https://github.com/espressif/esp-idf/tree/bb9200ace) in branch `release/v5.1` and main branch will continue to keep updating support for IDF v5.xx.

In a nutshell, you can start off your development with esp-matter main branch with IDF commit bb9200ace from `release/v5.1` branch.

aggaddam, 2023-03-29:
@shubhamdp  Thanks for the update. 

When is the official release planned for IDF v5.1?
When is the official release planned for ESP-Matter SDK?

dhrishi, 2023-03-29:
@aggaddam

ESP IDF v5.1 release is planned around end of May this year.

ESP Matter v1.0 release will soon be created from release/v1.0 branch which is based of upstream v1.0.0.2

ESP Matter v1.1 is now the main branch. This will be released after official release of Matter v1.1 from upstream, which isn't done yet.

dhrishi, 2023-04-04:
Closing this as the original question is answered.

closed 2023-04-04 by dhrishi