espressif/esp-matter #284: blemesh_bridge_attribute_update

jonsmirl, 2023-03-24:
blemesh_bridge_attribute_update() has started using the priv_data field. So how do I fix this in my code? priv_data is not NULL anymore for bridged endpoints.   Can we force the endpoint_id the bridge allocates into a specific range, say over 0x1000? This needs to only effect bridge allocated EPs. I am looking at dynamically allocated EPs for my own use too.
~~~
    if (type == PRE_UPDATE)
    {
        if (priv_data == NULL)  // change to if (endpoint_id >= 0x1000) ??
        {
            err = blemesh_bridge_attribute_update(endpoint_id, cluster_id, attribute_id, val, priv_data);
        }
        else
        {
            Ep_Light *epLight = (Ep_Light *)priv_data;
            err = epLight->attribute_update(endpoint_id, cluster_id, attribute_id, val);
        }
    }
~~~

jonsmirl, 2023-03-24:
In esp_matter_bridge.h this logic is not going to work...
~~~
#define MAX_BRIDGED_DEVICE_COUNT \
    CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT - 1 - CONFIG_ESP_MATTER_AGGREGATOR_ENDPOINT_COUNT
// There is an endpoint reserved as root endpoint
~~~
I have both fixed device endpoints and bridged device endpoint on the same device. This is ignoring my fixed endpoints which also come out of the CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT pool.

jonsmirl, 2023-03-24:
This should not be inside the "C" region
https://github.com/espressif/esp-matter/blob/main/examples/blemesh_bridge/main/blemesh_bridge.h#L21

jonsmirl, 2023-03-24:
The bridge code can not make the assumption it is the only thing running on the device. It is perfectly legal to have local devices in addition to the bridge and I'm doing that. The bridge code also can not assume it is the only entity using dynamic endpoints.

I don't need this, but it is certainly legal to run both the BLE and Zigbee bridge on a the same device and also have local devices. Someone is likely to do that on the esp32c6.

dhairyashah1, 2023-03-27:
@jonsmirl 

> Can we force the endpoint_id the bridge allocates into a specific range, say over 0x1000?
- This is a custom usecase. As per your requirement, you may pass `NULL` for `priv_data` as suitable in the [`app_main.cpp`](https://github.com/espressif/esp-matter/blob/main/examples/blemesh_bridge/main/app_main.cpp#L70) of  your extended version of the `blemesh_bridge` app. 
- You may handle it further in the [`blemesh_bridge.cpp`](https://github.com/espressif/esp-matter/blob/f2e77d239e0a2125cf2a5a2bc72ac86ab1631645/examples/blemesh_bridge/main/blemesh_bridge.cpp#L74) by checking whether the argument `app_bridged_device_t *bridged_device` (which is basically *priv_data) is NULL like follows -
```
esp_err_t blemesh_bridge_attribute_update(uint16_t endpoint_id, uint32_t cluster_id, uint32_t attribute_id,
                                          esp_matter_attr_val_t *val, app_bridged_device_t *bridged_device)
{
    if(bridged_device==NULL)
        *bridged_device = app_bridge_get_device_by_matter_endpointid(endpoint_id);
    if (bridged_device && bridged_device->dev && bridged_device->dev->endpoint) {
        if (cluster_id == OnOff::Id) {
    . . . 
```

> This is ignoring my fixed endpoints which also come out of the CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT pool.
- For the `blemesh_bridge` the `FIXED_ENDPOINT_COUNT` is `0` [by default](https://github.com/espressif/esp-matter/blob/f2e77d239e0a2125cf2a5a2bc72ac86ab1631645/examples/blemesh_bridge/main/zap-generated/endpoint_config.h#L46). This macro is passed on to the chip-code where it is [handled separately](https://github.com/espressif/connectedhomeip/blob/903e149bd636a5cd1a8821864cb073b4f7a9d3d8/src/app/util/attribute-storage.h#LL67-L71C48)
- Hence, the `FIXED_ENDPOINTS` donot come from the pool of `CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT`.
- If you have fixed endpoints, you can modify `MAX_BRIDGED_DEVICE_COUNT` as - 
```
#define MAX_BRIDGED_DEVICE_COUNT \
    CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT + FIXED_ENDPOINT_COUNT - 1 - CONFIG_ESP_MATTER_AGGREGATOR_ENDPOINT_COUNT
```

> I don't need this, but it is certainly legal to run both the BLE and Zigbee bridge on a the same device and also have local devices. Someone is likely to do that on the esp32c6.
- In that custom case, the `FIXED_ENDPOINT_COUNT` and `MAX_BRIDGED_DEVICE_COUNT` as stated above must be updated accordingly.

jonsmirl, 2023-04-06:
app_bridge_get_device_by_matter_endpointid() does not exist in current esp-matter code.

A function like that would solve my problem, if it is a bridged device it would return *bridged_device, and if not NULL. Now I can tell a bridged endpoint from a non-bridged one.

dhairyashah1, 2023-04-26:
The function `app_bridged_device_t app_bridge_get_device_by_matter_endpointid()` : [link](https://github.com/espressif/esp-matter/pull/266/files) was removed in the PR #266 (which made use of `*priv_data` to get the instance of the blemesh / zigbee bridge device, instead of searching it by the `endpoint_id`) for reducing complexity.

You can utilize its implementation reference.

jonsmirl, 2023-10-31:
I solved this by copying common/app_bridge_device.(cpp&h) into my project and then altering the structures. Altering the structures allowed me to unify the priv_data used by app_bridged_device_t with my own code. Since the struct is now unified I know the format and can decode it.

closed 2023-10-31 by jonsmirl